srsran/srsRAN_4G #412: Support for LTE B66?

dnelson82, 2019-11-20:
Before filing an issue, search for solutions here:
- srsLTE users mailing list (http://www.softwareradiosystems.com/mailman/listinfo/srslte-users)

## Issue Description ##
Does the latest release of _srslte_ support ENB transmission in LTE band 66 and has it been validated?

## Setup Details ##
I am running _srsepc_ and _srsenb_ on a local Ubuntu 18.04 LTS machine with an attached USRP B210. 

## Expected Behavior ##
The srsenb transmitting a LTE B66 carrier

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

codershiyar, 2022-12-20:
Is there any solution for this problem ?

ismagom, 2023-04-14:
Unfortunately that band is not supported

closed 2023-04-14 by ismagom

codershiyar, 2024-03-12:
why not? hope you can answer me, thanks in advanced

codershiyar, 2025-05-07:
Is there any updates or fixes related to Band 66 support since this thread was opened? Or maybe some configuration tips that might help?

Any clarification would be very helpful. Thanks in advance!    

@ismagom @dnelson82 @frankist  @andrepuschmann