srsran/srsRAN_4G #422: Packege not found issue when cmake

qiushishen, 2019-11-30:
Before filing an issue, search for solutions here:
- srsLTE users mailing list (http://www.softwareradiosystems.com/mailman/listinfo/srslte-users)

## Issue Description ##
I'm using Ubuntu 18.04 LTS, following the steps of **srslte-docs.pdf**. Some packages are missing when cmake, but actually I installed every packages using **sudo apt-get install cmake libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev**

## Setup Details ##

## Expected Behavior ##
cmake without errors

## Actual Behaviour ##
~/Desktop/SRS/srsLTE/build$ cmake ../
-- CMAKE_SYSTEM: Linux-5.0.0-23-generic
-- CMAKE_SYSTEM_PROCESSOR: x86_64
-- CMAKE_CXX_COMPILER: /usr/bin/c++
-- Build type not specified: defaulting to Release.
-- FFTW3F LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.a
-- FFTW3F INCLUDE DIRS: /usr/local/include
-- FFT_LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- Checking for module 'polarssl'
--   No package 'polarssl' found
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND
-- POLARSSL INCLUDE DIRS: POLARSSL_INCLUDE_DIRS-NOTFOUND
-- Could NOT find POLARSSL (missing: POLARSSL_LIBRARIES POLARSSL_INCLUDE_DIRS) 
-- Checking for module 'mbedtls'
--   No package 'mbedtls' found
-- MBEDTLS LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.a
-- MBEDTLS INCLUDE DIRS: /usr/include
-- Checking for module 'libpcsclite'
--   No package 'libpcsclite' found
-- Could NOT find PCSCLITE (missing: PCSCLITE_LIBRARY PCSCLITE_INCLUDE_DIR) 
-- PCSC LIBRARIES: PCSCLITE_LIBRARY-NOTFOUND
-- PCSC INCLUDE DIRS: PCSCLITE_INCLUDE_DIR-NOTFOUND
-- UHD LIBRARIES /usr/lib/x86_64-linux-gnu/libuhd.so
-- UHD INCLUDE DIRS /usr/include
-- Checking for module 'libbladeRF'
--   No package 'libbladeRF' found
-- libbladeRF not found.
-- FINDING SOAPY.
-- Checking for module 'SoapySDR'
--   No package 'SoapySDR' found
-- libSOAPYSDR not found.
-- FINDING ZEROMQ.
-- Checking for module 'ZeroMQ'
--   No package 'ZeroMQ' found
-- libZEROMQ not found.
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   program_options
-- SRSGUI LIBRARIES /usr/local/lib/libsrsgui.so
-- SRSGUI INCLUDE DIRS /usr/local/include
-- SSE4.1 is enabled - target CPU must support it
-- AVX is enabled - target CPU must support it
-- AVX2 is enabled - target CPU must support it
-- FMA is enabled - target CPU must support it
-- Using default glibc header for IPv6 support.
-- SSE4.1 is enabled - target CPU must support it
-- AVX is enabled - target CPU must support it
-- AVX2 is enabled - target CPU must support it
-- FMA is enabled - target CPU must support it
-- CMAKE_C_FLAGS is  -Werror=incompatible-pointer-types -Wall -Wno-comment -Wno-write-strings -Winline -Wno-unused-result -Wformat -Wmissing-field-initializers -Wtype-limits -std=c99 -D_GNU_SOURCE -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -mfma -DLV_HAVE_FMA -Ofast -funroll-loops -fvisibility=hidden -Werror
-- CMAKE_CXX_FLAGS is  -Wall -Wno-comment -Wno-reorder -Wno-unused-but-set-variable -Wno-unused-variable -Wtype-limits -std=c++11 -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -fvisibility=hidden -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE -DUSE_GLIBC_IPV6 -Werror
-- Using install prefix: /usr/local
-- Building for version: 19.9.0
--    examples will be installed.
-- No post-build command defined
-- Building with srsUE
-- No post-build-UE command defined
-- No post-build command defined
-- Building with srsENB
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/x86_64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/x86_64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-ENB command defined
-- Building with srsEPC
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/x86_64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/x86_64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-EPC command defined
-- Configuring done
-- Generating done
-- Build files have been written to: /home/shenqiushi/Desktop/SRS/srsLTE/build

## Steps to reproduce the problem ##
git clone https://github.com/srsLTE/srsLTE.git
cd srsLTE
mkdir build
cd build
cmake ../

## Additional Information ##

andrepuschmann, 2019-12-12:
run `make`

closed 2019-12-12 by andrepuschmann

ashwini1890, 2023-05-26:
hy i get the same error and if i run "make" it gives an error or uninitialized variable and stops.
the error is: 

/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:315:44: error: ‘old[7]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  315 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:690:31: note: ‘old[7]’ was declared here
  690 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:315:44: error: ‘old[6]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  315 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:690:31: note: ‘old[6]’ was declared here
  690 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:315:44: error: ‘old[5]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  315 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:690:31: note: ‘old[5]’ was declared here
  690 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:315:44: error: ‘old[4]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  315 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:690:31: note: ‘old[4]’ was declared here
  690 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:315:44: error: ‘old[3]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  315 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:690:31: note: ‘old[3]’ was declared here
  690 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:315:44: error: ‘old[2]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  315 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:690:31: note: ‘old[2]’ was declared here
  690 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:315:44: error: ‘old[1]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  315 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:690:31: note: ‘old[1]’ was declared here
  690 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:315:44: error: ‘old[0]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  315 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/home/ubuntu/srsRAN_4G/lib/include/srslte/phy/fec/turbodecoder_win.h:690:31: note: ‘old[0]’ was declared here
  690 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
cc1: all warnings being treated as errors
make[2]: *** [lib/src/phy/fec/CMakeFiles/srslte_fec.dir/build.make:216: lib/src/phy/fec/CMakeFiles/srslte_fec.dir/turbodecoder.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2836: lib/src/phy/fec/CMakeFiles/srslte_fec.dir/all] Error 2
make: *** [Makefile:166: all] Error 2

kindly help with the same