srsran/srsRAN_4G #423: Please Help to me ? I cant install srslte . How can I fix this phy files ?

muratoznn, 2019-12-05:
Please Help to me ? I cant install srslte . How can I fix this phy files ?
make is giving to me error

98% tests passed, 9 tests failed out of 552

Total Test time (real) = 612.76 sec

The following tests FAILED:
321 - phy_dl_test-p6-t2-q-m27 (Failed)
326 - phy_dl_test-p6-t3-q-m27 (Failed)
331 - phy_dl_test-p6-t4-q-m27 (Failed)
386 - phy_dl_test-p25-t3-m28 (Failed)
391 - phy_dl_test-p25-t4-m28 (Failed)
401 - phy_dl_test-p25-t2-q-m27 (Failed)
406 - phy_dl_test-p25-t3-q-m27 (Failed)
411 - phy_dl_test-p25-t4-q-m27 (Failed)
551 - nas_test (Child aborted)
Errors while running CTest
make: *** [Makefile:152: test] Error 8

xavierarteaga, 2019-12-05:
Hi, we know there is a defect in channel coding for some CPU architectures. Would you mind providing us with your CPU model and flags? `$ cat /proc/cpuinfo`

muratoznn, 2019-12-05:
This İs My Cpu İnformations . How Can I fix İt ?

root@kali:~/srsLTE/build# cat /proc/cpuinfo
processor : 0
vendor_id : GenuineIntel
cpu family : 6
model : 122
model name : Intel(R) Celeron(R) N4000 CPU @ 1.10GHz
stepping : 1
microcode : 0x32
cpu MHz : 2310.950
cache size : 4096 KB
physical id : 0
siblings : 2
core id : 0
cpu cores : 2
apicid : 0
initial apicid : 0
fpu : yes
fpu_exception : yes
cpuid level : 24
wp : yes
flags : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg cx16 xtpr pdcm sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave rdrand lahf_lm 3dnowprefetch cpuid_fault cat_l2 pti cdp_l2 ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust smep erms mpx rdt_a rdseed smap clflushopt intel_pt sha_ni xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts umip rdpid md_clear arch_capabilities
bugs : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass
bogomips : 2188.80
clflush size : 64
cache_alignment : 64
address sizes : 39 bits physical, 48 bits virtual
power management:

processor : 1
vendor_id : GenuineIntel
cpu family : 6
model : 122
model name : Intel(R) Celeron(R) N4000 CPU @ 1.10GHz
stepping : 1
microcode : 0x32
cpu MHz : 2050.759
cache size : 4096 KB
physical id : 0
siblings : 2
core id : 1
cpu cores : 2
apicid : 2
initial apicid : 2
fpu : yes
fpu_exception : yes
cpuid level : 24
wp : yes
flags : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg cx16 xtpr pdcm sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave rdrand lahf_lm 3dnowprefetch cpuid_fault cat_l2 pti cdp_l2 ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust smep erms mpx rdt_a rdseed smap clflushopt intel_pt sha_ni xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts umip rdpid md_clear arch_capabilities
bugs : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass
bogomips : 2188.80
clflush size : 64
cache_alignment : 64
address sizes : 39 bits physical, 48 bits virtual
power management:

root@kali:~/srsLTE/build#

________________________________
Gönderen: Xavier Arteaga <notifications@github.com>
Gönderildi: 5 Aralık 2019 Perşembe 10:39
Kime: srsLTE/srsLTE <srsLTE@noreply.github.com>
Bilgi: muratoznn <murat_1.2@hotmail.com>; Author <author@noreply.github.com>
Konu: Re: [srsLTE/srsLTE] Please Help to me ? I cant install srslte . How can I fix this phy files ? (#423)

Hi, we know there is a defect in channel coding for some CPU architectures. Would you mind providing us with your CPU model and flags? $ cat /proc/cpuinfo

xavierarteaga, 2019-12-05:
Thanks. There is a defect for Intel SSE4 architectures. You need to use a CPU with AVX2 until we fix this defect. However, you can still use srsLTE, just consider the DL BLER will be higher than it should be.

closed 2019-12-05 by xavierarteaga

muratoznn, 2019-12-05:
Okay I installed srslte with this errors but I am taking this tx calibration error. What Can I do for this ?

Why srslte is breaking my limesdr calibration opinions ?

After I am calibrating it with limesuite Start sudo srsenb  . Tx Calibration Settings is breaking : S

root@kali:~/srsLTE/srsenb# sudo srsenb

Built in Release mode using commit 0e89fa9f on branch master.

— Software Radio Systems LTE eNodeB —

Reading configuration file /root/.config/srslte/enb.conf…
Opening 1 RF devices with 1 RF channels…
[INFO] [UHD] linux; GNU C++ version 9.2.1 20191109; Boost_106700; UHD_3.14.1.1-1+b1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP with args:
[INFO] [UHDSoapyDevice] Make connection: ‘LimeSDR Mini [USB 2.0] 1D4C3893CE8AAC’
[INFO] [UHDSoapyDevice] Reference clock 40.00 MHz
[INFO] [UHDSoapyDevice] Device name: LimeSDR-Mini
[INFO] [UHDSoapyDevice] Reference: 40 MHz
[INFO] [UHDSoapyDevice] LMS7002M calibration values caching Disable
Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz
Setting Sampling frequency 11.52 MHz

Warning TX/RX time offset has not been calibrated for device uhd_unknown. Set a value manually

[FATAL] [UHDSoapyDevice] Tx Calibration: MCU error 5 (Loopback signal weak: not connected/insufficient gain?)
[INFO] [UHDSoapyDevice] Rx calibration finished

________________________________
Gönderen: Xavier Arteaga <notifications@github.com>
Gönderildi: 5 Aralık 2019 Perşembe 13:23
Kime: srsLTE/srsLTE <srsLTE@noreply.github.com>
Bilgi: muratoznn <murat_1.2@hotmail.com>; Author <author@noreply.github.com>
Konu: Re: [srsLTE/srsLTE] Please Help to me ? I cant install srslte . How can I fix this phy files ? (#423)

Thanks. There is a defect for Intel SSE4 architectures. You need to use a CPU with AVX2 until we fix this defect. However, you can still use srsLTE, just consider the DL BLER will be higher than it should be.

twei7, 2020-03-11:
Hi, I am also hitting the same test error. My CPU seems to have AVX instruction set.

99% tests passed, 8 tests failed out of 587

Total Test time (real) = 297.96 sec

The following tests FAILED:
	357 - phy_dl_test-p6-t2-q-m27 (Failed)
	362 - phy_dl_test-p6-t3-q-m27 (Failed)
	367 - phy_dl_test-p6-t4-q-m27 (Failed)
	422 - phy_dl_test-p25-t3-m28 (Failed)
	427 - phy_dl_test-p25-t4-m28 (Failed)
	437 - phy_dl_test-p25-t2-q-m27 (Failed)
	442 - phy_dl_test-p25-t3-q-m27 (Failed)
	447 - phy_dl_test-p25-t4-q-m27 (Failed)
Errors while running CTest
Makefile:140: recipe for target 'test' failed
make: *** [test] Error 8

processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 45
model name	: Intel(R) Xeon(R) CPU E5-2690 0 @ 2.90GHz
stepping	: 7
microcode	: 0x718
cpu MHz		: 1197.299
cache size	: 20480 KB
physical id	: 0
siblings	: 16
core id		: 0
cpu cores	: 8
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx lahf_lm epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips	: 5785.70
clflush size	: 64
cache_alignment	: 64
address sizes	: 46 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 45
model name	: Intel(R) Xeon(R) CPU E5-2690 0 @ 2.90GHz
stepping	: 7
microcode	: 0x718
cpu MHz		: 1196.923
cache size	: 20480 KB
physical id	: 0
siblings	: 16
core id		: 1
cpu cores	: 8
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx lahf_lm epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips	: 5785.70
clflush size	: 64
cache_alignment	: 64
[... 787 lines cut ...]
vendor_id	: GenuineIntel
cpu family	: 6
model		: 45
model name	: Intel(R) Xeon(R) CPU E5-2690 0 @ 2.90GHz
stepping	: 7
microcode	: 0x718
cpu MHz		: 1196.988
cache size	: 20480 KB
physical id	: 1
siblings	: 16
core id		: 7
cpu cores	: 8
apicid		: 47
initial apicid	: 47
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx lahf_lm epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips	: 5787.45
clflush size	: 64
cache_alignment	: 64
address sizes	: 46 bits physical, 48 bits virtual
power management: