srsran/srsRAN_4G #424: failed to build ubuntu 14.04 lts on 75%.

Keskebeu, 2019-12-12:
I have build and run the system in 16.04 with no problems. I am trying to build and run it in 14.04 where I read that srsLTE should not have any problems building in 14.04LTS. 

the output is as follows

In file included from /home/imsicatcher/srslte/lib/include/srslte/upper/ipv6.h:26:0,
                 from /home/imsicatcher/srslte/srsue/src/stack/upper/gw.cc:23:
/usr/include/linux/ipv6.h:19:8: error: redefinition of ‘struct in6_pktinfo’
 struct in6_pktinfo {
        ^
In file included from /home/imsicatcher/srslte/lib/include/srslte/phy/io/netsink.h:35:0,
                 from /home/imsicatcher/srslte/lib/include/srslte/srslte.h:77,
                 from /home/imsicatcher/srslte/lib/include/srslte/common/common.h:82,
                 from /home/imsicatcher/srslte/lib/include/srslte/common/buffer_pool.h:37,
                 from /home/imsicatcher/srslte/srsue/hdr/stack/upper/gw.h:26,
                 from /home/imsicatcher/srslte/srsue/src/stack/upper/gw.cc:22:
/usr/include/netinet/in.h:534:8: error: previous definition of ‘struct in6_pktinfo’
 struct in6_pktinfo
        ^
In file included from /home/imsicatcher/srslte/lib/include/srslte/upper/ipv6.h:26:0,
                 from /home/imsicatcher/srslte/srsue/src/stack/upper/gw.cc:23:
/usr/include/linux/ipv6.h:24:8: error: redefinition of ‘struct ip6_mtuinfo’
 struct ip6_mtuinfo {
        ^
In file included from /home/imsicatcher/srslte/lib/include/srslte/phy/io/netsink.h:35:0,
                 from /home/imsicatcher/srslte/lib/include/srslte/srslte.h:77,
                 from /home/imsicatcher/srslte/lib/include/srslte/common/common.h:82,
                 from /home/imsicatcher/srslte/lib/include/srslte/common/buffer_pool.h:37,
                 from /home/imsicatcher/srslte/srsue/hdr/stack/upper/gw.h:26,
                 from /home/imsicatcher/srslte/srsue/src/stack/upper/gw.cc:22:
/usr/include/netinet/in.h:541:8: error: previous definition of ‘struct ip6_mtuinfo’
 struct ip6_mtuinfo
        ^
make[2]: *** [srsue/src/stack/upper/CMakeFiles/srsue_upper.dir/gw.cc.o] Error 1
make[1]: *** [srsue/src/stack/upper/CMakeFiles/srsue_upper.dir/all] Error 2
make: *** [all] Error 2

Any help is appreciated it.

andrepuschmann, 2019-12-12:
Just run cmake with `-DUSE_GLIBC_IPV6=False` on those systems.

closed 2019-12-12 by andrepuschmann

Keskebeu, 2019-12-13:
Thank you for your prompt answer Andre. Did not mean to open an issue but I'd rather ask the developer. Thanks again.

zeroplus4, 2019-12-24:
Hello .please i dont know if you know how to resolve this sigploit ss7 parameters connection..how can i get all the details.