srsran/srsRAN_4G #427: Bearer DRB2 already configured. Reconfiguration not supported

jdomnitz, 2019-12-15:
## Issue Description ##
I'm attempting to setup an eNodeB with two APNs one for internet and one for ims.  The initial attachment to the ims APN works correctly but when the UE attempts to attach to the internet APN I get a "[PDCP] [W] Bearer DRB2 already configured. Reconfiguration not supported" error and a malformed RRCConnectionReconfiguration message is transmitted.  I do not see this issue when using a single APN.

## Setup Details ##
eNodeB configuration straight out of the *.sample files.  Using Open5gs as the EPC.

## Expected Behavior ##
After the attachment a valid connection reconfiguration should be transmitted.

## Actual Behaviour ##
A malformed packet is transmitted (appears to be an empty payload).
[ENB-Failure.zip](https://github.com/srsLTE/srsLTE/files/3965291/ENB-Failure.zip)
[ENB-Failure.zip](https://github.com/srsLTE/srsLTE/files/3965294/ENB-Failure.zip)

## Steps to reproduce the problem ##
See setup details

## Additional Information ##
See attached packet capture:
[ENB-Failure.zip](https://github.com/srsLTE/srsLTE/files/3965295/ENB-Failure.zip)

andrepuschmann, 2019-12-16:
We've never tested dual APN with srsENB. Can you send a full eNB log and PCAP of the S1AP?

jdomnitz, 2019-12-17:
Is there another way to get VoLTE working?  I think this is the last hurdle.  Everything else seems to be working.

eNB log and S1AP pcap attached.  I logged at the debug level since it shows the empty buffer before transmission.

[enb.zip](https://github.com/srsLTE/srsLTE/files/3970943/enb.zip)
[s1ap.zip](https://github.com/srsLTE/srsLTE/files/3970944/s1ap.zip)

andrepuschmann, 2019-12-17:
Hey Justin, we've identified an issue with the bearer setup. It should be fixed in the upcoming 19.12 release but I can't promise that there are no other issues with VoLTE. Anyway, wait until it's out and feel free to test it and reopen the issue again if it still fails.

closed 2019-12-17 by andrepuschmann

jdomnitz, 2019-12-18:
I didn't see an associated commit.  Is there a patch available?  I would like to test it before release and see if I can flush out any other VoLTE issues.

andrepuschmann, 2019-12-18:
Have a look at the `master_merge_next` branch. [This](https://github.com/srsLTE/srsLTE/commit/75672324b90564b2c519f070ffcb847dd0d4fad7) commit should fix the bearer setup.

jdomnitz, 2019-12-23:
Confirmed fixed :+1:

andrepuschmann, 2019-12-24:
Thanks for confirming Justin.