srsran/srsRAN_4G #429: How Can I Fix These errors ? I can Not install ?

muratoznn, 2019-12-16:
How Can I fix This errors ?  I am using intel celeron cpu. 
PLease help ?  I want to use srslte but I cant install it  ?

98% tests passed, 9 tests failed out of 552

Total Test time (real) = 641.50 sec

The following tests FAILED:
	321 - phy_dl_test-p6-t2-q-m27 (Failed)
	326 - phy_dl_test-p6-t3-q-m27 (Failed)
	331 - phy_dl_test-p6-t4-q-m27 (Failed)
	386 - phy_dl_test-p25-t3-m28 (Failed)
	391 - phy_dl_test-p25-t4-m28 (Failed)
	401 - phy_dl_test-p25-t2-q-m27 (Failed)
	406 - phy_dl_test-p25-t3-q-m27 (Failed)
	411 - phy_dl_test-p25-t4-q-m27 (Failed)
	551 - nas_test (Child aborted)
Errors while running CTest
make: *** [Makefile:152: test] Error 8

andrepuschmann, 2019-12-16:
There is currently a bug in the non-AVX implementation. Please use a CPU with AVX

closed 2019-12-16 by andrepuschmann

muratoznn, 2019-12-16:
İf I Try With Rasperry Pi ?  I Can Run İt ?

________________________________
Gönderen: Andre Puschmann <notifications@github.com>
Gönderildi: 16 Aralık 2019 Pazartesi 22:56
Kime: srsLTE/srsLTE <srsLTE@noreply.github.com>
Bilgi: muratoznn <murat_1.2@hotmail.com>; Author <author@noreply.github.com>
Konu: Re: [srsLTE/srsLTE] How Can I Fix These errors ? I can Not install ? (#429)

There is currently a bug in the non-AVX implementation. Please use a CPU with AVX