srsran/srsRAN_4G #434: Error parsing in srsenb

asp2k5, 2019-12-22:
## Issue Description ##

I'm having the following error when I attempt to run **sudo srsenb** with a
**USRP B210**. I havent changed any of default enb config file. can anyone pls help out. Thankx.

**---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/esb3283/.config/srslte/enb.conf...
Error parsing field m_ri in section cqi_report_cnfg
Error parsing section phy_cnfg
Error parsing Radio Resources configuration
Error processing arguments.**

frankist, 2019-12-22:
Overwrite the conf files in the `~/.config/srslte` directory, by first erasing the `.conf` in that directory, and then running `srslte_install_configs.sh` that is present in your build folder. Please, use the mailing list for issues

closed 2019-12-22 by frankist

asp2k5, 2019-12-24:
thnkx it worked.