srsran/srsRAN_4G #435: Dedicated Bearer Support

jdomnitz, 2019-12-23:
With the last set of fixes I have VoLTE registration succeeding.

Dedicated Bearers seem to be the last hurdle for VoLTE calls.  Since there is already some support in UE would it be possible to get them on the ENB roadmap?

andrepuschmann, 2019-12-24:
They are on the roadmap.

akibsayyed, 2020-01-20:
I would like to know if there is dedicated bearer support in SRSUE ??

andrepuschmann, 2020-01-20:
It's already supported in the UE, yes.

akibsayyed, 2020-01-28:
Does ue support ims apn?

jla91, 2020-01-29:
Is there any chance that the dedicated bearer support on srsenb will be in the next release?

herlesupreeth, 2020-02-04:
A quick update to people testing dedicated bearer support in srsenb, its already supported. I was able to do a volte call which uses dedicated bearer (srsLTE enb + open5gs + Kamailio IMS), please find the pcap attached

[srslte_debug.zip](https://github.com/srsLTE/srsLTE/files/4154388/srslte_debug.zip)

[image]

All you have to do to place a VoLTE call is the following changes in drb.conf file

drb.conf
<pre>
// All times are in ms. Use -1 for infinity, where available

qci_config = (

{
  qci=1;
  pdcp_config = {
    discard_timer = 50;                
    pdcp_sn_size = 12;                  
  }
  rlc_config = {
    ul_um = {
      sn_field_length = 10; 
    };
    dl_um = {
      sn_field_length = 10; 
      t_reordering    = 40;             
    };
  };
  logical_channel_config = {
    priority = 1; 
    prioritized_bit_rate   = 8; 
    bucket_size_duration  = 100; 
    log_chan_group = 0; 
  };
},
{
  qci=5;
  pdcp_config = {
    discard_timer = 50;
    status_report_required = true;
  }
  rlc_config = {
    ul_am = {
      t_poll_retx = 50;
      poll_pdu = 64;
      poll_byte = 100;
      max_retx_thresh = 16;
    };
    dl_am = {
      t_reordering = 40;
      t_status_prohibit = 25;
    };
  };
  logical_channel_config = {
    priority = 6;
    prioritized_bit_rate   = 8;
    bucket_size_duration  = 100;
    log_chan_group = 2;
  };
},
{
  qci=7;
  pdcp_config = {
    discard_timer = 100;                
    pdcp_sn_size = 12;                  
  }
  rlc_config = {
    ul_um = {
      sn_field_length = 10; 
    };
    dl_um = {
      sn_field_length = 10; 
      t_reordering    = 45;             
    };
  };
  logical_channel_config = {
    priority = 13; 
    prioritized_bit_rate   = -1; 
    bucket_size_duration  = 100; 
    log_chan_group = 2; 
  };
},
{
  qci=9;
  pdcp_config = {
    discard_timer = -1;
    status_report_required = true;
  }
  rlc_config = {
    ul_am = {
      t_poll_retx = 120;
      poll_pdu = 64;
      poll_byte = 750;
      max_retx_thresh = 16;
    };
    dl_am = {
      t_reordering = 50;
      t_status_prohibit = 50;
    };
  };
  logical_channel_config = {
    priority = 11; 
    prioritized_bit_rate   = -1; 
    bucket_size_duration  = 100; 
    log_chan_group = 3; 
  };
}

);
</pre>

twei7, 2020-03-23:
Hi, are there any instructions of how to test the VoLTE registration of srsUE (probably with commercial eNodeB)?

herlesupreeth, 2020-03-23:
i am not sure whether it possible, but I would definitely try experimenting the following things

1. Have a look at following page to know the list of open-source IMS clients and see which one is apt for you. And install whichever you like

https://nil.uniza.sk/six-open-source-ims-clients-features-overview-february-2012/

2. And, once you have srsUE connected to eNB, use the IMS client installed before to send a SIP register (make sure to hardcode PCSCF IP in IMS client)

akibsayyed, 2020-03-23:
>    1. And, once you have srsUE connected to eNB, use the IMS client

Yes but issue is we need to use 2 apn
One for data and other for ims

How do we solve this

herlesupreeth, 2020-03-23:
In that case, one could define a dedicated radio bearer for IMS signalling under the single apn (like 'default' apn or something). Since the IMS registration would be completely under your control you dont need two apns. Check in this link the Case 1 scenario

andrepuschmann, 2020-10-27:
Issue seems to be solved. Closing therefore.

closed 2020-10-27 by andrepuschmann

northmirko, 2021-06-21:
Hello, Does **SRSUE** support Dedicated Bearer ?

andrepuschmann, 2021-06-21:
Please see the post above. Quick answer, yes, they are supported.

joeldelosangeles, 2021-11-26:
Hi,

We are encountering an error in /tmp/enb.log.  [E] QCI=6 not configured.  During attach, the enodeB sends an InitialContextSetupFailure [Radionetwork-cause=unspecified.   We are integrating with a big telco and they do not want to adjust the QCI because the setup is working when non-srs enodeb's are being used.  How do we configure QCI from 1 to 9 or for now even just QCI=6.  The UE's can camp when QCI=8 but not when QCI=6.  Do we just need to put configuration in drb.conf?  If so how?  Or do we need to edit the code?

Thanks,
Joel

andrepuschmann, 2021-11-26:
Just add it to your `drb.conf` and you're done

joeldelosangeles, 2021-12-17:
getting this error:  

Invalid option: 1 for field "discard_timer"  

even though we did not configure any qci setting as such

joeldelosangeles, 2021-12-18:
Hi,

I followed the recommended drb.conf but it seems there is necessary formatting of the text content.  See error below:

/usr/local/src/srsLTE/srsenb/src/enb_cfg_parser.cc.481: Error can't find t_poll_retx in section ul_am

/usr/local/src/srsLTE/srsenb/src/enb_cfg_parser.cc.486: Error can't find poll_pdu in section ul_am

/usr/local/src/srsLTE/srsenb/src/enb_cfg_parser.cc.491: Error can't find poll_byte in section ul_am

/usr/local/src/srsLTE/srsenb/src/enb_cfg_parser.cc.497: Error can't find max_retx_thresh in section ul_am

andrepuschmann, 2021-12-20:
Share your config file and version and we can check. Without that information, nobody can help you.

joeldelosangeles, 2021-12-20:
Apologies Andre.  The version is 45486b6 (20.10.1) and the drb.conf is attached
[drb.txt](https://github.com/srsran/srsRAN/files/7744484/drb.txt)
.