srsran/srsRAN_4G #437: Configuring LimeSDR-mini with srsLTE

mikehibbett, 2019-12-26:
I have had many difficulties getting LimeSDR mini to work with srsLTE. None of the online guides worked for me.

I now have a tested setup that works, for me, which I would like to share with others. Apologies for sharing on the 'issues' list, but this is where most people will look for solutions. Admins, feel free to close - unless others repeat this sequence and have issues, I will be happy to assist resolving.

Setup
===

My system is an Intel i7-3770T CPU, 8GB RAM, 128GB SSD. 
I am running Ubuntu 18.04.3 in a dual boot configuration. The setup described below follows from a fresh installation of the OS, as I wanted a reproducible setup from a clean install. Having worked on this for days, I have got the time for the sequence of "erase old OS, install new OS, build system, get srsLTE running" down to a 25 minute 48 seconds process. I'm sure this time could be improved :)

This setup was created by following and then adjusting the steps from the following two links:

https://docs.srslte.com/en/latest/general/source/1_installation.html
https://en.dlyang.me/limesdr-mini-on-ubuntu/

Setup steps.
========

Start with the LimeSDR mini not plugged into the PC.

Install Ubuntu 18.04.3 64 bit desktop OS.
Wait for the OS to advise of updates; install these and reboot.

Open a terminal shell. Enter the following commands, one at a time:

sudo add-apt-repository -y ppa:pothosware/framework
sudo add-apt-repository -y ppa:pothosware/support
sudo add-apt-repository -y ppa:myriadrf/drivers
sudo apt-get update
sudo apt-get install pothos-all -y
sudo apt-get install python-pothos -y
sudo apt-get install python3-pothos -y
sudo apt-get install pothos-python-dev -y
sudo apt-get install soapysdr-tools -y
sudo apt-get install python-soapysdr python-numpy -y
sudo apt-get install python3-soapysdr python3-numpy -y
sudo apt-get install soapysdr-module-remote soapysdr-server -y
sudo apt remove limesdr0.6-module-audio -y
sudo apt install soapysdr-module-all -y
sudo add-apt-repository -y ppa:myriadrf/drivers
sudo apt-get update
sudo apt-get install limesuite liblimesuite-dev limesuite-udev limesuite-images -y
sudo apt-get install soapysdr soapysdr-module-lms7 -y

Now plug your LimeSDR mini into a USB3 port, and run the command:

SoapySDRUtil --probe

In the output displayed, confirm the SDR is detected, and that it is on a USB3 port. If not, try another USB port.

Now enter the following commands, one at a time:

sudo apt-get install libsoapysdr-dev cmake libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev git build-essential -y

git clone https://github.com/srsLTE/srsLTE.git
cd srsLTE
mkdir build
cd build
cmake ../
make
sudo make install
sudo ldconfig
sudo srslte_install_configs.sh user

That completes the installation. Before running the code, I edited the enb config file to get the transmission on a frequency (band20) that is used by my phone in Europe.

In the file ~/.config/srslte/enb.conf change the [rf] section to say
[rf]
dl_earfcn = 6200
tx_gain = 56
rx_gain = 38
device_name = soapy
device_args = rxant=LNAH,txant=BAND2

Now open up two terminal windows. In one, run the command
sudo srsepc

In the other, run
sudo srsenb

In the output of the above you should see the SDR is detected. The final output should be calibration completed and, in my case, a single "[WARNING] L" message. From this point, I have confirmed on a spectrum analyser that the enb is transmitting. 

I have yet to confirm cell information detection by an actual mobile phone, but those are next steps, followed by full connectivity with a test SIM.

SDR configuration
============

The configuration of the LimeSDR mini used in the tests above can been seen below.

SoapySDRUtil --probe

# resulting output:

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D4976B5C14D1A'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M calibration values caching Disable

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=FT601
  hardware=LimeSDR-Mini
  boardSerialNumber=0x1d4976b5c14d1a
  firmwareVersion=6
  gatewareVersion=1.30
  hardwareVersion=2
  protocolVersion=1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: YES
  Sensors: clock_locked, lms7_temp
  Registers: BBIC
  GPIOs: MAIN

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32, CS12, CS16
  Native format: CS16 [full-scale=2048]
  Stream args:
     * Buffer Length - The buffer transfer size over the link.
       [key=bufferLength, units=samples, default=0, type=int]
     * Link Format - The format of the samples over the link.
       [key=linkFormat, default=CS16, type=string, options=(CS16, CS12)]
     * Skip Calibration - Skip automatic activation calibration.
       [key=skipCal, default=false, type=bool]
  Antennas: NONE, LNAH, LNAL_NC, LNAW, Auto
  Corrections: DC removal, DC offset, IQ balance
  Full gain range: [-12, 61] dB
    TIA gain range: [0, 12] dB
    LNA gain range: [0, 30] dB
    PGA gain range: [-12, 19] dB
  Full freq range: [0, 3800] MHz
    RF freq range: [30, 3800] MHz
    BB freq range: [-10, 10] MHz
  Tune args:
     * LO Offset - Tune the LO with an offset and compensate with the baseband CORDIC.
       [key=OFFSET, units=Hz, default=0.0, type=float, range=[-1e+07, 1e+07]]
     * BB - Specify a specific value for this component or IGNORE to skip tuning it.
       [key=BB, units=Hz, default=DEFAULT, type=float, range=[-1e+07, 1e+07], options=(DEFAULT, IGNORE)]
  Sample rates: [0.1, 65] MSps
  Filter bandwidths: [1.4, 130] MHz
  Sensors: lo_locked
  Other Settings:
     * TSP DC Level - Digital DC level in LMS7002M TSP chain.
       [key=TSP_CONST, type=int, range=[0, 32767]]

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32, CS12, CS16
  Native format: CS16 [full-scale=2048]
  Stream args:
     * Buffer Length - The buffer transfer size over the link.
       [key=bufferLength, units=samples, default=0, type=int]
     * Link Format - The format of the samples over the link.
       [key=linkFormat, default=CS16, type=string, options=(CS16, CS12)]
     * Skip Calibration - Skip automatic activation calibration.
       [key=skipCal, default=false, type=bool]
  Antennas: NONE, BAND1, BAND2, Auto
  Corrections: DC offset, IQ balance
  Full gain range: [-12, 64] dB
    PAD gain range: [0, 52] dB
    IAMP gain range: [-12, 12] dB
  Full freq range: [0, 3800] MHz
    RF freq range: [30, 3800] MHz
    BB freq range: [-10, 10] MHz
  Tune args:
     * LO Offset - Tune the LO with an offset and compensate with the baseband CORDIC.
       [key=OFFSET, units=Hz, default=0.0, type=float, range=[-1e+07, 1e+07]]
     * BB - Specify a specific value for this component or IGNORE to skip tuning it.
       [key=BB, units=Hz, default=DEFAULT, type=float, range=[-1e+07, 1e+07], options=(DEFAULT, IGNORE)]
  Sample rates: [0.1, 65] MSps
  Filter bandwidths: [5, 40], [50, 130] MHz
  Sensors: lo_locked
  Other Settings:
     * TSP DC Level - Digital DC level in LMS7002M TSP chain.
       [key=TSP_CONST, type=int, range=[0, 32767]]

andrepuschmann, 2019-12-26:
Thanks Mike for the detailed report. Such a manual would actually nicely fit into the official user documentation which is hosted [here](https://github.com/srsLTE/srsLTE_docs). Feel free to submit a PR if you wish.
Regarding the config I have one comment. We don't have a LimeSDR Mini to test with but I thought it only uses the wideband antenna connectors, so using LNAW as Rx antenna would be correct. Other may correct me please.

mikehibbett, 2019-12-26:
Thanks Andre, I'd be happy to once I get confirmation that others have had the same result. I will test your suggestion out to see if it helps. The LimeSDR Mini has just two SMA connectors, one TX one RX. I presume that setting you mention configures what matching circuit the RX SMA connector is routed through. I haven't got as far as testing a mobile phone yet, but will test your comment.

JamesHagerman, 2020-02-16:
For others landing here, I wanted to share a little more information about the details here because it took me a bit to figure out how all this works.

Tl;dr: Based on my limited testing, `srsLTE` correctly selects antenna ports on the LimeSDR Mini based on the value of `dl_earfcn` and it shouldn't need to have them configured explicitly.

The LMS7002M has 2 RF channels, each channel has 3x RX ports and 2x TX ports.

The LimeSDR Mini only uses one of these channels. It has two RF switches that allow software selection of which port is connected to the SMA connectors.

On the RX side, only `RX1_H` and `RX1_W` are exposed to the SMA RX port through the RF switch (and two different matching networks). The `RX1_L` port is not connected at all.

On the TX side, both `TX1_1` and `TX1_2` are exposed, (again, through two different matching networks).

With the default configuration of `dl_earfcn = 3400`, `srsLTE` correctly selects `LNAH` for RX and `BAND1` for TX. Those are SoapySDR names for `RX1_H` and `TX1_1` which can be verified by looking at the output of `SoapySDRUtil --probe`.

This is equivalent to an `srsLTE` config of `device_args = rxant=LNAH,txant=BAND1` which correctly aligns the RX/TX ports matching networks with the corresponding Uplink and Downlink frequencies used on Band 7.

@mikehibbett's config is setting `dl_earfcn = 6200` and is also actively selecting ports with `rxant=LNAH,txant=BAND2`. But the matching networks of those two RF paths do not align completely with Band 20 (6200) which has a Downlink of 796.00 and an Uplink of 837.00. TX is correct, but RX should (as @andrepuschmann said) be on `LNAW`.

Thank you so much for posting this @mikehibbett and thank you so much for the clarity @andrepuschmann! It helped me learn a lot about both the LimeSDR Mini and `srsLTE`!

andrepuschmann, 2020-02-17:
Thanks a ton @JamesHagerman for the feedback. With all the knowledge gathered in this issue I think it would be great if we could somehow join forces and extend the  [srsLTE docs page](https://docs.srslte.com/).

mikehibbett, 2020-02-17:
Thanks for that clarification @JamesHagerman , that clears it up for me. I'll be doing more work on this in the next month, and will be delighted to build up some additional documentation to contribute @andrepuschmann

shakisparki, 2020-03-03:
Hi @mikehibbett @andrepuschmann 

I have done the above, however I get :

> Device Soapy not found. Switching to auto mode
and then a UHD: code 11, no compatible RF frontend found.

I can find the LimeSDR Mini using both 
`LimeUtil --find` and
`SoapySDRUtil --probe`

I'm not sure what to try again.
I have installed `libsoapysdr-dev`

Any Ideas? 

Thanks

andrepuschmann, 2020-03-03:
Your CMake output looks right. Can you post your console output of `SoapySDRUtil --probe` and also srsLTE?

shakisparki, 2020-03-04:
Thanks.
Here are the outputs

### SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] Make connection: 'LimeSDR Mini [USB 2.0] 1D40E0CB99ACD3'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M register cache: Disabled

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=FT601
  hardware=LimeSDR-Mini
  boardSerialNumber=0x1d40e0cb99acd3
  firmwareVersion=5
  gatewareVersion=1.26
  hardwareVersion=2
  protocolVersion=1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: YES
  Sensors: clock_locked, lms7_temp
  Registers: BBIC
  Other Settings:
     * SAVE_CONFIG - Save LMS settings to file
       [key=SAVE_CONFIG, type=string]
     * LOAD_CONFIG - Load LMS settings from file
       [key=LOAD_CONFIG, type=string]
     * OVERSAMPLING - oversampling ratio (0 - auto)
       [key=OVERSAMPLING, type=int, options=(0, 1, 2, 4, 8, 16, 32)]
  GPIOs: MAIN

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32, CS12, CS16
  Native format: CS16 [full-scale=32767]
  Stream args:
     * Buffer Length - The buffer transfer size over the link.
       [key=bufferLength, units=samples, default=0, type=int]
     * Link Format - The format of the samples over the link.
       [key=linkFormat, default=CS16, type=string, options=(CS16, CS12)]
     * Skip Calibration - Skip automatic activation calibration.
       [key=skipCal, default=false, type=bool]
     * align phase - Attempt to align phase of Rx channels.
       [key=alignPhase, default=false, type=bool]
  Antennas: NONE, LNAH, LNAL_NC, LNAW, Auto
  Corrections: DC removal, DC offset, IQ balance
  Full gain range: [-12, 61] dB
    TIA gain range: [0, 12] dB
    LNA gain range: [0, 30] dB
    PGA gain range: [-12, 19] dB
  Full freq range: [0, 3800] MHz
    RF freq range: [30, 3800] MHz
    BB freq range: [-7.68, 7.68] MHz
  Tune args:
     * LO Offset - Tune the LO with an offset and compensate with the baseband CORDIC.
       [key=OFFSET, units=Hz, default=0.0, type=float, range=[-7.68e+06, 7.68e+06]]
[... 335 lines cut ...]
# metrics_csv_enable:   Write eNB metrics to CSV file.
# metrics_csv_filename: File path to use for CSV metrics.
# pregenerate_signals:  Pregenerate uplink signals after attach. Improves CPU performance.
# tx_amplitude:         Transmit amplitude factor (set 0-1 to reduce PAPR)
# link_failure_nof_err: Number of PUSCH failures after which a radio-link failure is triggered. 
#                       a link failure is when SNR<0 and CRC=KO
# max_prach_offset_us:  Maximum allowed RACH offset (in us)
# eea_pref_list:        Ordered preference list for the selection of encryption algorithm (EEA) (default: EEA0, EEA2, EEA1).
# eia_pref_list:        Ordered preference list for the selection of integrity algorithm (EIA) (default: EIA2, EIA1, EIA0).
#
#####################################################################
[expert]
#pusch_max_its        = 8 # These are half iterations
#pusch_8bit_decoder   = false
#nof_phy_threads      = 3
#metrics_period_secs  = 1
#metrics_csv_enable   = false
#metrics_csv_filename = /tmp/enb_metrics.csv
#pregenerate_signals  = false
#tx_amplitude         = 0.6
#link_failure_nof_err = 50
#rrc_inactivity_timer = 60000
#max_prach_offset_us  = 30
#eea_pref_list = EEA0, EEA2, EEA1
#eia_pref_list = EIA2, EIA1, EIA0

shakisparki, 2020-03-09:
Hello,
It appears I might have butchered the installation of srsLTE somewhere or messed up my installation environment at some point. When I run srsenb directly from the build directory after make, without make installing, it runs just fine. Although there appears to be some error on the Tx calibration side, and initiating the SCTP socket

Built in Release mode using commit d045213f on branch master.

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/shakir/.config/srslte/enb.conf...
Opening 1 RF devices with 1 RF channels...
Soapy has found device #0: addr=24607:1027, driver=lime, label=LimeSDR Mini [USB 2.0] 1D40E0CB99ACD3, media=USB 2.0, module=FT601, name=LimeSDR Mini, serial=1D40E0CB99ACD3, 
[INFO] Make connection: 'LimeSDR Mini [USB 2.0] 1D40E0CB99ACD3'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M register cache: Disabled
Setting up Rx stream with 1 channel(s)
Setting up Tx stream with 1 channel(s)

[INFO] RX LPF configured
[INFO] RX LPF configured
Set Rx bandwidth to 2.50 MHz
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 5.00 MHz
[INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
[INFO] TX LPF configured
Set Tx bandwidth to 50.00 MHz
Available device sensors: 
 - clock_locked
 - lms7_temp
Available sensors for Rx channel 0: 
 - lo_locked
Setting Rx antenna to LNAH
Setting Tx antenna to BAND2
**[ERROR] Tx Calibration: MCU error 4 (SXT tune failed)**
[INFO] Rx calibration finished
State of gain elements for Rx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -4.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PAD: 45.00 dB
 - IAMP: 0.00 dB
Rx antenna set to LNAH
Tx antenna set to BAND2

Warning burst preamble is not calibrated for device lime_mini. Set a value manually

Setting frequency: DL=796.0 Mhz, UL=837.0 MHz
Setting Sampling frequency 11.52 MHz
**Failed to initiate SCTP socket. Attempting reconnection in 10 seconds**

==== eNodeB started ===
Type <t> to view trace
[INFO] RX LPF configured
Set Rx bandwidth to 8.64 MHz
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 8.64 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 8.64 MHz
[INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
[INFO] TX LPF configured
Set Tx bandwidth to 50.00 MHz

Warning TX/RX time offset has not been calibrated for device lime_mini. Set a value manually

[INFO] Tx calibration finished
[INFO] Rx calibration finished

jin-github1002, 2020-04-16:
Great tutorial!!
I have followed the steps with LimeSDR USB. Now i can run srsue in a machine, run srsenb and srsepc on another machine. Each machine works with a LimeSDR USB.
 I can see "attach the network successfully" on the terminal of the ue side. But it releases from the network soon. May you all give some comments about this issue?
Thanks for your steps. it really helps a lot.

manili, 2020-05-11:
Thanks a lot for the tutorial.
However I still have problem to identify the network on my iPhone 7 Plus/iPhone 6s with the LimeSDR-USB.
The srsENB could identify Soapy without any problem and it works successfully, but I can not see the network in manual mode. Any help?

Kind regards,
Manili

Uthmanhere, 2020-05-30:
I am facing the same issue. Although everything seems fine in the log, nothing appears on radio as the image suggests.

[image]

### srsENB Logs
```
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/user/.config/srslte/enb.conf...

Built in Release mode using commit c892ae56b on branch HEAD.

/home/user/lime/srsLTE/srsenb/src/enb_cfg_parser.cc.883: Force DL EARFCN for cell PCI=1 to 6200

/home/user/lime/srsLTE/srsenb/src/enb_cfg_parser.cc.887: Force DL freq for cell PCI=1 to 1932.000000 MHz

Opening 1 channels in RF device=soapy with args=rxant=LNAH,txant=BAND1
11:10:40.605163 DEBUG:  xtrxllpciev0_discovery:264 [PCIE] pcie: Found `pcie:///dev/xtrx0`
Soapy has found device #0: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 3.0] 9081C05C30F33, media=USB 3.0, module=FX3, name=LimeSDR-USB, serial=0009081C05C30F33, 
Soapy has found device #1: addr=pcie:///dev/xtrx0, dev=pcie:///dev/xtrx0, driver=xtrx, label=XTRX: pcie:///dev/xtrx0 (10Gbit), media=PCIe, module=SoapyXTRX, name=XTRX, serial=, type=xtrx, 
Selecting Soapy device: 0
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9081C05C30F33'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M calibration values caching Disable
Setting up Rx stream with 1 channel(s)
Setting up Tx stream with 1 channel(s)
Available device sensors: 
 - clock_locked
 - lms7_temp
Available sensors for Rx channel 0: 
 - lo_locked
Setting Rx antenna to LNAH
Setting Tx antenna to BAND1
[INFO] Tx calibration finished
[INFO] Rx calibration finished
State of gain elements for Rx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -4.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PAD: 45.00 dB
 - IAMP: 0.00 dB
Rx antenna set to LNAH
Tx antenna set to BAND1

==== eNodeB started ===
Type <t> to view trace
Starting plot for worker_id=0
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Setting frequency: DL=1932,0 Mhz, UL=837,0 MHz for cc_idx=0
[WARNING] L

```
Hangs and need to send SIGTERM to terminate.

### srsEPC Logs
```
Built in Release mode using commit c892ae56b on branch HEAD.

---  Software Radio Systems EPC  ---

Reading configuration file /home/user/.config/srslte/epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:001, MNC:01, PLMN: 61712
S1 Setup Request - TAC 0, B-PLMN 0
S1 Setup Request - Paging DRX v128
Sending S1 Setup Response
SCTP Association Shutdown. Association: 12
Deleting eNB context. eNB Id: 0x19b
Releasing UEs context
No UEs to be released

```

joeldelosangeles, 2020-06-29:
Hi,

I'm getting this error:  Device soapy not found. Switching to auto mode.
That is why srsenb tries to use the UHD driver.  Will appreciate your help.

Thanks,
Joel Delos Angeles

root@srslte:/sys/bus/usb/drivers/usb# sudo srsenb
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srslte/enb.conf...

Built in Release mode using commit c892ae56b on branch master.

/root/srsLTE/srsenb/src/enb_cfg_parser.cc.883: Force DL EARFCN for cell PCI=1 to 1875

Opening 2 channels in RF device=soapy with args=rxant=LNAH,txant=BAND2
Device soapy not found. Switching to auto mode
[INFO] [UHD] linux; GNU C++ version 7.4.0; Boost_106501; UHD_3.14.1.1-release
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: rxant=LNAH,txant=BAND2,master_clock_rate=23.04e6
/root/srsLTE/lib/src/phy/rf/rf_uhd_imp.c.187: Error opening UHD: code 11

/root/srsLTE/lib/src/phy/rf/rf_imp.c.130: No compatible RF frontend found

Error initializing radio.
Setting frequency: DL=1872.5 Mhz, UL=1777.5 MHz for cc_idx=0
Error - buffer pool is empty
Error - buffer pool is empty

==== eNodeB started ===
Type <t> to view trace
srsLTE crashed... we could not save backtrace in './srsLTE.backtrace.crash'...
---  exiting  ---

andrepuschmann, 2020-06-29:
Can you verify that Soapy picks up the Lime ok. There is a bit of LimeSDR documentation in the RPi4 appnote. See [here](https://docs.srslte.com/en/latest/app_notes/source/pi4/source/index.html)

joeldelosangeles, 2020-06-29:
Thanks Andre.  I will work on the steps in the RPi-related page tomorrow.  For the meanwhile, what follows are additional information.  I did the LimeUtil --update for the FW upgrade.   Thanks

root@srslte:~# SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 90706024F2421'
[WARNING] Gateware version mismatch!
  Expected gateware version 2, revision 22
  But found version 2, revision 17
  Follow the FW and FPGA upgrade instructions:
  http://wiki.myriadrf.org/Lime_Suite#Flashing_images
  Or run update on the command line: LimeUtil --update

.............

root@srslte:~# LimeUtil --update
Connected to [LimeSDR-USB [USB 3.0] 90706024F2421]
Gateware version mismatch!......
.....
Existing firmware is same as update (4)
[  0%]    32/578900 Bytes in progress... (/usr/share/LimeSuite/images/20.01/Lime[  0%]    64/578900 Bytes in progress... (/usr/share/LimeSuite/images/20.01/Lime[  0%]    96/578900 Bytes in progress... (/usr/share/LimeSuite/images/20.01/Lime[  0%]  
......
(/usr/share/LimeSuite/images/20.01/Lim[100%] 578880/578900 Bytes in progress... (/usr/share/LimeSuite/images/20.01/Lim[100%] 578900/578900 Bytes in progress... (/usr/share/LimeSuite/images/20.01/Lim[100%] 578900/578900 Bytes in progress... (/usr/share/LimeSuite/images/20.01/Lim[100%] 578900/578900 Bytes programming: completed (/usr/share/LimeSuite/images/20.01/LimeSDR-USB_HW_1.4_r2.22.rbf)
Programming update complete!

................
root@srslte:~# SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 90706024F2421'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M register cache: Disabled

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=FX3
  hardware=LimeSDR-USB
  boardSerialNumber=0x90706024f2421
  firmwareVersion=4
  gatewareVersion=2.22
  hardwareVersion=4
  protocolVersion=1

joeldelosangeles, 2020-06-30:
Hi,

The SDR was detected and srsenb is running.  However, my spectrum analyzer just shows a CW signal at 1865 MHz (not an LTE signal).  Will appreciate if anyone can help me troubleshoot.  Logs are as follows:

Thanks,
Joel

root@oai-enb:~# srsenb
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srslte/enb.conf...

Built in Release mode using commit c892ae56b on branch master.

/root/srsLTE/srsenb/src/enb_cfg_parser.cc.883: Force DL EARFCN for cell PCI=1 to 1800

Opening 2 channels in RF device=soapy with args=clock=gpsdo,rxant=LNAH,txant=BAND2
Soapy has found device #0: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 3.0] 90706024F2421, media=USB 3.0, module=FX3, name=LimeSDR-USB, serial=00090706024F2421,
Selecting Soapy device: 0
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 90706024F2421'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M register cache: Disabled
Setting up Rx stream with 2 channel(s)
Setting up Tx stream with 2 channel(s)
[INFO] RX LPF configured
[INFO] RX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Available device sensors:
 - clock_locked
 - lms7_temp
Available sensors for Rx channel 0:
 - lo_locked
Available sensors for Rx channel 1:
 - lo_locked
Setting Rx antenna to LNAH
Setting Tx antenna to BAND2
[INFO] Tx calibration finished
[INFO] Tx calibration finished
[INFO] Rx calibration finished
[INFO] Rx calibration finished
State of gain elements for Rx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -4.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PAD: 45.00 dB
 - IAMP: 0.00 dB
Rx antenna set to LNAH
Tx antenna set to BAND2
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
[ERROR] SetFrequencyCGEN(368.64 MHz) failed

==== eNodeB started ===
Type <t> to view trace
[INFO] RX LPF configured
[ERROR] setSampleRate(Rx, 0, 11.52 MHz) Failed
setSampleRate Rx fail: SoapyLMS7::setSampleRate() failed
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 11.52 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 11.52 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Setting frequency: DL=1865.0 Mhz, UL=1770.0 MHz for cc_idx=0
LLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLL

joeldelosangeles, 2020-07-01:
Hi,

I hope someone is familiar with these errors: (I'm only getting a CW signal at 1875 MHz.  Thanks

[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
[ERROR] SetFrequencyCGEN(368.64 MHz) failed

==== eNodeB started ===
Type <t> to view trace
[INFO] RX LPF configured
[ERROR] setSampleRate(Rx, 0, 11.52 MHz) Failed
setSampleRate Rx fail: SoapyLMS7::setSampleRate() failed
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 11.52 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 11.52 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Setting frequency: DL=1875.0 Mhz, UL=1780.0 MHz for cc_idx=0

Thanks,
Joel

andrepuschmann, 2020-07-01:
Joel, the issue you are seeing comes from the LimeSDR's Soapy driver failing to set the sample rate. Perhaps you can try with a single channel. Also make sure you have the latest and matching versions of host code and FPGA running. If all that doesn't help I guess you should ask on the Lime list.

joeldelosangeles, 2020-07-02:
Hi Andre,

Thanks for your reply.  Are you referring to the firmware and the FPGA image?  If so, do you know where I can check that or do I have to flush the LimeSDR?

Thanks,
Joel

joeldelosangeles, 2020-07-03:
Hi,

Able to do this:

0.01/LimeSDR-USB_HW_1.4_r2.22.rbf)
Programming update complete!

after gateware mismatch.   Tried setting to 5 MHz  with single antenna.  The error when running srsenb is:

Rx antenna set to LNAH
Tx antenna set to BAND2
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
[ERROR] SetFrequencyCGEN(368.64 MHz) failed

==== eNodeB started ===
Type <t> to view trace
[INFO] RX LPF configured
[ERROR] setSampleRate(Rx, 0, 5.76 MHz) Failed
setSampleRate Rx fail: SoapyLMS7::setSampleRate() failed

Thanks,
Joel

rembulan, 2020-07-09:
Hi All,
I got this error messages 

Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 soapysdr-module-all : Depends: soapysdr0.6-module-all (= 0.6.1-2) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

joeldelosangeles, 2020-07-10:
Hi rembulan

I also encountered the same error.  The only dependency which I was able to correct is sudo apt-get install  liblimesuite18.10-1.  After that the error is the same when doing sudo apt install soapysdr-module-all -y.

Anyway I proceeded to the next steps and soapy-sdr seems to be working after that.

sudo add-apt-repository -y ppa:myriadrf/drivers
sudo apt-get update
sudo apt-get install limesuite liblimesuite-dev limesuite-udev limesuite-images -y
sudo apt-get install soapysdr soapysdr-module-lms7 -y

Thanks,
Joel

jin-github1002, 2020-07-24:
@joeldelosangeles hi, I used LimeSDR USB as srsue and srsenb. I use dl_earfcn=1934, tx gain =60, rx gain=40, device name=soapy.

Theo-31, 2020-07-28:
Hi,

At least, did someone achieve to run fully srsLTE with LimeSDR mini ? On my side I can get my UE attached but I can't go further because of TX issue.

Dastgsm, 2020-08-05:
someone says on forum that srslte19.12 is only which run with limesdrmini

alexballas, 2020-08-25:
> @joeldelosangeles, How did you set the reference clock to 30.72 MHz from 40 MHz?

@joeldelosangeles uses a LimeSDR-USB, not a LimeSDR mini from what I see. It would be good to focus just on the mini for this thread.

andrepuschmann, 2020-10-27:
Several changes for the LimeSDR have been merged into srsLTE and LimeSuite/SoapyLMS over the recent past. Please try again with the newest versions and reopen or create new issues if needed. Thanks

closed 2020-10-27 by andrepuschmann

jin-github1002, 2020-10-27:
Hi @andrepuschmann which version of srsLTE includes the changes for the LimeSDR? I am using 19.12. I hope to confirm if I need to upgrade to version 20.xx. Thank you so much!

alexballas, 2020-10-27:
I tried with the latest (**master**) versions of both **LimeSuite** and **srsLTE**.

srsenb output:
```
$ sudo ./build/srsenb/src/srsenb 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /etc/srslte/enb.conf...

Built in Release mode using commit 0ec49a7b1 on branch master.

/home/alex/test/srsLTE/srsenb/src/enb_cfg_parser.cc.883: Force DL EARFCN for cell PCI=1 to 1875

Opening 1 channels in RF device=soapy with args=rxant=LNAH,txant=BAND2
Soapy has found device #0: addr=24607:1027, driver=lime, label=LimeSDR Mini [USB 3.0] 1D588F01B6A2F2, media=USB 3.0, module=FT601, name=LimeSDR Mini, serial=1D588F01B6A2F2, 
Selecting Soapy device: 0
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D588F01B6A2F2'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M register cache: Disabled
Setting up Rx stream with 1 channel(s)
Setting up Tx stream with 1 channel(s)
[INFO] RX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Available device sensors: 
 - clock_locked
 - lms7_temp
Available sensors for Rx channel 0: 
 - lo_locked
Setting Rx antenna to LNAH
Setting Tx antenna to BAND2
[INFO] Tx calibration finished
[INFO] Rx calibration finished
State of gain elements for Rx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -4.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PAD: 45.00 dB
 - IAMP: 0.00 dB
Rx antenna set to LNAH
Tx antenna set to BAND2

==== eNodeB started ===
Type <t> to view trace
[INFO] RX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 11.52 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Setting frequency: DL=1872.5 Mhz, UL=1777.5 MHz for cc_idx=0
```
srsepc output:

```
$ sudo ./build/srsepc/src/srsepc

Built in Release mode using commit 0ec49a7b1 on branch master.

---  Software Radio Systems EPC  ---

Reading configuration file /etc/srslte/epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SPGW GTP-U Initialized.
SPGW S11 Initialized.
[... 161 lines cut ...]
    PAD gain range: [0, 52] dB
    IAMP gain range: [-12, 12] dB
  Full freq range: [0, 3800] MHz
    RF freq range: [30, 3800] MHz
    BB freq range: [-7.68, 7.68] MHz
  Tune args:
     * LO Offset - Tune the LO with an offset and compensate with the baseband CORDIC.
       [key=OFFSET, units=Hz, default=0.0, type=float, range=[-7.68e+06, 7.68e+06]]
     * BB - Specify a specific value for this component or IGNORE to skip tuning it.
       [key=BB, units=Hz, default=DEFAULT, type=float, range=[-7.68e+06, 7.68e+06], options=(DEFAULT, IGNORE)]
  Sample rates: [0.1, 30.72] MSps
  Filter bandwidths: [5, 40], [50, 130] MHz
  Sensors: lo_locked
  Other Settings:
     * TSP_CONST - Digital DC test signal level in LMS7002M TSP chain.
       [key=TSP_CONST, default=16383, type=int, range=[0, 32767]]
     * CALIBRATE -  DC/IQ calibration bandwidth
       [key=CALIBRATE, type=float, range=[2.5e+06, 1.2e+08]]
     * ENABLE_GFIR_LPF - LPF bandwidth (must be set after sample rate)
       [key=ENABLE_GFIR_LPF, type=float]
     * TSG_NCO - Enable NCO test signal
       [key=TSG_NCO, default=4, type=int, options=(-1, 4, 8)]
```

Unfortunately it just hangs there. I don't see any outstanding errors when running srsenb.

jin-github1002, 2020-10-27:
hi @alexballas what do you mean "it just hangs“？
I saw that you used dl_earfcn = 1875, prb =50, for LimeSDR mini. I tried many dl_earfcn for LimeSDRUSB. I am confused how to find the frequency working stable on LimeSDR USB.
My problems is : Only the dl_earfch=1934 ,gain=60  (showed in the tutorial of srsLTE) are stable used on LimeSDR USB when srsLTE runs on desk PC. But when I run srsLTE on Jetson TX2 ( arm processor), above frequency and gain do not work stable. I got  stuck here.

alexballas, 2020-10-28:
Actually I need to run some more tests. I'll update this thread asap with the results.

vetrivelmnv13, 2020-11-06:
in limesdrmini i am giving sample rate 5Mhz. based on the sample rate i have to change the buffer size. i don't have an idea. can anyone give some suggestion?

hmartinezf, 2020-11-10:
Hola!!

I got my first SDR the LimeSDR-Mini v1.3 and I didn't have any issue at all, even when moving to Band 20. If you use cheap antenna for Rx 800 MHz you may consider 1m thin coax cable plus 50 ohms load for good reception.

What I don't understand is why PLMN configurations are not working. I tried a different MNC MCC but it seems that only 001-01 is accepted by SRSLTE. Actually I can see on my phone **Software Radio Systems LTE (TESTING)** TESTING is my SPN service provider name in USIM card. I need help, actually the _COTS UE Application Note_ describes how to do that but...

mbayesamb, 2022-07-08:
hello can we set up a 5G or 4G network using a single limesdr please