srsran/srsRAN_4G #438: NO RECEPTION OF LTE SIGNAL ON COMMERCIAL MOBILE

asp2k5, 2019-12-27:
Hi everyone.I am using **USRP B210** for srsLTE, how ever both EPC and eNB seems to be working fine (constant red LED on Tx+ constant Green on Rx of RF0) still am unable to detect any signal on my mobile. could anyone pls suggest any solution.n_prb is set as 50.
pls help out.thanks.

---  Software Radio Systems EPC  ---
Reading configuration file /home/esb3283/.config/srslte/epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf404, MNC: 0xff64
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:404, MNC:64, PLMN: 324678
S1 Setup Request - TAC 7, B-PLMN 0
S1 Setup Request - Paging DRX 2
Sending S1 Setup Response

-----------------------------------------------------------------------------------------------
Built in Release mode using commit 0e89fa9f on branch master.
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/esb3283/.config/srslte/enb.conf...
Opening 1 RF devices with 1 RF channels...
[INFO] [UHD] linux; GNU C++ version 8.3.0; Boost_106700; UHD_3.14.1.0-release
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP with args: type=b200,master_clock_rate=23.04e6
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 2.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Setting frequency: DL=2145.0 Mhz, UL=1745.0 MHz
Setting Sampling frequency 11.52 MHz

**==== eNodeB started ===**
Type <t> to view trace

Hansimov, 2020-01-03:
Notice that you are using USB 2.0, this may be something related to the data transfer rate.
Have you tried USB 3.0?

asp2k5, 2020-01-03:
No, am using USB 3.0 it port no 2 not the usb2. My system has 3 usb3.0 port. So thats not the issue.

andrepuschmann, 2020-01-07:
But UHD uses the port in USB2.0 mode, see output.

asp2k5, 2020-01-28:
thankx alot I think this is the issue. My UHD automatically select USB2 even through its connected to USB3 port. on numerous plug in out it select usb3. is there any issue?? how could I resolve it...??
am using usb3 cable only ...

andrepuschmann, 2020-01-28:
I don't know, but it's not an srsLTE issue. Closing therefore.

closed 2020-01-28 by andrepuschmann