srsran/srsRAN_4G #443: Please Help ? [ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules0.7/libLMS7Support.so)

muratoznn, 2020-01-20:
How Can I fix This Error ? Please Help ?  I am using Kali Linux And Limesdr Mini

root@kali:~/Downloads/srsLTE-master/srsenb# sudo srsenb /root/Downloads/srsLTE-master/srsenb/enb.conf

Built in Release mode using 19.12.0.

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/Downloads/srsLTE-master/srsenb/enb.conf...
Opening 1 RF devices with 1 RF channels...
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200104; Boost_106700; UHD_3.15.0.0-2
[INFO] [LOGGING] Fastpath logging disabled at runtime.
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules0.7/libLMS7Support.so)
  duplicate entry for lime (/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libLMS7Support.so)
Assertion 'close_nointr(fd) != -EBADF' failed at src/basic/fd-util.c:71, function safe_close(). Aborting.
srsLTE crashed... backtrace saved in './srsLTE.backtrace.crash'...
---  exiting  ---
This İs My Srscrach Backtrace File ::

--- command='srsenb /root/Downloads/srsLTE-master/srsenb/enb.conf' version=19.12.0 signal=6 date='20/01/2020 18:44:05' ---
	srsenb(+0x104d6e) [0x556827ab0d6e]
	/lib/x86_64-linux-gnu/libc.so.6(+0x3a100) [0x7fe3f882e100]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0x141) [0x7fe3f882e081]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x121) [0x7fe3f8819535]
	/lib/x86_64-linux-gnu/libudev.so.1(+0x580a) [0x7fe3f742980a]
	/lib/x86_64-linux-gnu/libudev.so.1(+0x66f2) [0x7fe3f742a6f2]
	/lib/x86_64-linux-gnu/libudev.so.1(+0xe877) [0x7fe3f7432877]
	/lib/x86_64-linux-gnu/libudev.so.1(+0x125fb) [0x7fe3f74365fb]
	/lib/x86_64-linux-gnu/libudev.so.1(+0x12d34) [0x7fe3f7436d34]
	/lib/x86_64-linux-gnu/libudev.so.1(+0x19a1b) [0x7fe3f743da1b]
	/lib/x86_64-linux-gnu/libudev.so.1(+0x19deb) [0x7fe3f743ddeb]
	/lib/x86_64-linux-gnu/libudev.so.1(udev_enumerate_scan_devices+0x277) [0x7fe3f7440627]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0x11126) [0x7fe3f790d126]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xfaab) [0x7fe3f790baab]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_init+0x19f) [0x7fe3f7902a3f]
	/lib/x86_64-linux-gnu/libairspy.so.0(airspy_open+0x49) [0x7fe3ec324779]
	/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libairspySupport.so(+0x8583) [0x7fe3f407f583]
	/lib/x86_64-linux-gnu/libSoapySDR.so.0.7(_ZNSt17_Function_handlerIFSt10unique_ptrINSt13__future_base12_Result_baseENS2_8_DeleterEEvENS1_12_Task_setterIS0_INS1_7_ResultISt6vectorISt3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESF_St4lessISF_ESaISt4pairIKSF_SF_EEESaISM_EEEES3_ENSt6thread8_InvokerISt5tupleIJPFSO_RKSM_ESM_EEEESO_EEE9_M_invokeERKSt9_Any_data+0x35) [0x7fe3f7b41645]
	/lib/x86_64-linux-gnu/libSoapySDR.so.0.7(+0x2c849) [0x7fe3f7b40849]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x106df) [0x7fe3f93586df]
	/lib/x86_64-linux-gnu/libSoapySDR.so.0.7(_ZNSt6thread11_State_implINS_8_InvokerISt5tupleIJZNSt13__future_base17_Async_state_implINS1_IS2_IJPFSt6vectorISt3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESC_St4lessISC_ESaISt4pairIKSC_SC_EEESaISJ_EERKSJ_ESJ_EEEESL_EC4EOSR_EUlvE_EEEEE6_M_runEv+0xf9) [0x7fe3f7b40f69]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xcfb20) [0x7fe3f8be2b20]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x8fb7) [0x7fe3f9350fb7]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7fe3f88ee2cf]

andrepuschmann, 2020-04-09:
Not sure what this is, perhaps a mismatch in the library version or similar.

closed 2020-04-09 by andrepuschmann

ccpp, 2021-12-05:
Hi,
I also got this issue. In my case it's a 3rd-party library, also based on libusb.
Could you resolve this?
Thank you