srsran/srsRAN_4G #444: Radio-Link Failure <how to debug HW setup>

Shik-shik, 2020-01-22:
Before filing an issue, search for solutions here:
- srsLTE users mailing list (http://www.softwareradiosystems.com/mailman/listinfo/srslte-users)

## Issue Description ##
- set DEFAULT configuration;
- adjust arfcn value to 3300 (no interference detected in this range);
- running srsue result to RLF situation in good signal condition and high SNR. 

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

CPU: I7-3630QM;
Linux: Ubuntu 18.04;
RF: bladerf x40 (both enb and ue)
driver/library:
$ bladeRF-cli -i --lib-version
2.2.0-2018.12-rc3-2-ppabionic

## Expected Behavior ##
successfull RRC connection setup following with ATTACH procedure

## Actual Behaviour ##
Failed RRC due to timer expire:
14:57:35.735166 [RRC ] [W] Proc "Connection Request" - Timer T300 expired: ConnectionRequest timed out

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
kindly ask you to test with bladerf x40 on default values. 
trying different TX/RX gain may change RSRP and SNR value but Radio-Link failure remain.

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
logs from UE and ENB in attachment to this case. 

[80_after_calibration.zip](https://github.com/srsLTE/srsLTE/files/4098342/80_after_calibration.zip)

Shik-shik, 2020-01-30:
Default config  files for bladeRF x40 result to RLF. 
after community request, other configuration was given for test: 

enb.conf
================
dl_earfcn = 3300
n_prb = 15
max_prach_offset_us  = 300
================

sib.conf
================
zero_correlation_zone_config = 0

================

ue.conf
================
dl_earfcn = 3300
time_adv_nsamples  = 40
================

still it is not best option (RRC connected after several attempts). But it gives sufficient information for further improvement/debug.

Thank you for community members (many thanks to justin.tallon@softwareradiosystems.com).

Issue can be closed. After stable RRC from first attempt, config info will be updated.

closed 2020-01-30 by Shik-shik

jin-github1002, 2020-05-16:
@Shik-shik thank you for your issue. I set the same properties when i use limesdr as the rf end at ue and enb. it is successful for ue to attach lte network. ue and enb are connected for at most 10minutes. It is much longer than before when i use the default .conf files(dl_earfcn=3400,prb=50).
But i dont know why the max online time for ue is no more than 10 minutes. Could you give some comments about how to find the exact properties for the ue and enb to make them always connected? Or Do you have any information for further improvement and debug it?
Thank you so much for your help and your posted issue.