srsran/srsRAN_4G #448: Internal Compiler Error when Building lib/src/asn1/CMakeFiles/rrc_asn1.dir/rrc_asn1.cc.o during Make

shakisparki, 2020-02-04:
### **Procedure for installation**
`git clone https://github.com/srsLTE/srsLTE.git
cd srsLTE
mkdir build
cd build
cmake ../
make
make test`

### **During `Make`** 
[  1%] Built target asn1_utils
[  2%] Built target srslte_asn1
[  2%] Building CXX object lib/src/asn1/CMakeFiles/rrc_asn1.dir/rrc_asn1.cc.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
lib/src/asn1/CMakeFiles/rrc_asn1.dir/build.make:62: recipe for target 'lib/src/asn1/CMakeFiles/rrc_asn1.dir/rrc_asn1.cc.o' failed
make[2]: *** [lib/src/asn1/CMakeFiles/rrc_asn1.dir/rrc_asn1.cc.o] Error 4
CMakeFiles/Makefile2:1180: recipe for target 'lib/src/asn1/CMakeFiles/rrc_asn1.dir/all' failed
make[1]: *** [lib/src/asn1/CMakeFiles/rrc_asn1.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

**An error is thrown. Can't seem to get around it.
System is running Ubuntu 16.04
SoapySDR is Installed**

### **Below is Cmake Output**
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_SYSTEM: Linux-2.6.32-042stab140.4
-- CMAKE_SYSTEM_PROCESSOR: x86_64
-- CMAKE_CXX_COMPILER: /usr/bin/c++
-- Build type not specified: defaulting to Release.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.4
-- FFTW3F LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.a
-- FFTW3F INCLUDE DIRS: /usr/include
-- Found fftw3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- FFT_LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- Checking for module 'polarssl'
--   No package 'polarssl' found
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND
-- POLARSSL INCLUDE DIRS: POLARSSL_INCLUDE_DIRS-NOTFOUND
-- Could NOT find POLARSSL (missing:  POLARSSL_LIBRARIES POLARSSL_INCLUDE_DIRS)
-- Checking for module 'mbedtls'
--   No package 'mbedtls' found
-- MBEDTLS LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: MBEDTLS_STATIC_LIBRARIES-NOTFOUND
-- MBEDTLS INCLUDE DIRS: /usr/include
-- Found MBEDTLS: /usr/lib/x86_64-linux-gnu/libmbedcrypto.so
-- Checking for module 'libpcsclite'
--   No package 'libpcsclite' found
-- Could NOT find PCSCLITE (missing:  PCSCLITE_LIBRARY PCSCLITE_INCLUDE_DIR)
-- PCSC LIBRARIES: PCSCLITE_LIBRARY-NOTFOUND
-- PCSC INCLUDE DIRS: PCSCLITE_INCLUDE_DIR-NOTFOUND
-- UHD LIBRARIES UHD_LIBRARIES-NOTFOUND
-- UHD INCLUDE DIRS UHD_INCLUDE_DIRS-NOTFOUND
-- Could NOT find UHD (missing:  UHD_LIBRARIES UHD_INCLUDE_DIRS)
-- Checking for module 'libbladeRF'
--   No package 'libbladeRF' found
-- libbladeRF not found.
-- FINDING SOAPY.
-- Checking for module 'SoapySDR'
--   Found SoapySDR, version 0.8.0
-- Found libSOAPYSDR: /usr/local/include/SoapySDR, /usr/local/lib/libSoapySDR.so
-- FINDING ZEROMQ.
-- Checking for module 'ZeroMQ'
--   No package 'ZeroMQ' found
-- libZEROMQ not found.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   program_options
-- SRSGUI LIBRARIES SRSGUI_LIBRARIES-NOTFOUND
-- SRSGUI INCLUDE DIRS SRSGUI_INCLUDE_DIRS-NOTFOUND
-- Could NOT find SRSGUI (missing:  SRSGUI_LIBRARIES SRSGUI_INCLUDE_DIRS)
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE - Success
-- Performing Test HAVE_SSE
-- Performing Test HAVE_SSE - Success
-- SSE4.1 is enabled - target CPU must support it
-- Performing Test HAVE_AVX
-- Performing Test HAVE_AVX - Success
-- AVX is enabled - target CPU must support it
-- Performing Test HAVE_AVX2
-- Performing Test HAVE_AVX2 - Failed
-- Performing Test HAVE_FMA
-- Performing Test HAVE_FMA - Failed
-- Performing Test HAVE_AVX512
-- Performing Test HAVE_AVX512 - Failed
-- Performing Test HAVE_VISIBILITY_HIDDEN_CXX
-- Performing Test HAVE_VISIBILITY_HIDDEN_CXX - Success
-- Performing Test HAVE_ERROR_INCOMPATIBLE
-- Performing Test HAVE_ERROR_INCOMPATIBLE - Success
-- Using default glibc header for IPv6 support.
-- SSE4.1 is enabled - target CPU must support it
-- AVX is enabled - target CPU must support it
-- Performing Test HAVE_VISIBILITY_HIDDEN_C
-- Performing Test HAVE_VISIBILITY_HIDDEN_C - Success
-- CMAKE_C_FLAGS is  -Wno-unused-but-set-variable -Werror=incompatible-pointer-types -Wall -Wno-comment -Wno-write-strings -Winline -Wno-unused-result -Wformat -Wmissing-field-initializers -Wtype-limits -std=c99 -D_GNU_SOURCE -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE -march=native -mfpmath=sse -mavx -DLV_HAVE_AVX -DLV_HAVE_SSE -Ofast -funroll-loops -fvisibility=hidden -Werror
-- CMAKE_CXX_FLAGS is  -Wall -Wno-comment -Wno-reorder -Wno-unused-variable -Wtype-limits -std=c++11 -Wno-unused-but-set-variable -march=native -mfpmath=sse -mavx -DLV_HAVE_AVX -DLV_HAVE_SSE -fvisibility=hidden -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE -DUSE_GLIBC_IPV6 -Werror
-- Using install prefix: /usr/local
-- Building for version: 19.12.0
--    examples will be installed.
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- Found SCTP: /usr/lib/x86_64-linux-gnu/libsctp.so
-- No post-build command defined
-- Building with srsUE
-- No post-build-UE command defined
-- No post-build command defined
-- Building with srsENB
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/x86_64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/x86_64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-ENB command defined
-- Building with srsEPC
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/x86_64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/x86_64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-EPC command defined
-- Configuring done
-- Generating done
-- Build files have been written to: /home/cabox/workspace/srsLTE/build

frankist, 2020-02-04:
can you share your computer memory specs?

shakisparki, 2020-02-05:
I'm running this on a basic container with 
2GB of Disc Storage
256MB RAM (+ 512 MB swap)

frankist, 2020-02-05:
ok, you basically run out of memory. Compiling rrc_asn1.cc is very memory intensive. The only solution is to break it down into smaller .cc files and linking after

shakisparki, 2020-02-05:
OK thanks a lot :) I never would have thought it was a memory issue. 
But yeah the memory is pretty shite.

closed 2020-02-05 by shakisparki