srsran/srsRAN_4G #449: Adapt to supported CPU extensions dynamically

FFY00, 2020-02-04:
Currently, support for CPU extensions needs to be compiled in. This is not great when distributing the binaries.

srsLTE is really performance sensible so, right now, to have the best experience we should compile it natively. This is obviously not optimal, users should be able to just install it out of their package managers and be able to use it.

This problem is not hard to resolve. What we should do is compile the binary to the highest possible architecture and figure ut the supported CPU extensions at runtime.

For this, I suggest using [libvolk](https://www.libvolk.org/), which is used and developed by the GNURadio project.

frankist, 2020-02-05:
Hey, Filipe. libvolk license is not suitable for us.

FFY00, 2020-02-05:
Hello, could you elaborate?

This project is licensed under AGPLv3, libvolk is licensed under GPLv3. Although GPLv3 is not strictly compatible with AGPLv3, it does allow for the code to be linked (or combined, ex. static libs) against AGPLv3.

From GPLv3:

> 13. Use with the GNU Affero General Public License.
> 
> Notwithstanding any other provision of this License, you have permission to link or combine any covered work with a work licensed under version 3 of the GNU Affero General Public License into a single combined work, and to convey the resulting work. The terms of this License will continue to apply to the part which is the covered work, but the special requirements of the GNU Affero General Public License, section 13, concerning interaction through a network will apply to the combination as such.

Also see https://www.gnu.org/licenses/license-list.en.html#AGPLv3.0

Is there any particular reason?

UHD and libzmq both are licensed under GPLv3 so I don't really understand.

ismagom, 2020-02-06:
>    1. Use with the GNU Affero General Public License.

Hi Filipe,

I think the reason is more technical. We started using it in our early days
but we discarted it soon. We don't want the run-time dynamic kernel
selection because we have very strict timing requirements and we need
deterministic performance on a given CPU.

Also, many of our kernels are customized for some special operations. In
addition, there are some blocks that are entirely written in SIMD, like
decoder, interleaver, etc. There is no volk alternative for that.

Most of our users compile our source code. It wasn't until recently that we
had the need to distribute platform-dependant binaries. We'll provide a
solution to that soon, but it will not be based on volk.

Regards and thanks a lot for your feedback

andrepuschmann, 2020-02-06:
Distribution packagers might also check out [this](https://github.com/srsLTE/srsLTE/blob/master/lib/src/common/arch_select.cc). The idea here is to build arch specific binaries and call them `srsue_avx2` for example. Then load the generic `srsue` binary which detects the ISA and executes the right arch. There might be some tuning needed for the detection but in principle, this approach should work.

FFY00, 2020-02-06:
> Hi Filipe,
>
> I think the reason is more technical. We started using it in our early days
> but we discarted it soon. We don't want the run-time dynamic kernel
> selection because we have very strict [...]

Okay, so it isn't a licensing issue :)

libvolk was just a suggestion, I find it better to come with possible solutions when reporting issues or asking for new features.

> Most of our users compile our source code. It wasn't until recently that we
> had the need to distribute platform-dependant binaries. We'll provide a
> solution to that soon, but it will not be based on volk.
>
> Regards and thanks a lot for your feedback

That is reasonable, and I am glad it is in your roadmap. Do you have any idea when that might come out? Just to have a rough idea.

> Distribution packagers might also check out this. The idea here is to build arch specific binaries and call them srsue_avx2 for example. Then load the generic srsue binary which detects the ISA and [...]

Yes, that is another approach. Although a little wasteful, easier to implement. I think it's a good first step.

For now, I will be providing two packages on Archlinux, one with no CPU extensions and one with AVX2.

andrepuschmann, 2020-04-29:
Closing this as I think we'll have a good solution with the `arch_select` tool and the CMake options to select individual ISAs for packagers to use. Will be available as of srsLTE 20.04

closed 2020-04-29 by andrepuschmann

FFY00, 2020-04-29:
@andrepuschmann there is another interesting alternative that we have not discussed. GCC has a `target_clones` function attribute that allows us to define specific function implementations for different CPU extensions.

https://gcc.gnu.org/onlinedocs/gcc/Common-Function-Attributes.html#index-target_005fclones-function-attribute

andrepuschmann, 2020-04-30:
cool, didn't know about that flag.