srsran/srsRAN_4G #456: srsenb <-> srsepc: user contexts go missing

mrlnc, 2020-02-12:
## Issue Description ##

Hi everybody,

there seem to be issues with user and connection management between eNB and EPC, and I can't clearly isolate them. PCAPs and logs are attached, hope it helps to understand what S1AP is doing here.

What I observed:
1. eNB receives `k_enb = 0x0` from MME, resulting in RRC security mode command reject by UE
2. MME sends InitialContextSetupRequest but fails to look up `MME-UE-S1AP-ID`, resulting in malformed S1AP message
3. if eNB receives no answer to the InitialUEMessage, it sends a malformed UEContextReleaseRequest

('malformed' in Wireshark, but judging from the logs, it correlates with failing `find_nas_ctx_from_mme_ue_s1ap_id` and alike)

## Setup Details ##

local machine:
* srsepc on d045213fb9cbf98c83c06d7c17197a9dcbfddacf
* srsenb on d045213fb9cbf98c83c06d7c17197a9dcbfddacf
(seems to be the same with testers)

UE: Xiaomi Pocophone F1

## Steps to reproduce the problem ##

First attach is a regular attach. Then I was enabling and disabling data to trigger Service Requests. The Extended Service Request is an outgoing call.

I'm using a call (with Extended Service Request) to trigger the behavior here, but it seems to happen on other occasions, too.

## Expected Behavior ##

When the UE starts an attach request and finishes AKA, the correct key is handed to the eNB.

## Actual Behaviour ##

Some user lookup fails, and `k_enb = 0x0` is send to the eNB. Then, RRC procedures fail due to the wrong key.

MME NAS derives the key:
```
19:16:47.957119 [NAS ] [I] Generating KeNB with UL NAS COUNT: 0
19:16:47.957172 [NAS ] [I] Key eNodeB (k_enb)
             0000: 42 25 6c c5 67 14 a2 b2 e2 a8 a7 86 66 44 3d 70 
             0010: 89 77 4f ec d5 1b 2a d0 30 da aa 33 9a 11 cc f1 
```
Then, something goes wrong. `k_enb` and `imsi` for that user are zero:

```
19:16:47.997563 [HSS ] [I] Found User 001020123456781
19:16:47.997570 [NAS ] [D] Getting subscription information -- QCI 7
19:16:47.997580 [MME GTPC] [I] Sending Create Session Request.
19:16:47.997603 [MME GTPC] [I] Next MME control TEID: 3
19:16:47.997613 [MME GTPC] [I] Allocated MME control TEID: 2
19:16:47.997630 [MME GTPC] [W] Create Session Request being called for an UE with an active GTP-C connection.
19:16:47.997641 [MME GTPC] [W] Deleting previous GTP-C connection.
19:16:47.997646 [MME GTPC] [D] Sending S-11 GTP-C PDU
19:16:47.997668 [MME GTPC] [D] MME S11 Sent 1360 Bytes.
19:16:47.997674 [S1AP] [D] Waiting for S1-MME or S11 Message
19:16:47.997680 [SPGW] [D] Message received at SPGW: S11 Message
19:16:47.997722 [SPGW GTPC] [D] Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
19:16:47.997798 [SPGW GTPC] [I] SPGW Received Create Session Request
19:16:47.997895 [SPGW GTPC] [I] Creating new GTP-C context
19:16:47.997914 [SPGW GTPC] [I] SPGW: get_new_ue_ipv4 pool ip addr 172.16.0.3
19:16:47.997948 [SPGW GTPC] [I] Sending Create Session Response
19:16:47.998001 [SPGW GTPC] [D] SPGW Sending S11 PDU! N_Bytes: 1360
19:16:47.998029 [SPGW GTPC] [D] SPGW S11 Sent 1360 Bytes.
19:16:47.998036 [MME GTPC] [D] Received S11 message
19:16:47.998044 [MME GTPC] [D] MME Received GTP-C PDU. Message type GTPC_MSG_TYPE_CREATE_SESSION_RESPONSE
19:16:47.998056 [MME GTPC] [I] Received Create Session Response
19:16:47.998126 [MME GTPC] [I] MME GTPC Ctrl TEID 2, IMSI 1020123456781
19:16:47.998148 [MME GTPC] [I] Create Session Response -- SPGW control TEID 2
19:16:47.998170 [MME GTPC] [I] Create Session Response -- SPGW S1-U Address: 127.0.1.100
19:16:47.998328 [S1AP] [I] Preparing to send Initial Context Setup request
19:16:47.998343 [S1AP] [I] Initial Context Setup Request -- Key eNB (k_enb)
             0000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
             0010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
19:16:47.998384 [S1AP] [I] Adding attach accept to Initial Context Setup Request
19:16:47.998395 [NAS ] [I] Packing Attach Accept
19:16:47.998409 [NAS ] [I] Attach Accept -- MCC 0xf001, MNC 0xff02
19:16:47.998423 [S1AP] [D] Allocated M-TMSI 0x29a1c71a to IMSI 000000000000000,
```

[logs_pcaps.zip](https://github.com/srsLTE/srsLTE/files/4194558/logs_pcaps.zip)

alvasMan, 2020-02-13:
Thanks a lot for the detailed report Merlin. I'll have a look into this.

alvasMan, 2020-06-22:
Hi Merlin, sorry for the delayed reply, but you would you mind checking if the issue is still there in the latest version of the code?

There were some changes made to the way that srsEPC cleans up the UE's context that should help with this.

mrlnc, 2020-06-24:
Yes it seems mostly resolved on c892ae56be5302eaee5ca00e270efc7a5ce6fbb2

There's just one exception I can spot now: connect to srsEPC and throw in an unexpected Identity Response, then srsEPC gets some hiccups.

[image]

I think the cause is that the S1AP IDs are both set to zero:
```
17:55:16.600421 [S1AP] [D] Sending message to eNB with SCTP association 132714. MME UE S1AP ID 0, eNB UE S1AP ID 0
```

Attaching the full logs & PCAPs: [Archive.zip](https://github.com/srsLTE/srsLTE/files/4827196/Archive.zip)

There's one attach procedure before (just to make sure everything works), and then adding the unsolicited `Identity Response`. Seems that timezones are messed up on my system, just look out for the `xx:55:16` timestamp.

Ignore all those errors after `57:40.116178`. It's just because I attach GDB to flush the buffers… srsEPC is crashing there and doesn't write the log completely on exit.

[image]

andrepuschmann, 2020-10-27:
There have been a few changes to srsEPC recently I believe so I am unsure if the issue still exists. I am closing the ticket but feel free to reopen if you still see the issue. Patches are more than welcome if so ;-)

closed 2020-10-27 by andrepuschmann