srsran/srsRAN_4G #457: UE PDCP/RRC: Packet after SecurityModeCommand is not decrypted

mrlnc, 2020-02-16:
## Setup Details ##
Depends on eNodeB. Some commercial eNodeBs combine RRC Security Mode Command and RRC Reconfiguration messages in one RLC frame.

## Expected Behavior ##

After RRC Security Mode Command, PDCP is re-configured to use encryption. All following packets use security.

## Actual Behaviour ##

Some eNodeBs send RRC Reconfiguration immediately after RRC Security Mode Command (without first waiting for SMC complete), so both are in the same RLC frame.

The UE PDCP receives Security Mode Command and pushes it to RRC. The following packet is also pushed to RRC, without waiting for the encryption to be activated. Thus, RRC cannot decode the RRC Reconfiguration because it has not been decrypted by PDCP.

## Steps to reproduce the problem ##
Connect to an eNodeB with that behavior.

## Additional Information ##
I'm using this patch as a quick workaround: https://github.com/mrlnc/srsLTE/commit/46527824cc9d64edb13b24f6aae5288df8ffa4f4

It will save every incoming PDCP packet. If RRC receives a Security Mode Command, it will command PDCP to decrypt the last saved PDCP packet and re-submit it to RRC.

There are some obvious limitations; it works only if there is exactly one PDCP packet after the one that contains RRC Security Mode Command. And we accept that every now and then, a malformed packet (due to missing decryption) reaches RRC. Also, it could lead to duplicate packets reaching RRC if Null-Security is used.

mrlnc, 2020-02-16:
RLC frame where Security Mode Command directly followed by encrypted PDCP packet.

[rrc_smc.pcap.zip](https://github.com/srsLTE/srsLTE/files/4209785/rrc_smc.pcap.zip)

```
14:16:54.860480 [RLC ] [I] SRB1 Rx data PDU SN=4 (142 B)
             0000: 04 32 02 20 e5 90 ce 97 05 e4 f6 7a e1 27 9b 6b 
             0010: 0e 60 86 69 c1 31 28 01 31 bc 2d 85 cc d5 cf 0e
```
RLC frame contains two messages.
```
14:16:54.860503 [RLC ] [I] SRB1 Status packet requested through polling bit
14:16:54.860507 [RLC ] [I] SRB1 Rx SDU (8 B)
             0000: 04 32 02 20 e5 90 ce 97 
14:16:54.860512 [PDCP] [I] RX SRB1 PDU (8 B), do_integrity = false, do_encryption = false
             0000: 04 32 02 20 e5 90 ce 97 
14:16:54.860515 [PDCP] [I] RX SRB1 PDU SN: 4
             0000: 32 02 20 
```
Contains the RRC Security Mode Command.
```
14:16:54.860519 [RLC ] [I] SRB1 Rx SDU (134 B)
             0000: 05 e4 f6 7a e1 27 9b 6b 0e 60 86 69 c1 31 28 01 
             0010: 31 bc 2d 85 cc d5 cf 0e 4f 01 2f 1e 93 58 08 a4 
14:16:54.860525 [PDCP] [I] RX SRB1 PDU (134 B), do_integrity = false, do_encryption = false
             0000: 05 e4 f6 7a e1 27 9b 6b 0e 60 86 69 c1 31 28 01 
             0010: 31 bc 2d 85 cc d5 cf 0e 4f 01 2f 1e 93 58 08 a4
```
This is the packet *after* RRC Security Mode Command; encryption & integrity not applied.
```
14:16:54.860529 [PDCP] [I] RX SRB1 PDU SN: 5
             0000: e4 f6 7a e1 27 9b 6b 0e 60 86 69 c1 31 28 01 31 
             0010: bc 2d 85 cc d5 cf 0e 4f 01 2f 1e 93 58 08 a4 b8 
14:16:54.860959 [RRC ] [D] SRB1 - Rx securityModeCommand (3 B)
             0000: 32 02 20 
14:16:54.860972 [RRC ] [D] Content:
[
  {
    "DL-DCCH-Message": {
      "message": {
        "c1": {
          "securityModeCommand": {
            "rrc-TransactionIdentifier": 1,
            "criticalExtensions": {
              "c1": {
                "securityModeCommand-r8": {
                  "securityConfigSMC": {
                    "securityAlgorithmConfig": {
                      "cipheringAlgorithm": "eea2",
                      "integrityProtAlgorithm": "eia2"
                    }
                  }
                }
              }
            }
          }
        }
      }
    }
  }
]
14:16:54.860978 [RRC ] [I] Received Security Mode Command eea: 128-EEA2, eia: 128-EIA2
14:16:54.860981 [RRC ] [D] UE K_asme
             0000: cc 2a a9 7b 28 e0 00 fa 11 af ef 97 c3 46 b5 91 
             0010: de 88 1c b8 7d 06 f8 13 e6 c9 bf 65 ef 02 70 02 
14:16:54.860986 [RRC ] [D] Generating K_enb with UL NAS COUNT: 0
14:16:54.861001 [RRC ] [I] RRC encryption key - k_rrc_enc
             0000: 17 e5 fb 37 78 3d a7 d4 d8 14 6c b9 63 cd 2b cc 
             0010: 16 39 e6 6b bf 57 f9 7e 79 93 29 bb bd ec 0e 4e 
14:16:54.861009 [RRC ] [I] RRC integrity key  - k_rrc_int
             0000: 5c 9b 86 17 33 33 e7 57 cb 32 1f ed e7 3a 10 0b 
             0010: a7 77 64 15 43 91 97 02 95 50 43 7c 20 a5 1e 2a 
14:16:54.861016 [RRC ] [I] UP encryption key  - k_up_enc
             0000: 16 86 42 16 d2 59 02 5e 5f ca 2c d0 fd 97 11 8b 
             0010: 7c 58 75 cc 4c 6c 08 bd aa fb cc 6b 3f 9e 62 e8 
14:16:54.861021 [PDCP] [I] Configuring security with 128-EIA2 and 128-EEA2
14:16:54.861023 [PDCP] [D] K_rrc_enc
             0000: 17 e5 fb 37 78 3d a7 d4 d8 14 6c b9 63 cd 2b cc 
             0010: 16 39 e6 6b bf 57 f9 7e 79 93 29 bb bd ec 0e 4e 
14:16:54.861029 [PDCP] [D] K_up_enc
             0000: 16 86 42 16 d2 59 02 5e 5f ca 2c d0 fd 97 11 8b 
             0010: 7c 58 75 cc 4c 6c 08 bd aa fb cc 6b 3f 9e 62 e8 
14:16:54.861033 [PDCP] [D] K_rrc_int
             0000: 5c 9b 86 17 33 33 e7 57 cb 32 1f ed e7 3a 10 0b 
             0010: a7 77 64 15 43 91 97 02 95 50 43 7c 20 a5 1e 2a 
14:16:54.861039 [PDCP] [D] K_up_int
             0000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
             0010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
14:16:54.861047 [RRC ] [D] Preparing Security Mode Complete
14:16:54.861057 [RRC ] [D] SRB1 - Tx securityModeComplete (2 B)
             0000: 2a 00 
14:16:54.861064 [RRC ] [D] Content:
[
  {
    "UL-DCCH-Message": {
      "message": {
        "c1": {
          "securityModeComplete": {
            "rrc-TransactionIdentifier": 1,
            "criticalExtensions": {
              "securityModeComplete-r8": {
              }
            }
          }
        }
      }
    }
  }
]
14:16:54.861067 [PDCP] [I] TX SRB1 SDU, SN: 5, do_integrity = true, do_encryption = false
             0000: 2a 00 
14:16:54.861072 [PDCP] [D] Integrity gen input: COUNT 5, Bearer ID 1, Direction Uplink
14:16:54.861074 [PDCP] [D] Integrity gen input msg:
             0000: 05 2a 00 
14:16:54.861077 [PDCP] [D] MAC (generated)
             0000: 62 bc dc 61 
14:16:54.861081 [RLC ] [I] SRB1 Tx SDU (7 B, tx_sdu_queue_len=0)
             0000: 05 2a 00 62 bc dc 61 
14:16:54.861977 [RRC ] [E] Failed to unpack DL-DCCH message
```
RRC is trying to decode the still encrypted contents, and fails.

andrepuschmann, 2020-02-16:
Thanks Merlin. We are aware of this issue. It will be fixed in the next release.

mrlnc, 2020-02-16:
Sure, I just want to share the workaround in case anybody runs into this issue.

andrepuschmann, 2020-04-09:
Hey Merlin, I hope you're doing good. I was wondering if you currently have access to the particular eNB causing the issue? If so, would you be happy to run a test against it to see if it is fixed?

mrlnc, 2020-04-09:
Seems resolved on a2ade9edd1fe8e701f5fa3bfc51f85b3789a1d11. I didn't perform any additional tests except the attach.

Thanks!

ue.log, ue.pcap, nas.pcap:
[smc_andre.zip](https://github.com/srsLTE/srsLTE/files/4455593/smc_andre.zip)

andrepuschmann, 2020-04-09:
Great. It will be available with the next public release.

closed 2020-04-09 by andrepuschmann