srsran/srsRAN_4G #464: rf_soapy_imp.c:348:9: error: too many arguments to function ‘SoapySDRDevice_setupStream’

epipenau, 2020-02-25:
I am using Soapy and LimeSuite from github fresh today and srsLTE wont compile

/root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c: In function ‘rf_soapy_open_multi’:
/root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:327:36: warning: passing argument 2 of ‘SoapySDRDevice_setupStream’ makes integer from pointer without a cast [-Wint-conversion]
                                    &handler->rxStream,
                                    ^~~~~~~~~~~~~~~~~~
In file included from /root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘int’ but argument is of type ‘SoapySDRStream **’ {aka ‘struct SoapySDRStream **’}
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:328:36: warning: passing argument 3 of ‘SoapySDRDevice_setupStream’ makes pointer from integer without a cast [-Wint-conversion]
                                    SOAPY_SDR_RX,
                                    ^~~~~~~~~~~~
In file included from /root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const char *’ but argument is of type ‘int’
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:329:36: warning: passing argument 4 of ‘SoapySDRDevice_setupStream’ from incompatible pointer type [-Wincompatible-pointer-types]
                                    SOAPY_SDR_CF32,
                                    ^~~~~~~~~~~~~~
In file included from /root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const size_t *’ {aka ‘const long unsigned int *’} but argument is of type ‘char *’
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:330:36: warning: passing argument 5 of ‘SoapySDRDevice_setupStream’ makes integer from pointer without a cast [-Wint-conversion]
                                    rx_channels,
                                    ^~~~~~~~~~~
In file included from /root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘size_t’ {aka ‘const long unsigned int’} but argument is of type ‘size_t *’ {aka ‘long unsigned int *’}
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:331:43: warning: passing argument 6 of ‘SoapySDRDevice_setupStream’ makes pointer from integer without a cast [-Wint-conversion]
                                    handler->num_rx_channels,
                                    ~~~~~~~^~~~~~~~~~~~~~~~~
In file included from /root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const SoapySDRKwargs *’ {aka ‘const struct <anonymous> *’} but argument is of type ‘size_t’ {aka ‘long unsigned int’}
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:326:9: error: too many arguments to function ‘SoapySDRDevice_setupStream’
     if (SoapySDRDevice_setupStream(handler->device,
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Device.h:295:31: note: declared here
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:349:36: warning: passing argument 2 of ‘SoapySDRDevice_setupStream’ makes integer from pointer without a cast [-Wint-conversion]
                                    &handler->txStream,
                                    ^~~~~~~~~~~~~~~~~~
In file included from /root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘int’ but argument is of type ‘SoapySDRStream **’ {aka ‘struct SoapySDRStream **’}
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:351:36: warning: passing argument 4 of ‘SoapySDRDevice_setupStream’ from incompatible pointer type [-Wincompatible-pointer-types]
                                    SOAPY_SDR_CF32,
                                    ^~~~~~~~~~~~~~
In file included from /root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const size_t *’ {aka ‘const long unsigned int *’} but argument is of type ‘char *’
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:352:36: warning: passing argument 5 of ‘SoapySDRDevice_setupStream’ makes integer from pointer without a cast [-Wint-conversion]
                                    tx_channels,
                                    ^~~~~~~~~~~
In file included from /root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘size_t’ {aka ‘const long unsigned int’} but argument is of type ‘size_t *’ {aka ‘long unsigned int *’}
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:353:43: warning: passing argument 6 of ‘SoapySDRDevice_setupStream’ makes pointer from integer without a cast [-Wint-conversion]
                                    handler->num_tx_channels,
                                    ~~~~~~~^~~~~~~~~~~~~~~~~
In file included from /root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const SoapySDRKwargs *’ {aka ‘const struct <anonymous> *’} but argument is of type ‘size_t’ {aka ‘long unsigned int’}
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:348:9: error: too many arguments to function ‘SoapySDRDevice_setupStream’
     if (SoapySDRDevice_setupStream(handler->device,
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /root/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Device.h:295:31: note: declared here
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~

jonah01, 2020-02-27:
I have the same problem. Is there a solution? Best regards :)

jonah01, 2020-02-28:
I had the same problem and found a remedy. 
You have to use an older version of Soapy-sdr (soapy-sdr-0.7.1). 
Make sure that you have installed everything in the correct order. 
I wrote a simple bash script for that, maybe it helps:

`#!/bin/sh
if [ "$(id -u)" == "0" ]; then
   echo "This script must be run as normal User" 1>&2
   exit 1
fi

user="$USER"
sudo add-apt-repository -y ppa:myriadrf/drivers
sudo apt update -y
sudo apt upgrade -y
sudo apt install g++ -y
sudo apt install libcurl4-openssl-dev -y
sudo apt install libssl-dev -y
sudo apt install libsqlite3-dev -y
sudo apt install libboost-program-options-dev -y
sudo apt install zlib1g-dev -y
sudo apt install plink -y
sudo apt install libpython-dev -y
sudo apt install python-numpy -y
sudo apt install swig -y
sudo apt install libgps-dev -y
sudo apt install gpsd -y
sudo apt install gpsd-clients -y
sudo apt install libjsoncpp-dev -y
sudo apt install libusb-1.0-0-dev -y
sudo apt install git -y
sudo apt install cmake -y
sudo apt install build-essential -y
sudo apt install libmbedtls-dev -y
sudo apt install libconfig++-dev -y
sudo apt install libsctp-dev -y
sudo apt install libfftw3-dev -y
sudo apt install libi2c-dev -y
sudo apt install libboost-all-dev -y

if [ -d  /home/$user/SoapySDR ]; then
    echo " - SoapySDR already exists. Skipping it."

else
  git clone https://github.com/pothosware/SoapySDR.git
  cd SoapySDR || exit
  #git pull origin master
  git pull https://github.com/pothosware/SoapySDR soapy-sdr-0.7.1
  git checkout soapy-sdr-0.7.1
  mkdir build
  cd build || exit
  cmake ..
  make -j4
  sudo make install
  sudo ldconfig #needed on debian systems
  cd /home/$user/ || exit
fi

if [ -d  /home/$user/LimeSuite ]; then
  echo " - LimeSuite already exists. Skipping it."

else
  sudo apt install libwxgtk3.0-dev -y
  sudo apt install freeglut3-dev -y

  git clone https://github.com/myriadrf/LimeSuite.git
  cd LimeSuite || exit
  git checkout stable
  mkdir builddir && cd builddir || exit
  cmake ../
  make -j4
  sudo make install
  sudo ldconfig
  cd /home/$user/LimeSuite/udev-rules || exit
  chmod +x install.sh
  sudo ./install.sh
  cd /home/$user/ || exit
fi

if [ -d  /home/$user/messbox ]; then
  echo " - messbox already exists. Skipping it."

else
  git cloen https://github.com/srsLTE/srsLTE.git
  cd srsLTE || exit
  mkdir build && cd build || exit
  cmake ..
  make -j4
  sudo make install
  sudo ldconfig
  cd /home/$user/ || exit
fi

echo "Install successful"`

andrepuschmann, 2020-02-28:
SoapySDR has updated their API in version 0.8. This will be updated on our side in the next release. Until then, make sure to use SoapySDR <0.8.

@jonah01 Thanks for the script.

closed 2020-02-28 by andrepuschmann

jonah01, 2020-02-28:
No problem, I'm glad I could help.