srsran/srsRAN_4G #465: make[1]: *** [lib/src/phy/rf/CMakeFiles/srslte_rf.dir/all] Error 2

esraltun, 2020-02-25:
Hi,
I tried to make srsLTE but it failed. I didn't find polarssl (libpolarssl-dev) and I didnt understand why not find sctp (I used both ubuntu 16.04 and 18.04) 

ubuntu@arm:~/srsLTE/build$ cmake ..                                             
-- CMAKE_SYSTEM: Linux-4.14.0-xilinx-                                           
-- CMAKE_SYSTEM_PROCESSOR: armv7l                                               
-- CMAKE_CXX_COMPILER: /usr/bin/c++                                             
-- Build type not specified: defaulting to Release.                             
-- FFTW3F LIBRARIES: /usr/lib/arm-linux-gnueabihf/libfftw3f.so                  
-- FFTW3F STATIC LIBRARIES: /usr/lib/arm-linux-gnueabihf/libfftw3f.a            
-- FFTW3F INCLUDE DIRS: /usr/include                                            
-- FFT_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libfftw3f.so                     
-- Checking for module 'polarssl'                                               
--   No package 'polarssl' found                                                
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND                              
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND                
-- POLARSSL INCLUDE DIRS: POLARSSL_INCLUDE_DIRS-NOTFOUND                        
-- Could NOT find POLARSSL (missing: POLARSSL_LIBRARIES POLARSSL_INCLUDE_DIRS)  
-- Checking for module 'mbedtls'                                                
--   No package 'mbedtls' found                                                 
-- MBEDTLS LIBRARIES: /usr/lib/arm-linux-gnueabihf/libmbedcrypto.so             
-- MBEDTLS STATIC LIBRARIES: /usr/lib/arm-linux-gnueabihf/libmbedcrypto.a       
-- MBEDTLS INCLUDE DIRS: /usr/include                                           
-- PCSC LIBRARIES: /usr/lib/arm-linux-gnueabihf/libpcsclite.so                  
-- PCSC INCLUDE DIRS: /usr/include/PCSC                                         
-- Building with PCSC support.                                                  
-- FINDING SOAPY.                                                               
-- Boost version: 1.65.1                                                        
-- Found the following Boost libraries:                                         
--   program_options                                                            
-- Using default glibc header for IPv6 support.                                 
-- Detected ARM processor                                                       
-- CMAKE_C_FLAGS is  -Wno-unused-but-set-variable -Wall -Wno-comment -Wno-write-
strings -Winline -Wno-unused-result -Wformat -Wmissing-field-initializers -Wtype
-limits -std=c99 -D_GNU_SOURCE -O3 -fno-trapping-math -fno-math-errno -DBUILD_TY
PE_RELEASE -DIS_ARM -DHAVE_NEON -mfloat-abi=hard -mfpu=neon -fvisibility=hidden 
-Werror                                                                         
-- CMAKE_CXX_FLAGS is  -Wall -Wno-comment -Wno-reorder -Wno-unused-variable -Wty
pe-limits -std=c++11 -Wno-unused-but-set-variable -O3 -fno-trapping-math -fno-ma
th-errno -DBUILD_TYPE_RELEASE -DUSE_GLIBC_IPV6 -DIS_ARM -mfloat-abi=hard -mfpu=n
eon -Werror                                                                     
-- Using install prefix: /usr/local                                             
-- Building for version: 19.12.0                                                
--    examples will be installed.                                               
-- Checking for module 'sctp'                                                   
--   No package 'sctp' found                                                    
-- SCTP LIBRARIES: /usr/lib/arm-linux-gnueabihf/libsctp.so                      
-- SCTP INCLUDE DIRS: /usr/include                                              
-- Skipping phy_dl_test-p6-t2-q-m27 on ARM                                      
-- Skipping phy_dl_test-p6-t3-q-m27 on ARM                                      
-- Skipping phy_dl_test-p6-t4-q-m27 on ARM                                      
-- Skipping phy_dl_test-p25-t3-m28 on ARM                                       
-- Skipping phy_dl_test-p25-t4-m28 on ARM                                       
-- Skipping phy_dl_test-p25-t2-q-m27 on ARM                                     
-- Skipping phy_dl_test-p25-t3-q-m27 on ARM                                     
-- Skipping phy_dl_test-p25-t4-q-m27 on ARM                                     
-- No post-build command defined                                                
-- Building with srsUE                                                          
-- No post-build-UE command defined                                             
-- No post-build command defined                                                
-- Building with srsENB                                                         
-- Found LibConfig++: /usr/lib/arm-linux-gnueabihf/libconfig++.so               
-- static LibConfig++ path: /usr/lib/arm-linux-gnueabihf/libconfig++.a          
-- Found LibConfig: /usr/lib/arm-linux-gnueabihf/libconfig.so                   
-- static LibConfig path: /usr/lib/arm-linux-gnueabihf/libconfig.a              
-- Checking for module 'sctp'                                                   
--   No package 'sctp' found                                                    
-- SCTP LIBRARIES: /usr/lib/arm-linux-gnueabihf/libsctp.so                      
-- SCTP INCLUDE DIRS: /usr/include                                              
-- No post-build-ENB command defined                                            
[... 127 lines cut ...]
In file included from /home/ubuntu/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:0:   
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const SoapySDRKw  a
rgs * {aka const struct <anonymous> *}’ but argument is of type ‘size_t {ak    a
 unsigned int}’                                                                 
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device
,                                                                               
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~                       
/home/ubuntu/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:348:9: error: too many argumen
ts to function ‘SoapySDRDevice_setupStream’                                     
     if (SoapySDRDevice_setupStream(handler->device,                            
         ^~~~~~~~~~~~~~~~~~~~~~~~~~                                             
In file included from /home/ubuntu/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:0:   
/usr/local/include/SoapySDR/Device.h:295:31: note: declared here                
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device
,                                                                               
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~                       
cc1: all warnings being treated as errors                                       
lib/src/phy/rf/CMakeFiles/srslte_rf.dir/build.make:86: recipe for target 'lib/sr
c/phy/rf/CMakeFiles/srslte_rf.dir/rf_soapy_imp.c.o' failed                      
make[2]: *** [lib/src/phy/rf/CMakeFiles/srslte_rf.dir/rf_soapy_imp.c.o] Error 1 
CMakeFiles/Makefile2:3351: recipe for target 'lib/src/phy/rf/CMakeFiles/srslte_r
f.dir/all' failed                                                               
make[1]: *** [lib/src/phy/rf/CMakeFiles/srslte_rf.dir/all] Error 2              
Makefile:162: recipe for target 'all' failed                                    
make: *** [all] Error 2

epipenau, 2020-02-25:
yep i've got the same problem.

andrepuschmann, 2020-02-28:
SoapySDR has updated their API in version 0.8. This will be updated on our side in the next release. Until then, make sure to use SoapySDR <0.8.

closed 2020-02-28 by andrepuschmann

rembulan, 2020-07-08:
can You guide me how to obtain the right SoapySDR to solve my same problem ?