srsran/srsRAN_4G #466: [ERROR] Tx Calibration: MCU error 4 (SXT tune failed)

epipenau, 2020-02-29:
Hi,

I installed Soapy 0.7 and stable LimeSuite and I get this Tx Calibration error.

Regards,

Michael.

Built in Release mode using commit d045213f on branch master.

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srslte/enb.conf...
Opening 1 RF devices with 1 RF channels...
Soapy has found device #0: addr=24607:1027, driver=lime, label=LimeSDR Mini [USB 2.0] 1D4C3615D72304, media=USB 2.0, module=FT601, name=LimeSDR Mini, serial=1D4C3615D72304, 
[INFO] Make connection: 'LimeSDR Mini [USB 2.0] 1D4C3615D72304'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M register cache: Disabled
Setting up Rx stream with 1 channel(s)
Setting up Tx stream with 1 channel(s)
[INFO] RX LPF configured
[INFO] RX LPF configured
Set Rx bandwidth to 2.50 MHz
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 5.00 MHz
[INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
[INFO] TX LPF configured
Set Tx bandwidth to 50.00 MHz
Available device sensors: 
 - clock_locked
 - lms7_temp
Available sensors for Rx channel 0: 
 - lo_locked
Setting Rx antenna to LNAH
Setting Tx antenna to BAND1
[ERROR] Tx Calibration: MCU error 4 (SXT tune failed)
[INFO] Rx calibration finished
State of gain elements for Rx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -4.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PAD: 45.00 dB
 - IAMP: 0.00 dB
Rx antenna set to LNAH
Tx antenna set to BAND1

Warning burst preamble is not calibrated for device lime_mini. Set a value manually

Setting frequency: DL=1950.0 Mhz, UL=1870.0 MHz
Setting Sampling frequency 11.52 MHz

==== eNodeB started ===
Type <t> to view trace
[INFO] RX LPF configured
Set Rx bandwidth to 8.64 MHz
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 8.64 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 8.64 MHz
[INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
[INFO] TX LPF configured
Set Tx bandwidth to 50.00 MHz
Setting manual TX/RX offset to 70 samples
[INFO] Tx calibration finished
[INFO] Rx calibration finished

andrepuschmann, 2020-02-29:
Not sure if it's related to your issue but check your USB port. You can't run a 10 MHz cell on USB2.

chneone, 2020-03-04:
> Hi,
> 
> I installed Soapy 0.7 and stable LimeSuite and I get this Tx Calibration error.
> 
> Regards,
> 
> Michael.
> 
> Built in Release mode using commit [d045213](https://github.com/srsLTE/srsLTE/commit/d045213fb9cbf98c83c06d7c17197a9dcbfddacf) on branch master.
> 
> --- Software Radio Systems LTE eNodeB ---
> 
> Reading configuration file /root/.config/srslte/enb.conf...
> Opening 1 RF devices with 1 RF channels...
> Soapy has found device #0: addr=24607:1027, driver=lime, label=LimeSDR Mini [USB 2.0] 1D4C3615D72304, media=USB 2.0, module=FT601, name=LimeSDR Mini, serial=1D4C3615D72304,
> [INFO] Make connection: 'LimeSDR Mini [USB 2.0] 1D4C3615D72304'
> [INFO] Reference clock 40.00 MHz
> [INFO] Device name: LimeSDR-Mini
> [INFO] Reference: 40 MHz
> [INFO] LMS7002M register cache: Disabled
> Setting up Rx stream with 1 channel(s)
> Setting up Tx stream with 1 channel(s)
> [INFO] RX LPF configured
> [INFO] RX LPF configured
> Set Rx bandwidth to 2.50 MHz
> [INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
> [INFO] TX LPF configured
> Set Tx bandwidth to 5.00 MHz
> [INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
> [INFO] TX LPF configured
> Set Tx bandwidth to 50.00 MHz
> Available device sensors:
> 
> * clock_locked
> * lms7_temp
>   Available sensors for Rx channel 0:
> * lo_locked
>   Setting Rx antenna to LNAH
>   Setting Tx antenna to BAND1
>   [ERROR] Tx Calibration: MCU error 4 (SXT tune failed)
>   [INFO] Rx calibration finished
>   State of gain elements for Rx channel 0 (AGC not supported):
> * TIA: 9.00 dB
> * LNA: 30.00 dB
> * PGA: -4.00 dB
>   State of gain elements for Tx channel 0 (AGC not supported):
> * PAD: 45.00 dB
> * IAMP: 0.00 dB
>   Rx antenna set to LNAH
>   Tx antenna set to BAND1
> 
> Warning burst preamble is not calibrated for device lime_mini. Set a value manually
> 
> Setting frequency: DL=1950.0 Mhz, UL=1870.0 MHz
> Setting Sampling frequency 11.52 MHz
> 
> ==== eNodeB started ===
> Type to view trace
> [INFO] RX LPF configured
> Set Rx bandwidth to 8.64 MHz
> [INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 8.64 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
> [INFO] TX LPF configured
> Set Tx bandwidth to 8.64 MHz
> [INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
> [INFO] TX LPF configured
> Set Tx bandwidth to 50.00 MHz
> Setting manual TX/RX offset to 70 samples
> [INFO] Tx calibration finished
> [INFO] Rx calibration finished

I have the same problem，

chneone, 2020-03-04:
if I use limesdr mini，the **.conf  need be changed?

epipenau, 2020-03-05:
Just tried this on a machine with USB 3.0.  Still get the same problem.

Built in Release mode using commit d045213f on branch master.

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /etc/srslte/enb.conf...
Opening 1 RF devices with 1 RF channels...
Soapy has found device #0: addr=24607:1027, driver=lime, label=LimeSDR Mini [USB 3.0] 1D4C3615D72304, media=USB 3.0, module=FT601, name=LimeSDR Mini, serial=1D4C3615D72304, 
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D4C3615D72304'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M register cache: Disabled
Setting up Rx stream with 1 channel(s)
Setting up Tx stream with 1 channel(s)
[INFO] RX LPF configured
[INFO] RX LPF configured
Set Rx bandwidth to 2.50 MHz
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 5.00 MHz
[INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
[INFO] TX LPF configured
Set Tx bandwidth to 50.00 MHz
Available device sensors: 
 - clock_locked
 - lms7_temp
Available sensors for Rx channel 0: 
 - lo_locked
Setting Rx antenna to LNAH
Setting Tx antenna to BAND2
[ERROR] Tx Calibration: MCU error 4 (SXT tune failed)
[INFO] Rx calibration finished
State of gain elements for Rx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -4.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PAD: 45.00 dB
 - IAMP: 0.00 dB
Rx antenna set to LNAH
Tx antenna set to BAND2

Warning burst preamble is not calibrated for device lime_mini. Set a value manually

Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz

andrepuschmann, 2020-03-05:
Is the Rxant setting correct? I thought the mini only had the LNAW connected.

epipenau, 2020-03-05:
As per your info I tried LNAW but the Tx Calibration error still occurs.

andrepuschmann, 2020-04-09:
I am afraid we can't really help with RF hardware issues here. Many people have the calibration issue and I am not aware of the cause of this. Perhaps the Lime mailing list is a better spot to ask. Closing this therefore.

closed 2020-04-09 by andrepuschmann

Dastgsm, 2020-08-05:
In my supposition, this error MCU4 could be disappear using soapySDRv0.8 but i never could be compiled srslte with soapySDRv0.8, it's failed after 92%