srsran/srsRAN_4G #469: Could not create SCTP socket

jay-peng, 2020-03-03:
## Issue Description ##
I am using srsLTE to setup test environment. After I do local compile on a linux box which is Intel CPU supported, it works fine. But when I do cross compile for Ultrazed board which is for ARM platform and run srsEPC, I got error in console - "Could not create SCTP socket".

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
Use Ultrazed board to run srsLTE software, it is Ubuntu 19.04. srsLTE is almost the latest version.
srsEPC is acting as LTE core network and srsENB is acting as LTE access network. Both of two applications need to run on the same Ultrazed board. 

## Expected Behavior ##
When I run srsepc on linux box (Intel CPU), it is no any SCTP error. Then run srsenb and observed the S1AP connection is established between epc and enb.

## Actual Behaviour ##
In console after run srsepc
Reading configuration file /home/zynqmp/srslte/epc.conf...
HSS Initialized.
Could not create SCTP socket
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf208, MNC: 0xff92
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.

## Steps to reproduce the problem ##
In linux box, do cross compile for Ultrazed board which is for ARM CPU. Copy binary code and all the required library to ultrazed board and run srsEPC. Then got error in console - "Could not create SCTP socket".

alvasMan, 2020-03-09:
Can you run the `checksctp` command?
Also `lsmod | grep sctp` ?

andrepuschmann, 2020-04-09:
I guess it's an issue on the user side. Perhaps the kernel doesn't support SCTP, or the module isn't loaded. Since no follow up was provided I am closing the issue.

closed 2020-04-09 by andrepuschmann

jin-github1002, 2020-07-24:
@alvasMan  I have the same problems. I use lsmod. The output is as follows. Could you help with this? Thank you!

nvidia@nvidia-desktop:~$ lsmod | grep sctp
nf_conntrack_proto_sctp    14044  0
nf_conntrack          105820  1 nf_conntrack_proto_sctp

SoDhan, 2020-08-30:
what is the hardware that you are loading this on?

jin-github1002, 2020-08-30:
> what is the hardware that you are loading this on?

I use LimeSDR USB. Do you have any comments?

SoDhan, 2020-08-30:
likely EPC/eNB config issue. Please check ports etc and double check you have SCTP support compiled in and your kernel supports SCTP ports.

SoDhan, 2020-08-30:
please continue the discussion on the new open bug of your that you have started.

jin-github1002, 2020-08-30:
> please continue the discussion on the new open bug of yours that you have started.

Do you mean I can open a new topic about this? There is really some process on this issues. I can describe the process. I think the issue is relate to the arm processor and srsLTE. But it is little related to the hardware.

SoDhan, 2020-08-30:
I see that from your cmake log, there was a problem in detection of sctp lib.

SoDhan, 2020-08-30:
> Do you mean I can open a new topic about this? There is really some process on this issues. I can describe the process. I think the issue is relate to the arm processor and srsLTE. But it is little related to the hardware.
Please discuss further about the process.

SoDhan, 2020-08-30:
> Do you mean I can open a new topic about this? There is really some process on this issues. I can describe the process. I think the issue is relate to the arm processor and srsLTE. But it is little related to the hardware.

So, please describe in detail.

jin-github1002, 2020-08-30:
> > Do you mean I can open a new topic about this? There is really some process on this issues. I can describe the process. I think the issue is relate to the arm processor and srsLTE. But it is little related to the hardware.
> 
> So, please describe in detail.

OK. I will do tonight. Thank you for your view.