srsran/srsRAN_4G #475: Failing Attach Requests after removing UE context from srsue

jpkorva, 2020-03-25:
I’m testing srsue with commercial eNB + proprietary EPC. Initial attach succeeds. The problem is that if I detach and remove the .ctxt file, all following attach attempts fail, unless I change nas.eia parameter value. In failing attempts the dlInformationTransfer seems to be integrity protected and ciphered, instead of plain NAS message with authentication parameters, and srsue’s PDCP discards it due to an "Integrity check failure".

Is it as expected that srsue sets nas_ksi to 0 in nas::gen_attach_request() if it has no context? I understood from spec that it should be 111b i.e. “no key is available (UE to network)”. Changing the value seems to fix the issue.

andrepuschmann, 2020-03-25:
Yes, that's probably a bug in the NAS. Thanks for flagging this.

andrepuschmann, 2020-04-09:
Thanks again for flagging. It's fixed in the new version.

closed 2020-04-09 by andrepuschmann