srsran/srsRAN_4G #476: srsUE can't connect? to srsENB

p3ngdump, 2020-04-01:
Hello

my srsLTE is operated with zeroMQ fake RF.
After setting the srsLTE, running the srsEPC, srsENB, srsUE.
srsEPC and srsENB are run well.

srsUE connect to srsENB case, srsENB killed.

I don't know why srsENB was killed.

my command is following:

---- srsENB ----
sudo srsenb --rf.device_name=zmq --rf.device_args="fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=23.04e6" --expert.nof_phy_threads=1

---- srsUE ----
sudo ip netns exec ue1 srsue --rf.device_name=zmq --rf.device_args="tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6" --phy.nof_phy_threads=1

and srsENB print is following:

Built in Release mode using commit d045213f on branch master.

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file enb.conf...
Opening 1 RF devices with 1 RF channels...
Using base rate=23.04e6
Using ID=enb
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
Channel 0. Using rx_port=tcp://localhost:2001
Channel 0. Using tx_port=tcp://*:2000

Warning burst preamble is not calibrated for device zmq. Set a value manually

Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz
Setting Sampling frequency 11.52 MHz
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)

Warning TX/RX time offset has not been calibrated for device zmq. Set a value manually

==== eNodeB started ===
Type <t> to view trace
Killed

Thank you.

andrepuschmann, 2020-04-01:
Never seen that, maybe send logs and stdout of UE as well.

p3ngdump, 2020-04-01:
Thank for your response.

my logs and stdout of UE is following:

Reading configuration file ue.conf...

Built in Release mode using commit d045213f on branch master.

Opening 1 RF devices with 1 RF channels...
Using base rate=23.04e6
Using ID=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
Channel 0. Using rx_port=tcp://localhost:2000
Channel 0. Using tx_port=tcp://*:2001

Warning burst preamble is not calibrated for device zmq. Set a value manually

Waiting PHY to initialize ... Killed

Thank you.

p3ngdump, 2020-04-01:
Just before, I resolution the problem.

solution is run the /build/srsue/srsue program

Until, I don't know why print the error in command case.
Anyway, the problem is resolution.

Thank for your response!

andrepuschmann, 2020-04-01:
Not sure what you mean but glad you solved the issue.

closed 2020-04-01 by andrepuschmann