srsran/srsRAN_4G #479: Set Network Name

daisystevens100, 2020-04-08:
Hiya

Is it possible to change the network name from software radio systems to something custom?

kind regards

andrepuschmann, 2020-04-09:
It's not exposed as a parameter but you can change [this](https://github.com/srsLTE/srsLTE/blob/d045213fb9cbf98c83c06d7c17197a9dcbfddacf/srsepc/src/mme/nas.cc#L1512) line.

closed 2020-04-09 by andrepuschmann

Matioupi, 2020-06-11:
Hello,
I'm not sure to talk about the same thing as this issue does, but the advertised network name as displayed on the smartphone is always "MCC+MNC" and not the string that I can see at the given [line](https://github.com/srsLTE/srsLTE/blob/d045213fb9cbf98c83c06d7c17197a9dcbfddacf/srsepc/src/mme/nas.cc#L1512).
Am I missing something ?

andrepuschmann, 2020-06-11:
maybe the phone hasn't received the EMM_INFO yet, so it's just displaying the MCC+MNC

Matioupi, 2020-06-12:
Indeed, with info level activated on all logs, I do not see a trace of 

`  m_nas_log->info("Packing EMM Information\n");`

so it looks like execution path is not going through this.

At this time, I do not have custom sims so I'm not able to achieve a full connection, but my understanding is that this should be broadcasted even without legit connexion or with "unlegit" connexion attempt right ?

At least the phone can  see advertised commercial networks under a human readable name and not only MCC+MNC.

Matioupi, 2020-06-12:
actually it's only called from within https://github.com/srsLTE/srsLTE/blob/d94078d543ceed275cd7b6dd2261592cbf1f0d38/srsepc/src/mme/nas.cc#L1012
so it's probably available probably upon complete registration ?
is there a way to have it broadcasted "a priori" ?

andrepuschmann, 2020-06-12:
I don't know if that would work. You can give it a try.

mrlnc, 2020-06-12:
> At this time, I do not have custom sims so I'm not able to achieve a full connection, but my understanding is that this should be broadcasted even without legit connexion or with "unlegit" connexion attempt right ?

Only a few messages are being processed prior complete registration (actually, before security is activated). EMM Info is not one of these messages: if you send an unprotected EMM Info, the phone should discard the message.

I wonder where the network names during cell scan come from. Maybe a PLMN list somewhere on the phone.

kbarlee, 2020-06-12:
The network name can also be set by the 'SPN' field stored on the SIM card, i.e. if you have the name set to 'Software Radio Systems LTE' on the core, but the SIM SPN is set to 'MCC+MNC', the phone may display either.

Matioupi, 2020-06-12:
I've just been testing by using a commercial network MCC+MNC (without any change to srsLTE source code or default args) that I never connected to (not my operator) and that I do not receive in my area, and the phone will display the correct commercial operator name, so I guess it's a clue in favor of a phone hosted matching table...