srsran/srsRAN_4G #480: Incorrect decoding Full network name from EMM information message

trsvdev, 2020-04-08:
Hello,

When connecting the UE to the EPC (open5gs), the Full network name is incorrectly accepted from the EMM information NAS message. When encoding the network name, EPC uses UCS2 as described by 3GPP TS 24.008 (clause 10.3.5.5a)

Does srsLTE support this Coding Scheme?

Thank you.

andrepuschmann, 2020-04-09:
If it's wrong I'd say we don't support it. Can you provide logs and a PCAP with the messages in question?

trsvdev, 2020-04-09:
Thanks for the quick reply.

I have already found that this is not supported.
When learning, I use srsLTE as a library, and I do not have logs. 
In debugging, I found the **_liblte_mme_unpack_network_name_ie_** function and after executing **_emm_info-> full_net_name_** it contains: 0a 28 45 70 26 00 00 00 00 00 00 00 00 ....
and should be: 00 45 00 55 00 52 00 45 00 43 00 4f 00 4d, the value is "EURECOM"
Flag UCS2 in that function not process.

[emm information_s1ap.zip](https://github.com/srsLTE/srsLTE/files/4456601/emm.information_s1ap.zip)

andrepuschmann, 2020-04-24:
Thanks for the PCAP, we'll have a look at it.