srsran/srsRAN_4G #483: limesdr usb attached, then disconnected.

jin-github1002, 2020-04-17:
hi , srsLTE
Did anybody know whats wrong?

I use two limesdr usb. one limesdr is as ue in machine. The other on is as enb+epc in another machine. I run ue and enb+epc. in the ue terminal i saw "attached sucessfully" but soon it seemed to be failed. then it it was reconnecting.    After i saw "disconnected", i ping the srsapn, it is ok.

I am so conused that 1\why the ue can be attached for less than one minute. Then it is disconnected. 2\why under disconnected , the ping command is ok. it means the link between ue and unb+epc is ok.

the output in the ue terminal is as follows:

dell@dell-OptiPlex-9020:~/srsLTE/srsue$ sudo srsue ue.conf
Reading configuration file ue.conf...

Built in Release mode using commit d045213f on branch master.

Opening 1 RF devices with 1 RF channels...
Soapy has found device #0: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 3.0] 9081C05C13425, media=USB 3.0, module=FX3, name=LimeSDR-USB, serial=0009081C05C13425, 
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9081C05C13425'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M register cache: Disabled
Setting up Rx stream with 1 channel(s)
Setting up Tx stream with 1 channel(s)
[INFO] RX LPF configured
[INFO] RX LPF configured
Set Rx bandwidth to 2.50 MHz
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 5.00 MHz
[INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
[INFO] TX LPF configured
Set Tx bandwidth to 50.00 MHz
Available device sensors: 
 - clock_locked
 - lms7_temp
Available sensors for Rx channel 0: 
 - lo_locked
[INFO] Tx calibration finished
[INFO] Rx calibration finished
State of gain elements for Rx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -4.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PAD: 45.00 dB
 - IAMP: 0.00 dB
Rx antenna set to LNAL
Tx antenna set to BAND1

Warning burst preamble is not calibrated for device lime. Set a value manually

Waiting PHY to initialize ... done!
Attaching UE...
Searching cell in DL EARFCN=3400, f_dl=2685.0 MHz, f_ul=2565.0 MHz
Set Rx bandwidth to 2.50 MHz
[INFO] Rx calibration finished
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 5.00 MHz
[INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
[INFO] TX LPF configured
Set Tx bandwidth to 50.00 MHz
[INFO] Tx calibration finished
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=0.4 KHz
[INFO] RX LPF configured
Set Rx bandwidth to 11.52 MHz
[INFO] Rx calibration finished
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 11.52 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
[... 795 lines cut ...]
[INFO] RX LPF configured
Set Rx bandwidth to 2.50 MHz
[ERROR] Rx calibration: MCU error 5 (Loopback signal weak: not connected/insufficient gain?)
^CStopping srsUE... Press Ctrl+C 9 more times to force stop
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 5.00 MHz
[INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
[INFO] TX LPF configured
Set Tx bandwidth to 50.00 MHz
[INFO] Tx calibration finished
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CFO=0.7 KHz
Saving MAC PCAP file
Saving NAS PCAP file (DLT=148)
[INFO] RX LPF configured
Set Rx bandwidth to 11.52 MHz
[INFO] Rx calibration finished
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 11.52 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 11.52 MHz
[INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
[INFO] TX LPF configured
Set Tx bandwidth to 50.00 MHz
[INFO] Tx calibration finished

andrepuschmann, 2020-04-24:
You have loads of RF issues, lates, underruns. Those are causing your troubles.

closed 2020-04-24 by andrepuschmann

frankist, 2020-04-24:
Hey Jin, please use our mailing list for questions and troubleshooting

jin-github1002, 2020-04-24:
@frankist thank you so much! i have posted several times about the problems to mailing list. I didnt receive resoponse yet. DO i need to post again to the mailing list? Any help will be appreciated.