srsran/srsRAN_4G #484: Connecting Phone to srsLTE network

CarlBugeja99, 2020-04-17:
Before filing an issue, search for solutions here:
- srsLTE users mailing list (http://www.softwareradiosystems.com/mailman/listinfo/srslte-users)

## Issue Description ##
Hi All, I am still quite new to the concept of srsLTE so please excuse my lack of knowledge in certain areas.
As part of my thesis I am currently trying to connect my cellular device to my srslte network however the srslte network is not available after scanning from my phone. I fear that it could be because of missing sib files within the sib.conf file. Would appreciate any enlightment on this or if any of you managed to connect an actual phone to the srslte network. 

## Setup Details ##

I am currently using a limesdr usb device running on an ubuntu 18.04 VM.

Attached you can find device details obtained via SoapySDRUtil - probe command 

## Expected Behavior ##
Im expecting that at least the advertised network would be visible from my phone. However I believe that the SIBs need to be modified in order to ensure that my enodeB has priority over live cell towers.

## Actual Behaviour ##
While srsenb and srsepc are not producing any errors, my cell phone is not seeing the advertised. Not sure if its a calibration issue  or an issue regarding missing SIBs. 

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

Attached to this issue please find my configs. Please note that any modifications I made where based on information gather by an LTE scanner on my phone.

[Configs and drivers.zip](https://github.com/srsLTE/srsLTE/files/4495548/Configs.and.drivers.zip)

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2020-04-24:
Please check your eNB console for "lates" or other RF issues. It's very likely to be that. Also try to go to a native setup and make this work before you run in a VM. I am closing the issue because there are loads of other threads speaking about exactly the same issue.

closed 2020-04-24 by andrepuschmann

CarlBugeja99, 2020-04-24:
Hi Andre,

I found out that the issue was with the on board clock which was inaccurate. Now u so an external clock and all is working fine.

Thanks and Regards,
Carl Bugeja.