srsran/srsRAN_4G #486: PDCCH-Miss is very high when run pdsch_ue with specified rnti(=p-rnti/fffe)

lydipad, 2020-04-21:
PDCCH-Miss is very low when run pdsch_ue with default rnti(=si-rnti/ffff),
but PDCCH-Miss is very high when run pdsch_ue with specified rnti(=p-rnti/fffe)

./pdsch_ue -f 2330000000 -r ffff -n 10000

          CFO: -178.43 Hz
         RSRP: +48.8 dBm | +50.1 dBm
          SNR: +24.8 dB
           TM: 2
           Rb:   0.06 /   0.14 /   0.09 Mbps (net/maximum/processing)
   PDCCH-Miss:  0.00%[499/499]
   PDSCH-BLER:  0.00%
         TB 0: mcs=2; tbs=144
         TB 1: mcs=0; tbs=0
Press enter maximum printing debug log of 1 subframe.

./pdsch_ue -f 2330000000 -r fffe -n 100

          CFO:  +97.73 Hz
         RSRP: +38.4 dBm | +38.3 dBm
          SNR:  +1.3 dB
           TM: 2
           Rb:   0.00 /   0.25 /   0.00 Mbps (net/maximum/processing)
   PDCCH-Miss: 98.80%[1/83]
   PDSCH-BLER:  0.00%
         TB 0: mcs=6; tbs=256
         TB 1: mcs=0; tbs=0
Press enter maximum printing debug log of 1 subframe.

## Setup Details ##

rpi4 + limeSDR-mini
srsue sync to commercial network
srsLTE with the latest release

yagoda, 2020-04-21:
This specific rnti may not be present in the transmission you are receiving

lydipad, 2020-04-21:
By setting specific parameters(-r rnti ), I want to get paging messages  from pdsch_ue.

From the code, it should work, it just doesn't run the output as expected.

andrepuschmann, 2020-04-24:
You're trying to decode P-RNTI in every TTI, if there is no paging there will not be any grant for that RNTI, and hence the error rate increases. This is totally expected.

closed 2020-04-24 by andrepuschmann