srsran/srsRAN_4G #491: srsue running on 19.12 but not on 20.04

okallio, 2020-05-06:
Platform: Ubuntu 18.04.4
Device: LimeSDR Mini FW:6 GW:1.30 / LimeSuite 20.01.0

We had running config using srsLTE 19.12. Updated to srsLTE20.04.

After update to 20_04 srsUE is showing following errors:

Waiting PHY to initialize ... done!
Attaching UE...
.
Found Cell:  Mode=FDD, PCI=51, PRB=75, Ports=2, CFO=0.5 KHz
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
[ERROR] SetFrequencyCGEN(122.88 MHz) failed
[ERROR] Tune Rx Filter: MCU error 2 (CGEN tune failed)
[ERROR] setSampleRate(Rx, 0, 15.36 MHz) Failed
setSampleRate Rx fail: SoapyLMS7::setSampleRate() failed
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
[ERROR] SetFrequencyCGEN(122.88 MHz) failed
[ERROR] Tune Tx Filter: MCU error 2 (CGEN tune failed)
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
[ERROR] SetFrequencyCGEN(61.44 MHz) failed
[ERROR] setSampleRate(Tx, 0, 15.36 MHz) Failed
setSampleRate Tx fail: SoapyLMS7::setSampleRate() failed
RF status: O=0, U=0, L=1
^CStopping ..
---  exiting  ---

On running srsUE config 19_12 we had:

Attaching UE...
Searching cell in DL EARFCN=3100, f_dl=2655.0 MHz, f_ul=2535.0 MHz
Set Rx bandwidth to 2.50 MHz
[INFO] Rx calibration finished
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 5.00 MHz
[INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
[INFO] TX LPF configured
Set Tx bandwidth to 50.00 MHz
[INFO] Tx calibration finished
Setting manual TX/RX offset to 15 samples
.
Found Cell:  Mode=FDD, PCI=51, PRB=75, Ports=2, CFO=0.6 KHz
[INFO] RX LPF configured
Set Rx bandwidth to 11.52 MHz
[INFO] Rx calibration finished
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 11.52 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 11.52 MHz
[INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
RF status: O=0, U=0, L=1
[INFO] TX LPF configured
Set Tx bandwidth to 50.00 MHz
[INFO] Tx calibration finished
Setting manual TX/RX offset to 15 samples
Found PLMN:  Id=11111, TAC=1
Random Access Transmission: seq=31, ra-rnti=0x2
LRandom Access Transmission: seq=58, ra-rnti=0x2
Random Access Complete.     c-rnti=0x35ed, ta=10
RRC Connected
Network attach successful. IP: 45.45.0.14

After update also srsENB is showing following errors:

==== eNodeB started ===
Type <t> to view trace
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
[ERROR] SetFrequencyCGEN(245.76 MHz) failed
[ERROR] setSampleRate(Rx, 0, 1.92 MHz) Failed
setSampleRate Rx fail: SoapyLMS7::setSampleRate() failed
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
[ERROR] SetFrequencyCGEN(245.76 MHz) failed
[ERROR] setSampleRate(Tx, 0, 1.92 MHz) Failed
setSampleRate Tx fail: SoapyLMS7::setSampleRate() failed
Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz for cc_idx=0
^CStopping ..
---  exiting  ---

Using LimeSuite to perform CLKGEN cal everything works ok. Also SXR/SXT setting and calibrations run fine.

When srsUE (or srsENB) has been run using srsLTE v20_04 there is "VCO Compa" both High:1 and Low:1  set. Using "modules" -> "SPI" and reading address 008C we get value 367B ????

Running srsUE (orsrsENB) v19_12 "VCO Compa" there is High:1 and Low:0. Using "modules" -> "SPI" and reading address 008C we get value 267B ????

Have done quite a bit of debug on this but still can't find out what is wrong? We have similar HW setups on both versions 19_12 and 20_04 and also using similar values on enb.conf and ue.conf files.

demirten, 2020-05-13:
I have similar issues with version 20_04_1

I've been using USRP B210 and srsLTE 19_02 for a while wtihout any problem.

After upgrading to 20_04_1, iphone clients can not complete the PDN attachment process, session ends with "release-due-to-eutran-generated-reason" on every attempt.

When I go back to 19_02 setup, it always works.

I have some Android phones too and they are working with 20_04_1 without any problem.

[image]

andrepuschmann, 2020-05-14:
Hey, you are mixing up issues here quite a bit, posting about UE and the eNB. Do you mind creating separate issues for both things your seeing? This would help us looking at them in a more structured manner.  At least for the UE it seems to be a Lime driver thing. Are you sure you're using the exact same LimeSuite version? And only by changing between 20.04.1 and 19.12 you see this change in behavior?

andrepuschmann, 2020-05-14:
Also, please send complete logs and PCAPs.

iacovosi, 2020-07-29:
> 
> 
> Platform: Ubuntu 18.04.4
> Device: LimeSDR Mini FW:6 GW:1.30 / LimeSuite 20.01.0
> 
> We had running config using srsLTE 19.12. Updated to srsLTE20.04.
> 
> After update to 20_04 srsUE is showing following errors:
> 
> Waiting PHY to initialize ... done!
> Attaching UE...
> .
> Found Cell: Mode=FDD, PCI=51, PRB=75, Ports=2, CFO=0.5 KHz
> [ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
> [ERROR] SetFrequencyCGEN(122.88 MHz) failed
> [ERROR] Tune Rx Filter: MCU error 2 (CGEN tune failed)
> [ERROR] setSampleRate(Rx, 0, 15.36 MHz) Failed
> setSampleRate Rx fail: SoapyLMS7::setSampleRate() failed
> [ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
> [ERROR] SetFrequencyCGEN(122.88 MHz) failed
> [ERROR] Tune Tx Filter: MCU error 2 (CGEN tune failed)
> [ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
> [ERROR] SetFrequencyCGEN(61.44 MHz) failed
> [ERROR] setSampleRate(Tx, 0, 15.36 MHz) Failed
> setSampleRate Tx fail: SoapyLMS7::setSampleRate() failed
> RF status: O=0, U=0, L=1
> ^CStopping ..
> --- exiting ---
> 
> On running srsUE config 19_12 we had:
> 
> Attaching UE...
> Searching cell in DL EARFCN=3100, f_dl=2655.0 MHz, f_ul=2535.0 MHz
> Set Rx bandwidth to 2.50 MHz
> [INFO] Rx calibration finished
> [INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
> [INFO] TX LPF configured
> Set Tx bandwidth to 5.00 MHz
> [INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
> [INFO] TX LPF configured
> Set Tx bandwidth to 50.00 MHz
> [INFO] Tx calibration finished
> Setting manual TX/RX offset to 15 samples
> .
> Found Cell: Mode=FDD, PCI=51, PRB=75, Ports=2, CFO=0.6 KHz
> [INFO] RX LPF configured
> Set Rx bandwidth to 11.52 MHz
> [INFO] Rx calibration finished
> [INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 11.52 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
> [INFO] TX LPF configured
> Set Tx bandwidth to 11.52 MHz
> [INFO] Filter calibrated. Filter order-2nd, set to 50 MHz
> RF status: O=0, U=0, L=1
> [INFO] TX LPF configured
> Set Tx bandwidth to 50.00 MHz
> [INFO] Tx calibration finished
> Setting manual TX/RX offset to 15 samples
> Found PLMN: Id=11111, TAC=1
> Random Access Transmission: seq=31, ra-rnti=0x2
> LRandom Access Transmission: seq=58, ra-rnti=0x2
> Random Access Complete. c-rnti=0x35ed, ta=10
> RRC Connected
> Network attach successful. IP: 45.45.0.14
> 
> After update also srsENB is showing following errors:
> 
> ==== eNodeB started ===
> Type to view trace
> [ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
> [ERROR] SetFrequencyCGEN(245.76 MHz) failed
> [ERROR] setSampleRate(Rx, 0, 1.92 MHz) Failed
> setSampleRate Rx fail: SoapyLMS7::setSampleRate() failed
> [ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
> [ERROR] SetFrequencyCGEN(245.76 MHz) failed
> [ERROR] setSampleRate(Tx, 0, 1.92 MHz) Failed
> setSampleRate Tx fail: SoapyLMS7::setSampleRate() failed
> Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz for cc_idx=0
> ^CStopping ..
> --- exiting ---
> 
> Using LimeSuite to perform CLKGEN cal everything works ok. Also SXR/SXT setting and calibrations run fine.
> 
> When srsUE (or srsENB) has been run using srsLTE v20_04 there is "VCO Compa" both High:1 and Low:1 set. Using "modules" -> "SPI" and reading address 008C we get value 367B ????
> 
> Running srsUE (orsrsENB) v19_12 "VCO Compa" there is High:1 and Low:0. Using "modules" -> "SPI" and reading address 008C we get value 267B ????
> 
> Have done quite a bit of debug on this but still can't find out what is wrong? We have similar HW setups on both versions 19_12 and 20_04 and also using similar values on enb.conf and ue.conf files.

Dear Sir,
Did you solve your problem?

andrepuschmann, 2020-09-14:
Closing this issue as now updates have been provided on this from the original author.

closed 2020-09-14 by andrepuschmann