srsran/srsRAN_4G #498: Wrong throughput reported by metrics hub

igorskh, 2020-05-19:
##  Issue Description ##
TX bitrate at srsEnb reported from metrics hub is lower than expected.

I have recalculated throughput and number of packets from srsENB logs with `info` level enabled, aggregating PDSCH reports. Basically summing up the TBS from these lines:
```
11:47:51.985216 [PHY0] [I] [  357] PDSCH: cc=0, rnti=0x2, nof_prb=3, nof_re=450, tbs={11}, mod={2}, rv={0}, tti_tx_dl=361
```
Results from logs match iperf3 reporting. 

To validate I have added number of packets and tx errors to CSV metrics output:
```cpp
tx_packets_sum += metrics.stack.mac[i].tx_pkts;
tx_errors_sum += metrics.stack.mac[i].tx_errors;
```

Turned out that `metrics.stack.mac[i].nof_tti` is 2000-4000 (about 2-4 TTIs per ms) resulting in  throughput being lower by the factor 2 to 3: 
https://github.com/srsLTE/srsLTE/blob/d94078d543ceed275cd7b6dd2261592cbf1f0d38/srsenb/src/metrics_csv.cc#L78

I have also checked that number of TTIs used for calculation is incremented multiple times per millisecond.
https://github.com/srsLTE/srsLTE/blob/e3e38482100f5bf13f15af3bc09df882e937bc60/srsenb/src/stack/mac/mac.cc#L672

## Setup Details ##
RF front-end: ZeroMQ

CPU: Intel® Core™ i7-6700K CPU @ 4.00GHz × 4 

Memory: 61,0 GiB

Operating system: 
```
Linux 4.15.0-76-lowlatency #86-Ubuntu SMP PREEMPT Fri Jan 17 18:08:33 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
Commit: c892ae56be5302eaee5ca00e270efc7a5ce6fbb2

Number of PHY threads are configured to 1, as well as actual number of workers. No CA or MIMO.

## Expected Behavior ##
Number of TTIs used for normalising throughput does not exceed 1000 per second in given conditions.

Throughput from metrics hub matches throughput aggregated from logs. 

## Actual Behaviour ##
Number of TTIs incremented multiple times per millisecond resulting in incorrect throughput.

My assumption that at this point:
https://github.com/srsLTE/srsLTE/blob/e3e38482100f5bf13f15af3bc09df882e937bc60/srsenb/src/stack/mac/mac.cc#L672

The TTI is not transmitted yet and it cannot be counted for normalising throughput.

## Steps to reproduce the problem ##
Configuration files:
[config.zip](https://github.com/srsLTE/srsLTE/files/4650196/config.zip)

Run commands:
```bash
sudo ./srsepc/src/srsepc ../../config/epc.conf
./srsenb/src/srsenb ../../config/enb.conf --expert.nof_phy_threads=1
sudo ./srsue/src/srsue ../../config/ue.conf
```

Traffic:
```bash
sudo ip netns exec ue1 iperf3 -c 172.16.0.1 -u -R -b 1M
```

## Additional Information ##
Logs:
[logs.zip](https://github.com/srsLTE/srsLTE/files/4650729/logs.zip)

Aggregated CSV report:
[enb_metrics_eval.txt](https://github.com/srsLTE/srsLTE/files/4650201/enb_metrics_eval.txt)

TTIs per ms used for calculating throughput:
[image]

igorskh, 2020-05-20:
Maybe it's related, with ZeroMQ frontend, end-to-end (PDN-GW->UE) DL latency is suspiciously low:
```text
2020/05/20 16:15:42 latency = 1.61 ms, rtt = 8.73 ms
2020/05/20 16:15:43 latency = 1.17 ms, rtt = 10.12 ms
2020/05/20 16:15:44 latency = 0.95 ms, rtt = 11.52 ms
2020/05/20 16:15:45 latency = 1.86 ms, rtt = 10.67 ms
2020/05/20 16:15:46 latency = 1.47 ms, rtt = 8.54 ms
```

Considering each transmission being scheduled in 4 TTIs, it shouldn't fall below 4 ms? That what I see using USRP as an RF Frontend.

and the log show that the transmission is in 4 TTIs of current one:
```text
14:15:42.754479 [PHY0] [I] [ 9010] PDSCH: cc=0, rnti=0x46, nof_prb=3, nof_re=450, tbs={85}, mod={4}, rv={0}, tti_tx_dl=9014
```

igorskh, 2020-05-20:
I've injected the probe here for ZMQ and UHD, the configs are exactly the same. From the time I've calculated the difference between probe i and i-1.
https://github.com/srsLTE/srsLTE/blob/e3e38482100f5bf13f15af3bc09df882e937bc60/srsenb/src/stack/mac/mac.cc#L672

[image]
[image]

andrepuschmann, 2020-05-20:
Igor, I am actually not entirely sure if I understand the problem you are having.

Here are some thoughts on some of the points you were making.

> Considering each transmission being scheduled in 4 TTIs, it shouldn't fall below 4 ms? That what I see using USRP as an RF Frontend.

The ZMQ radio is, on purpose, time less. One TTI takes as long as it takes to send one subframe worth of samples from the transmitter to the receiver. If you are using 100 PRB and send one TTI over a 28k modem, this TTI may take minutes. But if you're running all on localhost it may take only a fraction of a ms. In the latter case, you would actually observe lower RTT (ping) between a UE and P-GW than the typical ~10-20ms you use over real HW, e.g. the USRP, where one ms takes (more or less) one ms.

igorskh, 2020-05-21:
Andre, thank you for this explanation. I assumed that ZeroMQ is just replacing the RF frontend, but time less concept is somehow confusing. This does, however, explain the number of TTIs and lower latency. 

Now, I am having two issues:
1. The initial one that the throughput reported by metrics hub, both in STDOUT and CSV, is lower than the actual one when using ZMQ. And this is connected to number of TTIs per second.
2. If ZMQ is time less, does it mean that eventually more than 1000 subframes can be transmitted per second? How would I get comparable to other RF frontend results if timing is different? One critical use case that I see is latency evaluation, using ZMQ would be really helpful for development before testing on real hardware.

andrepuschmann, 2020-05-21:
> I assumed that ZeroMQ is just replacing the RF frontend, but time less concept is somehow confusing.

It does replace the RF front end indeed. And it makes a lot of sense if you think what a real RF front end does when you configure a certain sample rate. It provides you with samples at the configured rate (more or less). And the timing is somehow implicitly given from the clock in the device. ZMQ is basically free-running, best effort radio. You can check if your ZMQ radio is faster or slower by looking at the TTIs that are processed every second. Note that the console trace still runs with system timers. So in the example below, every second we process ~600 TTIs, which means it runs slower than a read radio.

```
rnti cqi  ri mcs brate   ok  nok  (%)  snr  phr mcs brate   ok  nok  (%)   bsr
  4f  15 2.0  28  150M  620    0   0%  105 30.0  20  110k    8    0   0%   0.0
  4f  15 2.0  28  150M  582    0   0%  105 30.0  20  103k    7    0   0%   0.0
  4f  15 2.0  28  151M  612    0   0%  105 30.0  20  112k    8    0   0%   0.0
  4f  15 2.0  28  150M  590    0   0%  105 30.0  20  101k    7    0   0%   0.0
  4f  15   0  28  150M  606    0   0%  105 30.0  20  113k    8    0   0%   0.0
```

> If ZMQ is time less, does it mean that eventually more than 1000 subframes can be transmitted per second? How would I get comparable to other RF frontend results if timing is different? One critical [...]

More or less subframes, yes. It depends on your setup. If you want to really mimic a RF device, also in terms of timing, you would need to add something like the "throttle block" from GNU Radio. Basically it matches the sample rate in free-running applications.

andrepuschmann, 2020-05-22:
Igor, with the analysis and your current understanding, do you still see an issue in how we calculate the throughput? In other words, can the issue be closed?

igorskh, 2020-05-22:
Thanks. The latency and number of TTIs are clear for me now. I'll work on matching sampling rates.

However, I still don't get what exactly bitrate shows in this case, it's not the Mbit/s in terms of system clock, but can be converted, if I know total number of TTIs. Although if I don't run the full load, the total number is not reported.

So if I understand  the calculation correctly, in the 1st line of the example the real achieved bitrate per second was (150*0.62)M? 

For example with 1 Mbit/s load:

  dl_brate | tx_packets | tx_errors | dl_brate_mbps | dl_brate_logs | dl_count_logs | dl_brate_mbps_logs
  -- |  -- | -- | -- | -- | -- | --
 470723,375 |  32 | 0 | 0,45 | 1006768 | 32 | 0,96

Only 32 packets were transmitted in about 2133 TTIs, but looking at the metrics report there is no way of knowing that. What does the normalised value represent? Maybe it would be helpful to see actual number of processed TTIs per report period along with N/ACKs?

andrepuschmann, 2020-05-22:
> However, I still don't get what exactly bitrate shows in this case, it's not the Mbit/s in terms of system clock, but can be converted, if I know total number of TTIs. Although if I don't run the full load, the total number is not reported.

That's correct. We currently don't report that value on the UE console. It will be added shortly. Meanwhile you can copy+paste the eNB code, it's done there.

> So if I understand the calculation correctly, in the 1st line of the example the real achieved bitrate per second was (150*0.62)M?

Yes, that's correct.

> Only 32 packets were transmitted in about 2133 TTIs, but looking at the metrics report there is no way of knowing that. What does the normalised value represent? Maybe it would be helpful to see actual number of processed TTIs per report period along with N/ACKs?

Yes, see above. Use the eNB console for that to get the TTIs or add it to the UE yourself.

andrepuschmann, 2020-05-22:
Another approach to think about it is "simulated throughput" vs. "simulation throughput". I.e. the first is what you see over the (possibly slowed down) ZMQ based radio, for example, you get full 150Mbit simulated throughput if this was going through LTE. The latter, i.e. "simulation throughput", is what your iperf application reports. This is usually less than the first.

igorskh, 2020-05-22:
>  Use the eNB console for that to get the TTIs

That's exactly what I use, but I don't see the total number of TTIs in the eNB metrics output, only (N)ACKs, am I missing it?

eNB outputs:
```text
------DL--------------------------------UL------------------------------------
rnti cqi  ri mcs brate   ok  nok  (%)  snr  phr mcs brate   ok  nok  (%)   bsr
  47 10.0   0  20  438k   58    0   0% 96.1 30.0  19   19k   10    0   0%   0.0
  47 10.0   0  20  432k   59    0   0% 96.9 30.0  19   18k   10    0   0%   0.0
  47 10.0   0  20  438k   58    0   0% 96.5 30.0  19   71k   20    0   0%   0.0
```

Iperf outputs:
```text
[  4] local 172.16.0.15 port 57004 connected to 172.16.0.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   235 KBytes  1.93 Mbits/sec                  
[  4]   1.00-2.00   sec  33.9 KBytes   278 Kbits/sec                  
[  4]   2.00-3.00   sec   136 KBytes  1.12 Mbits/sec
```

If I run traffic generator with full load I see total number of TTIs, but it varies, so can be used only as an approximation 
```text
------DL--------------------------------UL------------------------------------
rnti cqi  ri mcs brate   ok  nok  (%)  snr  phr mcs brate   ok  nok  (%)   bsr
  46 10.0   0  20   20M 1709    0   0% 98.3 30.0  18  323k  190    0   0%   0.0
  46 10.0   0  20   20M 1730    0   0% 97.8 30.0  18  370k  192    0   0%   0.0
  46 10.0   0  20   20M 1729    0   0% 97.9 30.0  18  323k  192    0   0%   0.0
```

> Another approach to think about it is "simulated throughput" vs. "simulation throughput".

Worth noting somewhere along with the `time less` concept, it's wasn't so obvious for me before this discussion. Although, I guess it's common for the simulations of this kind.

Now, after clarifications from your side it's clear how it works. I guess reporting total number of processed TTIs would resolve this issue and help avoiding confusion.

Thanks a lot.

andrepuschmann, 2020-05-22:
> That's exactly what I use, but I don't see the total number of TTIs in the eNB metrics output, only (N)ACKs, am I missing it?

No, it's not printed explicitly. But if you have a metric periodicity of 1s, you have exactly 1000 TTIs in that second. And with a single user and TM1, you can send 1000 transport blocks to that user. So 1000 would be the max.

If you see 58, it means only 58 TBs have been send. But if you only have a very low rate it may be enough to send all data. This is usually a sign that your PHY or IQ transport isn't maxed out.

So its worth to note that we are not reporting the number of TTIs here, just the number of TBs.

> If I run traffic generator with full load I see total number of TTIs, but it varies, so can be used only as an approximation

This looks like you're PHY or IQ transport is the limitation. But you process almost twice as many TBs as you would do using real RF.

> Worth noting somewhere along with the time less concept, it's wasn't so obvious for me before this discussion. Although, I guess it's common for the simulations of this kind.

Yes, I agree. We will expand the app note. Thanks for raising this.

igorskh, 2020-05-22:
> No, it's not printed explicitly. But if you have a metric periodicity of 1s, you have exactly 1000 TTIs in that second. And with a single user and TM1, you can send 1000 transport blocks to that user. So 1000 would be the max.

That was my initial assumption, but my understanding that it is only possible to derive how fast or slow ZMQ RF compared to the real HW just by looking at number of TBs when data is transmitted every TTI.
In other words, without combining with iPerf logs I can't tell if that was 58 out of 1050 or 58 out of 1600 TTIs.

> This looks like you're PHY or IQ transport is the limitation. But you process almost twice as many TBs as you would do using real RF.

Sorry, there was a PHY limitation by enabled channel emulator.

What I mean is to see something like this:
```text
rnti cqi  ri mcs brate ttis   ok  nok  (%)  snr  phr mcs brate   ok  nok  (%)   bsr
  46  15   0  28  383k 2627   30    0   0%    0  0.0   0     0    0    0   0%   0.0
  46  15   0  28  409k 2619   32    0   0%    0  0.0   0     0    0    0   0%   0.0
  46  15   0  28   36M 1637 1637    0   0%  109 30.0  20  504k  182    0   0%   0.0
  46  15   0  28   36M 1631 1631    0   0%  109 30.0  20  501k  181    0   0%   0.0
  46  15   0  28  6.1M 2411  411    0   0%    0  0.0   0     0    0    0   0%   0.0
  46  15   0  28  6.4M 2381  426    0   0%    0  0.0   0     0    0    0   0%   0.0
```

Even with the same run with different traffic load, number of processed TTIs vary. This reporting can already be used for evaluation without parsing info-level logs or integrating additional probes into the code, if you think it's reasonable for other users maybe.

From my side everything is more or less clear now, thank you.

andrepuschmann, 2020-05-24:
Great to hear that we could solve your concerns. Printing the total number of TTIs seems useful for ZMQ runs alright. It's less useful for normal RF though. I'll create an internal issue and we discuss how we can handle both cases. Thanks for flagging this

closed 2020-05-24 by andrepuschmann