srsran/srsRAN_4G #500: MISSING LIBS

MoneyCatcher, 2020-05-20:
## Issue Description ##
make install do not install srslte_common, srslte_phy, srslte_radio, srslte_upper, srslte_asn1

## Setup Details ##
first seen on a ubuntu 18.04 machine after update to srslte release 20.04
apt-get update
or
apt-get install srslte srslte-dev

## Expected Behavior ##
sudo find / -name "libsrslte_common.a"
/usr/local/lib/libsrslte_common.a

## Actual Behaviour ##
no file in libs found (cmake and make create the libs in the build folder)
make install does not copy

## Steps to reproduce the problem ##
just install or make from source in a ubuntu system (maybe 18.04) 

## Additional Information ##
going back to sources from release  19.12
I was able to "make install" with libs
I think this is since 20.04.

MoneyCatcher, 2020-09-21:
The same thing, missing libs also in release 20.04.2.

Was there a renaming of the libs or an extra ubuntu package to install next to srslte-dev?

MoneyCatcher, 2020-09-21:
https://launchpad.net/~srslte/+archive/ubuntu/releases/+files/srslte-dev_20.04.2-0ubuntu1~srslte3~20.04_amd64.deb

Looking into this file for example reveals the missing /usr/lib/ folder.

In SRSLTE release 19.12 there is a /usr/lib/ folder

https://launchpad.net/~srslte/+archive/ubuntu/releases/+files/srslte-dev_19.12-0ubuntu1~srslte1~19.04_amd64.deb

MoneyCatcher, 2020-10-06:
Does anybody know how to install SRSLTE Libs and not only the header files in Ubuntu LTS 18.04 or 20.04? Any other repository or changes in Makefiles?

MoneyCatcher, 2020-10-19:
Official log also shows the missing static libraries, only headers in this file!

https://launchpadlibrarian.net/498106553/buildlog_ubuntu-focal-amd64.srslte_20.04.2-0ubuntu1~srslte3~20.04_BUILDING.txt.gz

Search for : "This package contains the development files - static libraries and headers" and check that no static libraries are not copied to this file.

closed 2020-10-20 by andrepuschmann

andrepuschmann, 2020-10-20:
Ah, GH closed this automatically bc I referenced it in the commit msg. But the result is the same, it will work again in 20.10