srsran/srsRAN_4G #502: srsenb: iphone devices failed to connect after 20_04_1 release

demirten, 2020-05-26:
I've a working srsenb lab setup with several UE's including iPhone and Android devices (with Qualcomm and MTK chipsets).

After upgrading release_20_04_1 tag, iPhone devices unable to complete initial PDN attachment procedure. User-Agent of the device is: iOS/13.3.1 (17D50) iPhone

When I go back to previous release, everything starts to working as expected. (Android devices working fine with both of the releases)

I'm attaching enb capture and log files, please let me know if you have any suggestion and then I can retry the test again.

[enb.pcap.gz](https://github.com/srsLTE/srsLTE/files/4683848/enb.pcap.gz)
[enb.log](https://github.com/srsLTE/srsLTE/files/4683849/enb.log)

andrepuschmann, 2020-05-26:
Hey Murat,
just to understand correctly: with the 20.04.1 release, Android phones work fine but your IPhone doesn't. With the 19.12, both Android an Iphone works, correct?
If so, can you send the eNB logs again when the Iphone attaces with `--log.all_level=debug --log.all_hex_limit=1000`?

Thanks
Andre

demirten, 2020-05-26:
Yes, there is no problem with the Android devices in both of the 19.12 and 20.04.1 releases.

Here is the log attached with your requested levels.

Please note that I'm testing srsenb with a commercial epc (including mme, spgw, pcrf and hss)

[enb.log](https://github.com/srsLTE/srsLTE/files/4684788/enb.log)

andrepuschmann, 2020-05-26:
Thanks, somehow PHY and MAC are not included. Can you also set `--log.phy_level=info --log.mac_level=info`, please?

demirten, 2020-05-26:
Ok, attached

[enb.log.gz](https://github.com/srsLTE/srsLTE/files/4684881/enb.log.gz)

andrepuschmann, 2020-05-26:
Hmm, we need to investigate what's going on here. The phone stops responding after receiving the ReconfigurationComplete. Can you perhaps take the same logs with the 19.12 release so we can compare? What model/version is the phone?

demirten, 2020-05-26:
It is an iphone 8plus with ios 13.3.1

Attached success enb log with 19.12 release:
[enb.1912.log.gz](https://github.com/srsLTE/srsLTE/files/4685030/enb.1912.log.gz)

andrepuschmann, 2020-09-14:
hey @demirten,
we've identified an issue in 20.04.1. Could you perhaps try with the recent 20.04.2 version again?
Thanks

demirten, 2020-09-14:
I just confirmed that iphone devices working properly on version 20.04.2

Here I'm attaching whole iphone IMS VoLTE (with ipsec) registration process capture:

[iphone-srslte-ipsec-register.pcapng.gz](https://github.com/srsLTE/srsLTE/files/5219860/iphone-srslte-ipsec-register.pcapng.gz)

andrepuschmann, 2020-09-14:
Very good. Thanks @demirten Closing the issue then

closed 2020-09-14 by andrepuschmann