srsran/srsRAN_4G #504: RX failed: Operation timed out;

shodan007, 2020-05-30:
## Issue Description ##
[Describe the issue in detail]

uname -a

Linux OpenLTE 5.6.0-kali1-amd64 #1 SMP Debian 5.6.7-1kali1 (2020-05-12) x86_64 GNU/Linux

bladeRF> version

  bladeRF-cli version:        1.8.0-0.2019.07-4+b1
  libbladeRF version:         2.2.1-0.2019.07-4+b1

  Firmware version:           2.3.2
  FPGA version:               0.11.0 (configured by USB host)

Reading configuration file /home/radio/.config/srslte/ue.conf...

Built in Release mode using commit c892ae56b on branch master.

Opening 1 channels in RF device=default with args=default
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
Waiting PHY to initialize ... Warning: Failed to create thread with real-time priority. Creating it with normal priority: Success
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Success
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Success
Warning: Failed to create thread with real-time priority. Creating it with normal priority: Resource temporarily unavailable
done!
Attaching UE...
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz
RF status: O=4, U=0, L=0
set RX frequency to 2684999998
set TX frequency to 2564999998
RF status: O=4, U=0, L=0
/home/radio/srsLTE/lib/src/phy/rf/rf_blade_imp.c.434: RX failed: Operation timed out; nsamples=1920;
                                                                                                                                                                      
./home/radio/srsLTE/lib/src/phy/rf/rf_blade_imp.c.434: RX failed: Operation timed out; nsamples=9600;
                                                                                                                                                                      
/home/radio/srsLTE/lib/src/phy/ue/ue_sync.c.775: Error receiving samples
                                                                                                                                                                      
/home/radio/srsLTE/lib/src/phy/ue/ue_cell_search.c.312: Error calling srslte_ue_sync_work()
                                                                                                                                                                      
/home/radio/srsLTE/lib/src/phy/ue/ue_cell_search.c.272: Error searching cell

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

TheDuchy, 2020-06-22:
Same here with a x40 board. Previously discussed here: https://github.com/srsLTE/srsLTE/issues/373. Issues looks like an upstream issue with libbladerf.

andrepuschmann, 2020-06-23:
The issue is currently investigated by the folks at Nuand. See [here](https://github.com/Nuand/bladeRF/issues/778).

jarek556, 2020-11-08:
Temporary solution: use FPGA firmware version 0.9.0.

yesimxev, 2021-04-18:
> Temporary solution: use FPGA firmware version 0.9.0.

Does not help for me, still experiencing the same issue 
```
[INFO] setSampleRate(Tx, 0, 1.920000 MHz), actual = 1.920000 MHz
[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
0[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
00000000000000[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
0000000000000[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
00000000000000[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
00000000000000[INFO] setSampleRate(Rx, 0, 1.920000 MHz), actual = 1.920000 MHz
```

Using the latest github repos, and also tried fpga from 0.12 to 0.8
```
  bladeRF-cli version:        1.8.0-git-e09200c3
  libbladeRF version:         2.4.0-git-e09200c3

  Firmware version:           2.4.0-git-a3d5c55f
  FPGA version:               0.9.0 (configured by USB host)
```

nomidaepapi, 2021-12-21:
The new version of SRS does not give me this issue anymore. Update everything on the blade and SRS

closed 2023-04-14 by ismagom