srsran/srsRAN_4G #505: Device zmq not found for srsLTE release 18.09

srinadhpotnuru, 2020-06-03:
## Issue Description ##
I wanted to use srsLTE release 18.09 and wanted to use it with no-RF front end application note. I placed the file "FindZeroMQ.cmake" from srsLTE release 20.04.1 and edited the CMakeLists.txt and .conf files as noted below.

Are older releases compatible for no-RF front end application note? If yes, how can I set them up?

## Setup Details ##
I placed the "FindZeroMQ.cmake" from `cmake/modules` of release 20.04.01 into `cmake/modules` of release 18.09. I changed the CMakeLists.txt in parent directory as below after the line `endif(BLADERF_FOUND OR UHD_FOUND OR SOAPYSDR_FOUND)`:

```
find_package(ZeroMQ)
if(ZEROMQ_FOUND)
  include_directories(${ZEROMQ_INCLUDE_DIRS})
  link_directories(${ZEROMQ_LIBRARY_DIRS})
  set(RF_FOUND TRUE)
  message(STATUS "RF_FOUND set TRUE")
endif(ZEROMQ_FOUND)
```
Post this, cmake builds with zmq with following output:
```
-- FINDING ZEROMQ.
-- Checking for module 'ZeroMQ'
--   No package 'ZeroMQ' found
-- Found libZEROMQ: /usr/local/include, /usr/local/lib/libzmq.so
-- RF_FOUND set TRUE
```
Executables are created for srsenb and srsue.

## Expected Behavior ##
srsue and srsenb to start with no-RF front end.

## Actual Behaviour ##
The connection is not setting up with following message on both srsenb and srsue

```
Reading configuration file /home/srinath/.srs/ue.conf...

Built in Release mode using commit  on branch .

---  Software Radio Systems LTE UE  ---

Opening RF device with 1 RX antennas...
Device zmq not found. Switching to auto mode
No compatible RF frontend found
Error opening RF device
Failed to find device zmq with args tx_port=tcp://*:5555,rx_port=tcp://localhost:5554,id=ue,base_srate=23.04e6

```

## Steps to reproduce the problem ##
I used srsLTE release 18.09 and changed the files as mentioned above

andrepuschmann, 2020-06-03:
srsLTE 18.09 is not going to work with ZMQ like the recent version. I suggest you update.

srinadhpotnuru, 2020-06-03:
Okay. Thanks for quick reply

closed 2020-06-03 by srinadhpotnuru