srsran/srsRAN_4G #506: srsenb+srsepc: Sierra Wireless modules (e.g. MC7455, EM7565) cannot register

aleksander0m, 2020-06-16:
## Issue Description ##
The testing framework I'm using to validate modem connections is currently failing with Sierra Wireless like the MC7455 or the EM7565, both running the "generic" firmware variant. I'm using SIM cards from sysmocom, and I've validated the setup works correctly with the same SIM cards, same eNodeB+EPC configuration and different modems (e.g. using Quectel devices instead). When using the Sierra Wireless modules, the network registration is failing, and I'm afraid I lack deep LTE stack knowledge to understand why this happens. From what I can tell, the registration process goes ok and suddenly the UE ends up detaching, but no idea why that may happen.

## Setup Details ##
Running both srsenb and srsepc on a Ubuntu 16.04 machine, 64bit, self-compiled from git master (commit c892ae56be5302eaee5ca00e270efc7a5ce6fbb2). The radio hardware is a USRP B210. The UE side is a Sierra Wireless MC7455 module, running firmware 02.33.03.00_GENERIC_002.072_000.

For what it's worth, I've also replaced srsepc with open5gs and the outcome is the same.

## Expected Behavior ##
The modules should successfully register in the network.

## Actual Behaviour ##
The modules fail to register in the network.

## Steps to reproduce the problem ##
In my case, just powering up the Sierra module triggers the automatic registration attempt.

## Additional Information ##
Attached to the bugreport are srsenb and srsepc logs, as well as PCAP capture from the EPC.

aleksander0m, 2020-06-16:
[enb.log](https://github.com/srsLTE/srsLTE/files/4787898/enb.log)
[epc.log](https://github.com/srsLTE/srsLTE/files/4787899/epc.log)
[epc.zip](https://github.com/srsLTE/srsLTE/files/4787900/epc.zip)

alvasMan, 2020-06-22:
Hi Aleksander --

It seems to me that there is a mismatch between the UE's APN and the EPC's APN. The EU is requesting as an APN "telefonica,es" and the EPC is replying with "srsapn".

Could you configure the UE's or the EPC's APN so that they match and let me know if the issue persists?

aleksander0m, 2020-06-22:
Hey @alvasMan thanks thanks for the check! That was actually it. It looks like the process in the Sierra modem is extremely strict w.r.t. the initial default LTE bearer settings. As soon as I configure the APN in the modem to match the one provided by the network, the process works flawlessly.

Closing this github issue, as it's definitely not a problem in srsLTE :D

closed 2020-06-22 by aleksander0m

alvasMan, 2020-06-23:
No problem, I'm glad that fixed it :)

srsEPC should really do a better job at notifying the mis-configuration to the user though, it's on my pipeline to fix that.