srsran/srsRAN_4G #507: The following tests FAILED: 	724 - nas_test (Child aborted)

WT4444, 2020-06-23:
After makeing  test ，an error occurred:The following tests FAILED: 	724 - nas_test (Child aborted),Errors while running CTest

andrepuschmann, 2020-06-24:
no further background provided. impossible to check

closed 2020-06-24 by andrepuschmann

WT4444, 2020-07-05:
After I finish installing srslte, when I execute the make test command,an error occurred:The following tests FAILED: 724 - nas_test (Child aborted),Errors while running CTest。

WT4444, 2020-07-05:
> no further background provided. impossible to check
According to the official website tutorial, after installing the uhd driver and related dependent libraries on ubuntu18.04, execute the install srsLTE command and execute the make test command prompt The following tests FAILED: 724-nas_test (Child aborted)