srsran/srsRAN_4G #510: srsenb: LimeSDR Mini Tx Calibration Issue

shanku9, 2020-06-29:
## Issue Description ##
I am trying to use LimeSDR mini to run enb function. I followed this (https://github.com/srsLTE/srsLTE/issues/437#issue-542647486) guide to setup my environment with LimeSDR Mini but keep running into different Tx Calibration Issues. Attached are the enb and ue logs with different EARFCN, Rx/Tx Antenna settings

## Setup Details ##
Operating System: Ubuntu 18.04.4
RF front-end: LimeSDR-Mini
LimeSuite: v20.01.0-myriadrf1~bionic

## Expected Behavior ##
No Tx calibration error. S1 Setup 

## Actual Behaviour ##
[ERROR] Tx Calibration: MCU error 4 (SXT tune failed)
srsLTE/lib/src/phy/rf/rf_soapy_imp.c.476: Failed to set Tx antenna

## Steps to reproduce the problem ##
Attached config files from enb, ue and epc

## Additional Information ##
For [ERROR] Tx Calibration: MCU error 4 (SXT tune failed), I followed instructions [here](https://discourse.myriadrf.org/t/how-to-fix-the-error-tx-calibration-mcu-error-4-sxt-tune-failed/4465/16) and was able to calibrate both Rx and Tx using LimeSuiteGUI and all board tests passed using LimeQuickTest but ran into same issue after running srsenb

[epc.txt](https://github.com/srsLTE/srsLTE/files/4843560/epc.txt)
[ue.txt](https://github.com/srsLTE/srsLTE/files/4843561/ue.txt)
[enb.txt](https://github.com/srsLTE/srsLTE/files/4843563/enb.txt)
[SoapySDRUtil.txt](https://github.com/srsLTE/srsLTE/files/4843571/SoapySDRUtil.txt)
[srsue_log.txt](https://github.com/srsLTE/srsLTE/files/4843572/srsue_log.txt)
[srsenb_logs.txt](https://github.com/srsLTE/srsLTE/files/4843573/srsenb_logs.txt)
[LimeQuickTest.txt](https://github.com/srsLTE/srsLTE/files/4843576/LimeQuickTest.txt)
[image]

kbarlee, 2020-06-29:
Hey

I believe the LimeSDR-Mini has an antenna path switch on it, so there is no need to specify
`device_args = rxant=LNAW,txant=BAND1`
in the config files. (I don't anyway). If you omit that line, are the results any different?

Kenny

shanku9, 2020-06-29:
@kbarlee Thanks for sharing the configuration setting but the results were same. Are you also running latest release on Lime mini? Attached is the enb config file and the enb/epc logs 

[enb.conf.txt](https://github.com/srsLTE/srsLTE/files/4846893/enb.conf.txt)
[srsenb.txt](https://github.com/srsLTE/srsLTE/files/4846895/srsenb.txt)
[srsepc.txt](https://github.com/srsLTE/srsLTE/files/4846894/srsepc.txt)

Theo-31, 2020-07-28:
@shanku9 Could you solve your issue ?

How far can you go with srsLTE with this TX calibration issue ?

mrlnc, 2020-07-29:
Coincidentally I was trying this too. Had no time yet to figure this out any further, but I want to share my notes:

* older releases of soapy & limesdr did somewhat work; very badly still but at least UE registration came through. I couldn't really find where the issues go from really bad to unusable.
* [there are instructions](https://discourse.myriadrf.org/t/how-to-fix-the-error-tx-calibration-mcu-error-4-sxt-tune-failed/4465/16) on doing the same tuning step with LimeSDR-GUI. The *same error appears*. Would that rule out issues in srsLTE and the soapy-layer?
* the failing tuning is frequency-dependent. follow the prior instructions for LimeSDR-GUI and select a very low frequency. If the low-freq VCO is selected, tuning works, if the mid-band VCO is selected, it fails with the error we're all talking about.

The LimeQuickTest finishes without error, through it should test the exact same thing.

As a side note, osmocom had a similar issue (i think they're not using the soapy-layer, though): https://osmocom.org/issues/3861

andrepuschmann, 2020-10-27:
As mentioned in #437 we've made several changes for the LimeSDR and have merged them into srsLTE. Also LimeSuite/SoapyLMS has been updated in th recent past. Please try again with the newest versions and reopen or create new issues if needed. Thanks

closed 2020-10-27 by andrepuschmann