srsran/srsRAN_4G #512: ZeroMQ Application in separated Machines for eNB - UE

jrmejiaa, 2020-07-03:
Hi Community,

I don't have any RF Equipment, therefore I want to use the ZeroMQ Application which fines really well in a single machine for eNB and the UE. However, I want to try to separate both things because I want to implement a more complicated machine for the UE Equipment but I don't find any information for this possibility.

I think that I have to change the arguments of **tx_port** and **rx_port** of the device in the configuration but I don't know how to do it correctly. Maybe I am doing something wrong if someone could help me, I really appreciate. Both machines are connected via ethernet with a interface who has a /24 Mask. 

I have tried the next configuration, but because the machine don't recognize one of the interfaces, there is an error about the creation of the socket to start the process. 

```bash
# enb.conf
tx_port=tcp://192.168.60.8:2000,rx_port=tcp://192.168.60.129:2001
# ue.conf
tx_port=tcp://192.168.60.129:2000,rx_port=tcp://192.168.60.8:2001
```

[image]

Thanks for your time and for the good work.

andrepuschmann, 2020-07-03:
the port that is tx on one side, needs to be rx on the other, so just swap them in your device args

jrmejiaa, 2020-07-03:
> the port that is tx on one side, needs to be rx on the other, so just swap them in your device args

Hi Andrea, thank your response, sorry for the dummy mistake. I made the wrong configuration I use this

```bash
# enb.conf
tx_port=tcp://192.168.60.8:2000,rx_port=tcp://192.168.60.129:2001
# ue.conf
tx_port=tcp://192.168.60.129:2000,rx_port=tcp://192.168.60.8:2001
```

when according with my own diagram was this one: 

```bash
# enb.conf
tx_port=tcp://192.168.60.129:2000,rx_port=tcp://192.168.60.8:2001
# ue.conf
tx_port=tcp://192.168.60.8:2000,rx_port=tcp://192.168.60.129:2001
```

An it works perfect, thank you and sorry for the dummy mistake. 

Happy coding, thanks for your work

closed 2020-07-03 by jrmejiaa