srsran/srsRAN_4G #514: srsUE: Adding MAC Address to the UE interface

jrmejiaa, 2020-07-03:
I want to use a SDN Controller to control the traffic of the simulated UE. For this purpose I need to create a VXLAN between the two UE and the SDN Controller (ONOS) uses the LLDP protocol to see if the links exists. I am having ping between the two machines, but I can't see the link between the two devices in the SDN Controller. After a troubleshooting using Wireshark in the interfaces of the controller and the UE, I found that the UE receives the LLDP package but the bridge does not send the MATCH to the Controller to acknowledge that the link exists. 

My though is that the difference is that the interface needs a MAC Address to create a complete preamble of the LLDP package (as far as I read about this protocol), so my question is: Can I add a MAC address to the interface that the srsUE creates? 

```bash
ue1: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 45.45.0.36  netmask 255.255.255.0  destination 45.45.0.36
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 5  bytes 420 (420.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 6  bytes 468 (468.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Thank you for your time

jrmejiaa, 2020-07-06:
Hi everyone, 

I was searching for this particular case, I know is pretty particular, but maybe there are some other cases, where a MAC Address in the simulated UE is necessary. I have found that the TUN device, the virtual interface that it is used to receive the IP from the EPC, does not support the MAC Address according with the documentation of [Ubuntu here](http://manpages.ubuntu.com/manpages/xenial/man5/systemd.netdev.5.html#[netdev]%20section%20options). For the developers it is a possibility to change the `TUN` devices for a `veth` device where the MAC Address and IP Address are indeed supported?

Thanks for the amazing work, I have been working with the srsLTE and OAI eNB and at least with the simulation without RF Hardware, srsLTE is by far the best version of them.

andrepuschmann, 2020-10-27:
Thanks for the warm words. But I am afraid we can't help you with your networking issue very much. Maybe StackOverflow is a better place to ask/search for issues like this. I am closing the ticket.

closed 2020-10-27 by andrepuschmann

triple-it, 2020-11-04:
What I have learned is that:
- Tun devices are raw-ip only (so no MAC), point-to-point, and no ARP.
- Tap devices are ethernet-style

The POINTOPOINT means that on this interface there is no Layer 2 addressing (no MAC address) on this interface:
no ARP requests (IPv4);
no NDP requests (IPv6);
the neighbour table is useless for this interface (ip neighbour);
in routing table entries for this interface the via directive is ignored;
packets on this interface are always send to the same (only) next-hop.

Examples of POINTOPOINT devices
- PPP interfaces: There is not Layer 2 address for PPP as this type of interface connects a single host to another host (hence the name "point-to-point protocol")
- TUN interfaces: They are IP only interfaces without a Layer 2.

Here are some great readings on this subject:
- https://backreference.org/2010/03/26/tuntap-interface-tutorial/
- http://rrendec.mindbit.ro/post/tun-tap-tunnels/
And of course the OpenVPN guides
- https://community.openvpn.net/openvpn/wiki/BridgingAndRouting