srsran/srsRAN_4G #515: uhd driver

WT4444, 2020-07-05:
## Issue Description ##
After the uhd driver installation is complete, execute the sudo uhd_usrp_probe command and [WARNING] [UDP] The send buffer could not be resized sufficiently.

## Setup Details ##
Run the installation driver command as follows:
sudo add-apt-repository ppa:ettusresearch/uhd
sudo apt-get update
sudo apt-get install libuhd-dev libuhd003 uhd-host
## Expected Behavior ##
No errors and warnings

## Actual Behaviour ##

[WARNING] [UDP] The send buffer could not be resized sufficiently.
Target sock buff size: 24266666 bytes.
Actual sock buff size: 1048576 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.wmem_max=24266666

## Steps to reproduce the problem ##

## Additional Information ##
ubuntu18.04

MoneyCatcher, 2020-07-06:
I think this is a ETTUS issue and not SRSLTE.

e.g. https://files.ettus.com/manual/page_usrp_x3x0_config.html#x3x0cfg_hostpc_netcfg_sockbuff

Did you try to run sudo sysctl -w net.core.wmem_max=24266666

BR

andrepuschmann, 2020-07-06:
> I think this is a ETTUS issue and not SRSLTE.

Exactly. Therefore closing the issue.

closed 2020-07-06 by andrepuschmann

WT4444, 2020-07-07:
> I think this is a ETTUS issue and not SRSLTE.
> e.g. https://files.ettus.com/manual/page_usrp_x3x0_config.html#x3x0cfg_hostpc_netcfg_sockbuff
> Did you try to run sudo sysctl -w net.core.wmem_max=24266666
> BR
I try to run sudo sysctl -w net.core.wmem_max=24266666.Then execute the sudo uhd_usrp_probe command.No errors and warnings occur. After reboot ubuntu, execute uhd_usrp_probe command, the same warning message will appear.

WT4444, 2020-07-07:
> > 我认为这是一个ETTUS问题，而不是SRSLTE。
> 
> 究竟。因此，解决问题。
How can I solve this problem? Is it because the uhd driver does not match the usrp-2943 image?

WT4444, 2020-07-07:
Issue Description
step1:After installing the uhd driver, a warning appears when executing the sudo uhd_usrp_probe command.

   ” [WARNING] [UDP] The send buffer could not be resized sufficiently.
   Target sock buff size: 24266666 bytes.
   Actual sock buff size: 1048576 bytes.
   See the transport application notes on buffer resizing.
   Please run: sudo sysctl -w net.core.wmem_max=24266666“
step 2:Ignore the first step warning. Install dependencies and srsLTEE according to the official website.When installing srsLTE, an error occurred when executing the make test command.
“ ....
718/734 Test #718: rrc_meas_test ..................................... Passed 0.07 sec
Start 719: nas_test
719/734 Test #719: nas_test ..........................................***Exception: Child aborted 0.41 sec
Start 720: tft_test
720/734 Test #720: tft_test .......................................... Passed 0.09 sec
...

  99% tests passed, 1 tests failed out of 734
  Total Test time (real) = 215.78 sec
  The following tests FAILED:
  719 - nas_test (Child aborted)
  Errors while running CTest
  Makefile:140: recipe for target 'test' failed
 make: *** [test] Error 8”
step 3: Ignore the error of the second step,Run the install srsLTE command. After installing srsLTE, an error occurs when executing the sudo srsenb command.

”Failed to initiate SCTP socket. Attempting reconnection in 10 seconds
connect(): Connection refused
Failed to initiate SCTP socket. Attempting reconnection in 10 seconds“

Setup Details
Operation System:ubuntu 18.04
Hardware : USRP-2943R
uhd driver versions: V3.14.1.1 release

Specific steps are as follows:
1、sudo apt-get install cmake libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev。
2、Install srsGUI according to the official website tutorial。
3、Install libpcsclite-dev according to the official website tutorial。
4、Install UHD driver according to the official website tutorial。
sudo add-apt-repository ppa:ettusresearch/uhd
sudo apt-get update
sudo apt-get install libuhd-dev libuhd003 uhd-host

After installing the uhd driver, execute the sudo uhd_usrp_probe command, a warning appears, but the USRP device can be found。
” [WARNING] [UDP] The send buffer could not be resized sufficiently.
Target sock buff size: 24266666 bytes.
Actual sock buff size: 1048576 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.wmem_max=24266666“

5、Install srsLTE according to the official website tutorial.
git clone https://github.com/srsLTE/srsLTE.git
cd srsLTE
mkdir build
cd build
cmake ../
make
make test
When executing the make test command, an error occurs.
"....
718/734 Test #718: rrc_meas_test ..................................... Passed 0.07 sec
Start 719: nas_test
719/734 Test #719: nas_test ..........................................***Exception: Child aborted 0.41 sec
Start 720: tft_test
720/734 Test #720: tft_test .......................................... Passed 0.09 sec
...

  99% tests passed, 1 tests failed out of 734
  Total Test time (real) = 215.78 sec
  The following tests FAILED:
  719 - nas_test (Child aborted)
  Errors while running CTest
  Makefile:140: recipe for target 'test' failed
  make: *** [test] Error 8“
6、Execute installation command。
sudo make install
srslte_install_configs.sh user

After installation, execute the sudo srsenb command, prompting an error.

"Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz for cc_idx=0
connect(): Connection refused
Failed to initiate SCTP socket. Attempting reconnection in 10 seconds
connect(): Connection refused
Failed to initiate SCTP socket. Attempting reconnection in 10 seconds
connect(): Connection refused"

What should I do?

MoneyCatcher, 2020-07-07:
Use /etc/sysctl.conf to make it permanent. And please try to avoid to mix all errors together. Also look into #516 which is maybe the same question. #517 is regarding a failed test and belongs to SRSLTE.

MoneyCatcher, 2020-07-07:
Please try to read the manuals of all components u use. Especially USRP UHD driver hints.

MoneyCatcher, 2020-07-07:
Is it possible to update to 3.15 UHD, I'm wondering why you build 3.14 from sources instead of installing 3.15 in Ubuntu 18.04?

```
apt install -y software-properties-common

 # UHD (3.15)
 # https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd
 add-apt-repository -y ppa:ettusresearch/uhd

  # update
  apt update -y

  #apt upgrade
  apt full-upgrade -y

  apt install -y libboost-all-dev libfftw3-dev libmbedtls-dev libconfig++-dev libsctp-dev \
    libuhd-dev uhd-host libusb-1.0-0-dev python3-mako
```

Please also check, that you have a proper firmware suitable to you UHD driver on the Ettus (NI) Device.

WT4444, 2020-07-08:
I can choose to install the version from the source code. How to match UHD driver version and srsLTE version on Ubuntu 18.04.

WT4444, 2020-07-08:
> Use /etc/sysctl.conf to make it permanent. And please try to avoid to mix all errors together. Also look into #516 which is maybe the same question. #517 is regarding a failed test and belongs to SRSLTE.

I have tried it but still not work.

MoneyCatcher, 2020-07-08:
Please go first into verifying the operation of ettus (NI) device, independent from SRSLTE.

https://kb.ettus.com/Verifying_the_Operation_of_the_USRP_Using_UHD_and_GNU_Radio

Use tools like **benchmark_rate**, as described in this documentation. If this works without any warning or error you can return to SRSLTE.

Try to be focused on every component you want to use.
First, remove the warnings regarding ETTUS UHD driver permanently.

After this making your remaining issues more clearer (focused on one single clear topic), because this is not a helpdesk or forum.
It is very unlikely, that you found 3 issues regarding SRSLTE at the same time, because it is very stable, periodicly checked and used by many people.

Good luck
BR

joeldelosangeles, 2021-09-22:
Hi, 

It seems I have upgraded successfully to UHD 3.15:
uhd_find_devices
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-66-g35e6c41b

uhd_usrp_probe
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-66-g35e6c41b

Why is srsenb still showing or using UHD 3.14?

root@srslte:/usr/local/bin# srsenb
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srslte/enb.conf...

Built in Release mode using commit c892ae56b on branch master.

/root/srsLTE/srsenb/src/enb_cfg_parser.cc.883: Force DL EARFCN for cell PCI=1 to 1850

Opening 1 channels in RF device=default with args=num_recv_frames=64,num_send_frames=64
[INFO] [UHD] linux; GNU C++ version 7.4.0; Boost_106501; UHD_3.14.1.1-release

Why is srsenb still showing or using UHD 3.14?