srsran/srsRAN_4G #517: srsLTE Test error

WT4444, 2020-07-06:
## Issue Description ##
step1:After installing the uhd driver, a warning appears when executing the sudo uhd_usrp_probe command.

       ” [WARNING] [UDP] The send buffer could not be resized sufficiently.
       Target sock buff size: 24266666 bytes.
       Actual sock buff size: 1048576 bytes.
       See the transport application notes on buffer resizing.
       Please run: sudo sysctl -w net.core.wmem_max=24266666“

step 2:Ignore the first step warning.  Install dependencies and srsLTEE according to the official website.When installing srsLTE, an error occurred when executing the make test command.
      “  ....
        718/734 Test #718: rrc_meas_test .....................................   Passed    0.07 sec
        Start 719: nas_test
       719/734 Test #719: nas_test ..........................................***Exception: Child aborted  0.41 sec
        Start 720: tft_test
       720/734 Test #720: tft_test ..........................................   Passed    0.09 sec
       ...

      99% tests passed, 1 tests failed out of 734
      Total Test time (real) = 215.78 sec
      The following tests FAILED:
      719 - nas_test (Child aborted)
      Errors while running CTest
      Makefile:140: recipe for target 'test' failed
     make: *** [test] Error 8”

step 3: Ignore the error of the second step,Run the install srsLTE command. After installing srsLTE, an error occurs when executing the sudo srsenb command.

”Failed to initiate SCTP socket. Attempting reconnection in 10 seconds
connect(): Connection refused
Failed to initiate SCTP socket. Attempting reconnection in 10 seconds“

## Setup Details ##
 Operation System:ubuntu 18.04
 Hardware              : USRP-2943R
 uhd driver versions: V3.14.1.1 release
 
Specific steps are as follows:
1、sudo apt-get install cmake libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev。
2、Install srsGUI according to the official website tutorial。
3、Install libpcsclite-dev according to the official website tutorial。
4、Install UHD driver according to the official website tutorial。
     sudo add-apt-repository ppa:ettusresearch/uhd
     sudo apt-get update
     sudo apt-get install libuhd-dev libuhd003 uhd-host

   After installing the uhd driver, execute the sudo uhd_usrp_probe command, a warning appears, but the USRP device can    be found。
  ” [WARNING] [UDP] The send buffer could not be resized sufficiently.
  Target sock buff size: 24266666 bytes.
  Actual sock buff size: 1048576 bytes.
  See the transport application notes on buffer resizing.
  Please run: sudo sysctl -w net.core.wmem_max=24266666“
   
5、Install srsLTE according to the official website tutorial.
     git clone https://github.com/srsLTE/srsLTE.git
    cd srsLTE
    mkdir build
    cd build
    cmake ../
    make
    make test
   When executing the make test command, an error occurs.
   "....
        718/734 Test #718: rrc_meas_test .....................................   Passed    0.07 sec
        Start 719: nas_test
       719/734 Test #719: nas_test ..........................................***Exception: Child aborted  0.41 sec
        Start 720: tft_test
       720/734 Test #720: tft_test ..........................................   Passed    0.09 sec
       ...

      99% tests passed, 1 tests failed out of 734
      Total Test time (real) = 215.78 sec
      The following tests FAILED:
      719 - nas_test (Child aborted)
      Errors while running CTest
      Makefile:140: recipe for target 'test' failed
      make: *** [test] Error 8“

6、Execute installation command。
     sudo make install
    srslte_install_configs.sh user

    After installation, execute the sudo srsenb command, prompting an error.

    "Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz for cc_idx=0
    connect(): Connection refused
    Failed to initiate SCTP socket. Attempting reconnection in 10 seconds
   connect(): Connection refused
   Failed to initiate SCTP socket. Attempting reconnection in 10 seconds
   connect(): Connection refused"

  
What should I do?

andrepuschmann, 2020-07-08:
Please use only one GH issue per issue. You are mixing things up. I am closing this one. If you want to trace down the NAS one, please open a new issue, provide full logs and console output, etc. Otherwise we can't help. We will close all issues that don't follow the guidelines.

closed 2020-07-08 by andrepuschmann

WT4444, 2020-07-08:
> Please use only one GH issue per issue. You are mixing things up. I am closing this one. If you want to trace down the NAS one, please open a new issue, provide full logs and console output, etc. Otherwise we can't help. We will close all issues that don't follow the guidelines.

Install dependencies and srsLTEE according to the official website.When installing srsLTE, an error occurred when executing the make test command.
“ ....
718/734 Test #718: rrc_meas_test ..................................... Passed 0.07 sec
Start 719: nas_test
719/734 Test #719: nas_test ..........................................***Exception: Child aborted 0.41 sec
Start 720: tft_test
720/734 Test #720: tft_test .......................................... Passed 0.09 sec
...

  99% tests passed, 1 tests failed out of 734
  Total Test time (real) = 215.78 sec
  The following tests FAILED:
  719 - nas_test (Child aborted)
  Errors while running CTest
  Makefile:140: recipe for target 'test' failed
 make: *** [test] Error 8”