srsran/srsRAN_4G #520: 99% tests passed, 1 tests failed out of 462

WT4444, 2020-07-09:
Install srsLTE according to the official website tutorial，After executing make and cmake commands, executing make test prompts an error.

99% tests passed, 1 tests failed out of 462

Total Test time (real) = 124.20 sec

The following tests FAILED:
	462 - nas_test (SEGFAULT)
Errors while running CTest
Makefile:140: recipe for target 'test' failed

Setup Details:
Operation System : ubuntu 18.04
         
Hardware : USRP-2943R
uhd driver versions: V3.14.1.1 release

CPU     ：
cat /proc/cpuinfo
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 158
model name	: Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz
stepping	: 10
microcode	: 0x96
cpu MHz		: 3191.994
cache size	: 12288 KB
physical id	: 0
siblings	: 1
core id		: 0
cpu cores	: 1
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes

andrepuschmann, 2020-07-09:
Can you please run the nas_test manually and provide the console output?

alvasMan, 2020-07-09:
There is an concurrency issue in the NAS test where you might try to stop the stack thread before you are finished starting it. 
That might be causing your issues.

Could you try the following:
```
diff --git a/srsue/test/upper/nas_test.cc b/srsue/test/upper/nas_test.cc
index a9665cd43..803fbec39 100644
--- a/srsue/test/upper/nas_test.cc
+++ b/srsue/test/upper/nas_test.cc
@@ -311,6 +311,8 @@ int mme_attach_request_test()
     gw.init(gw_args, logger, &stack);
     stack.init(&nas);
 
+    usleep(5000); // Wait for stack to initialize before stoping it.
+
     // trigger test
     stack.switch_on();
     stack.stop();
```

WT4444, 2020-07-09:
> Can you please run the nas_test manually and provide the console output?

How to manually run the nas_test command?

andrepuschmann, 2020-07-09:
In your build folder, run `./srsue/test/upper/nas_test`

WT4444, 2020-07-09:
> Can you please run the nas_test manually and provide the console output?

Run the nas_test command manually. the prompt is as follows

20:15:00.457535 [NAS] [I] Sending Security Mode Complete nas_current_ctxt.tx_count=0, RB=lcid
Error deallocating PDU: Addr=0x0x55587e8c1430
20:15:00.458019 [USIM] [E] Invalid length for OPc: 0 should be 32
Invalid length for OPc: 0 should be 32
20:15:00.461468 [USIM] [I] Read Home PLMN Id=00101

20:15:00.461689 [NAS] [I] Connection established correctly. Waiting for Attach
20:15:00.461696 [NAS] [E] Timeout or received attach reject while trying to attach
Failed to Attach
20:15:00.461705 [NAS] [E] Could not attach in attach request

20:15:00.461799 [GW] [I] TUN file descriptor = 4
20:15:00.461813 [GW] [D] Failed to set TUN device name: Operation not permitted
20:15:00.461825 [GW] [E] init_if failed
20:15:00.461831 [NAS] [E] Failed to set gateway address - (null)
20:15:00.461841 [NAS] [I] Sending Attach Complete

andrepuschmann, 2020-07-09:
apply the fix and run again. Please post full output

WT4444, 2020-07-09:
> apply the fix and run again. Please post full output
21:22:31.672211 [NAS] [I] DL lcid PDU
             0000: 07 52 01 0c 63 a8 54 13 e6 a4 ce d9 86 fb e5 ce 
             0010: 9b 62 5e 10 67 57 b3 c2 b9 70 90 01 0c 72 8a 67 
             0020: 57 92 52 b8 
21:22:31.672404 [NAS] [I] DL lcid Decrypted PDU
             0000: 07 52 01 0c 63 a8 54 13 e6 a4 ce d9 86 fb e5 ce 
             0010: 9b 62 5e 10 67 57 b3 c2 b9 70 90 01 0c 72 8a 67 
             0020: 57 92 52 b8 
21:22:31.672497 [NAS] [I] Received Authentication Request
21:22:31.672544 [NAS] [I] MCC=61441, MNC=65281
21:22:31.672586 [NAS] [D] Authentication request RAND
             0000: 0c 63 a8 54 13 e6 a4 ce d9 86 fb e5 ce 9b 62 5e 
21:22:31.672634 [NAS] [D] Authentication request AUTN
             0000: 67 57 b3 c2 b9 70 90 01 0c 72 8a 67 57 92 52 b8 
21:22:31.672726 [NAS] [I] Network authentication successful
21:22:31.672767 [NAS] [I] Sending Authentication Response
21:22:31.672798 [NAS] [I] Generated k_asme:
             0000: 8d 42 d1 bb 80 4b b4 53 cc 00 6d ed 4b 82 dd 55 
             0010: de 66 8b 2b 61 99 73 5d 2d ae 72 fa 24 70 97 ed 
21:22:31.672866 [NAS] [I] DL lcid PDU
             0000: 37 37 c7 67 ae 00 07 5d 02 01 02 e0 60 c1 
21:22:31.672913 [NAS] [I] DL lcid Decrypted PDU
             0000: 37 37 c7 67 ae 00 07 5d 02 01 02 e0 60 c1 
21:22:31.672963 [NAS] [I] Received Security Mode Command ksi: 1, eea: EEA0, eia: 128-EIA2
21:22:31.672994 [NAS] [I] NAS encryption key - k_nas_enc
             0000: 38 61 66 94 5c 30 0e 92 49 7e fb ce 3b 2c ad 96 
             0010: ef 3c 83 4b f8 a5 eb 4c e3 85 db 60 e9 92 85 9e 
21:22:31.673060 [NAS] [I] NAS integrity key - k_nas_int
             0000: ff 79 be dd c2 b9 39 eb 0b 2f cc 30 ef 6d 00 37 
             0010: 5b ae 51 b3 c0 26 2a 09 2e 2f b3 e5 d3 bc 0f 44 
21:22:31.673127 [NAS] [D] Generating integrity check. integ_algo:2, count_dl:0, lcid:1
21:22:31.673183 [NAS] [I] Integrity check ok. Local: count=0, Received: count=0
21:22:31.673269 [NAS] [I] Sending Security Mode Complete nas_current_ctxt.tx_count=0, RB=lcid
Error deallocating PDU: Addr=0x0x55e421441430
21:22:31.673788 [USIM] [E] Invalid length for OPc: 0 should be 32
Invalid length for OPc: 0 should be 32
21:22:31.677159 [USIM] [I] Read Home PLMN Id=00101
21:22:31.677289 [NAS] [I] Skip reading context from file.
21:22:31.677341 [NAS] [I] Attach Request
21:22:31.677369 [NAS] [I] No PLMN selected. Starting PLMN Search...
21:22:31.677397 [NAS] [I] Found PLMN:  Id=00101, TAC=65535
Found PLMN:  Id=00101, TAC=65535
21:22:31.677446 [NAS] [I] Selecting Home PLMN Id=00101
21:22:31.677484 [NAS] [I] Generating attach request
21:22:31.677561 [NAS] [I] Generating PDN Connectivity Request
Setting PDN protocol to IPv4
21:22:31.677642 [NAS] [I] Requesting IMSI attach (IMSI=001010123456789)
NAS generated SDU (len=22):
[07 41 01 08 09 10 10 10 32 54 76 98 02 e0 60 00 05 02 01 d0 11 d1 ];
21:22:31.677728 [NAS] [I] Connection established correctly. Waiting for Attach
21:22:31.677756 [NAS] [E] Timeout or received attach reject while trying to attach
Failed to Attach
21:22:31.677805 [NAS] [E] Could not attach in attach request
21:22:31.677832 [NAS] [I] DL lcid PDU
             0000: 27 00 00 00 00 00 07 42 01 3e 06 00 00 f1 10 00 
             0010: 01 00 2a 52 01 c1 01 04 1b 07 74 65 73 74 31 32 
             0020: 33 06 6d 6e 63 30 30 31 06 6d 63 63 30 30 31 04 
             0030: 67 70 72 73 05 01 c0 a8 05 02 27 01 80 50 0b f6 
             0040: 00 f1 10 80 01 01 35 16 6d bc 64 01 00 
21:22:31.677992 [NAS] [I] Integrity check ok. Local: count=0, Received: count=0
21:22:31.678023 [NAS] [I] DL lcid Decrypted PDU
             0000: 27 00 00 00 00 00 07 42 01 3e 06 00 00 f1 10 00 
             0010: 01 00 2a 52 01 c1 01 04 1b 07 74 65 73 74 31 32 
             0020: 33 06 6d 6e 63 30 30 31 06 6d 63 63 30 30 31 04 
             0030: 67 70 72 73 05 01 c0 a8 05 02 27 01 80 50 0b f6 
             0040: 00 f1 10 80 01 01 35 16 6d bc 64 01 00 
21:22:31.678173 [NAS] [I] Received Attach Accept
21:22:31.678241 [NAS] [I] Network attach successful. APN: test123.mnc001.mcc001.gprs, IP: 192.168.5.2
Network attach successful. IP: 192.168.5.2
21:22:31.678368 [GW] [I] TUN file descriptor = 4
21:22:31.678412 [GW] [D] Failed to set TUN device name: Operation not permitted
21:22:31.678444 [GW] [E] init_if failed
21:22:31.678471 [NAS] [E] Failed to set gateway address - (null)
21:22:31.678504 [NAS] [I] Sending Attach Complete
21:22:31.683108 [NAS] [I] Skip reading context from file.
21:22:31.684655 [NAS] [I] DL lcid PDU
             0000: 27 00 00 00 00 00 02 5a d9 
21:22:31.684726 [NAS] [I] Integrity check ok. Local: count=0, Received: count=0
21:22:31.684767 [NAS] [I] DL lcid Decrypted PDU
             0000: 27 00 00 00 00 00 02 5a d9 
21:22:31.684836 [NAS] [I] ESM information request received for beaser=0, transaction_id=90
21:22:31.684895 [NAS] [D] Including APN srslte in ESM info response
21:22:31.684936 [NAS] [D] Including CHAP authentication for user srsuser in ESM info response
21:22:31.684967 [NAS] [I] Sending ESM information response
             0000: 27 00 00 00 00 00 02 5a da 28 07 06 73 72 73 6c 
             0010: 74 65 27 3f 80 c2 23 1c 01 00 00 1c 10 c5 ae 9e 
             0020: 87 e4 e3 ec 35 1b 6a b4 d9 d6 93 97 bd 73 72 73 
             0030: 75 73 65 72 c2 23 1c 02 00 00 1c 10 89 da 11 88 
             0040: 54 8d 44 40 19 43 92 10 c8 91 1d d0 73 72 73 75 
             0050: 73 65 72

andrepuschmann, 2020-07-09:
so is it solved?

WT4444, 2020-07-09:
> so is it solved?
The error still exists

WT4444, 2020-07-09:
> > so is it solved?
> > The error still exists

Wǒ yīnggāi rúhé jiějué zhège wèntí
11/5000
How should I solve this problem.

WT4444, 2020-07-09:
> > so is it solved?
> > The error still exists
How should I solve this problem

andrepuschmann, 2020-07-09:
After you run `./srsue/test/upper/nas_test` can you post the output of `echo $?`, but right after it

WT4444, 2020-07-09:
> echo $?

WT4444, 2020-07-09:
> After you run `./srsue/test/upper/nas_test` can you post the output of `echo $?`, but right after it

After executing echo $? command, the error still exists.

WT4444, 2020-07-09:
> > After you run `./srsue/test/upper/nas_test` can you post the output of `echo $?`, but right after it
> 
> After executing echo $? command, the error still exists.
After executing the echo $? command, it returns 0, but the error still exists after executing the nas_test command

suttonpd, 2020-07-09:
Hi @WT4444 , if you continue to have problems please use the srslte-users mailing list - http://www.softwareradiosystems.com/mailman/listinfo/srslte-users

WT4444, 2020-07-09:
> diff --git a/srsue/test/upper/nas_test.cc

> 大家好@ WT4444，如果您仍然遇到问题，请使用srslte-users邮件列表-http: [//www.softwareradiosystems.com/mailman/listinfo/srslte-users](http://www.softwareradiosystems.com/mailman/listinfo/srslte-users)

WT4444, 2020-07-09:
> Hi @WT4444 , if you continue to have problems please use the srslte-users mailing list - http://www.softwareradiosystems.com/mailman/listinfo/srslte-users

ok, I will try it.

WT4444, 2020-07-09:
> There is an concurrency issue in the NAS test where you might try to stop the stack thread before you are finished starting it.
> That might be causing your issues.
>
> Could you try the following:
>
> ``` [...]

Can you elaborate, how will I execute my command.

closed 2020-07-09 by WT4444

WT4444, 2020-07-09:
> apply the fix and run again. Please post full output
What does it mean about the" apply the fix".

alvasMan, 2020-07-09:
> What does it mean about the" apply the fix".

OK, I'm not sure if you applied the fix or not, so I'll just explain it very thoroughly, just in case you didn't understand.

1 - Open `srsue/test/upper/nas_test.cc ` in you favorite text editor.
2 - Add the line that I told you previously, namely `usleep(5000);`, just after the `stack.init(&nas)` line.
3 - Re-compile srsLTE (`make && sudo make install && sudo ldconfig`)
4 - Run the tests again `ctest -j8`

If the tests are still failing, please send an e-mail to the mailing list and we will continue the discussion there.

WT4444, 2020-07-10:
> OK, I'm not sure if you applied the fix or not, so I'll just explain it very thoroughly, just in case you didn't understand.
>
> 1 - Open `srsue/test/upper/nas_test.cc ` in you favorite text editor.
> 2 [...]

Report an error after executing ctest.
99% tests passed, 2 tests failed out of 737

Total Test time (real) = 222.06 sec

The following tests FAILED:
	714 - ue_rf_failure_max_channels (Child killed)
	722 - nas_test (Child aborted)
Errors while running CTest

WT4444, 2020-07-10:
> OK, I'm not sure if you applied the fix or not, so I'll just explain it very thoroughly, just in case you didn't understand.
>
> 1 - Open `srsue/test/upper/nas_test.cc ` in you favorite text editor.
> 2 [...]

Manually execute the nas_test command, the error is as follows:
/srsue/test/upper$ ./nas_test
01:37:31.913043 [NAS ] [I] New state DEREGISTERED
01:37:31.913211 [NAS ] [I] DL lcid PDU
             0000: 07 52 01 0c 63 a8 54 13 e6 a4 ce d9 86 fb e5 ce 
             0010: 9b 62 5e 10 67 57 b3 c2 b9 70 90 01 0c 72 8a 67 
             0020: 57 92 52 b8 
01:37:31.913314 [NAS ] [I] DL lcid Decrypted PDU
             0000: 07 52 01 0c 63 a8 54 13 e6 a4 ce d9 86 fb e5 ce 
             0010: 9b 62 5e 10 67 57 b3 c2 b9 70 90 01 0c 72 8a 67 
             0020: 57 92 52 b8 
01:37:31.913402 [NAS ] [I] Received Authentication Request
01:37:31.913426 [NAS ] [I] MCC=61441, MNC=65281
01:37:31.913449 [NAS ] [D] Authentication request RAND
             0000: 0c 63 a8 54 13 e6 a4 ce d9 86 fb e5 ce 9b 62 5e 
01:37:31.913494 [NAS ] [D] Authentication request AUTN
             0000: 67 57 b3 c2 b9 70 90 01 0c 72 8a 67 57 92 52 b8 
01:37:31.913542 [NAS ] [I] Network authentication successful
01:37:31.913565 [NAS ] [D] Not applying security for PDU. No context configured.
01:37:31.913589 [NAS ] [I] Sending Authentication Response
01:37:31.913611 [NAS ] [I] Generated k_asme:
             0000: 8d 42 d1 bb 80 4b b4 53 cc 00 6d ed 4b 82 dd 55 
             0010: de 66 8b 2b 61 99 73 5d 2d ae 72 fa 24 70 97 ed 
01:37:31.913677 [NAS ] [I] DL lcid PDU
             0000: 37 4e fd 57 11 00 07 5d 02 01 02 f0 70 c1 
01:37:31.913722 [NAS ] [I] DL lcid Decrypted PDU
             0000: 37 4e fd 57 11 00 07 5d 02 01 02 f0 70 c1 
01:37:31.913766 [NAS ] [I] Received Security Mode Command ksi: 1, eea: EEA0, eia: 128-EIA2
01:37:31.913792 [NAS ] [I] NAS encryption key - k_nas_enc
             0000: 38 61 66 94 5c 30 0e 92 49 7e fb ce 3b 2c ad 96 
             0010: ef 3c 83 4b f8 a5 eb 4c e3 85 db 60 e9 92 85 9e 
01:37:31.913857 [NAS ] [I] NAS integrity key - k_nas_int
             0000: ff 79 be dd c2 b9 39 eb 0b 2f cc 30 ef 6d 00 37 
             0010: 5b ae 51 b3 c0 26 2a 09 2e 2f b3 e5 d3 bc 0f 44 
01:37:31.913922 [NAS ] [D] Generating integrity check. integ_algo:2, count_dl:0, lcid:1
01:37:31.913954 [NAS ] [I] Integrity check ok. Local: count=0, Received: count=0  [4e fd 57 11]
01:37:31.913989 [NAS ] [I] Sending Security Mode Complete nas_current_ctxt.tx_count=0, RB=lcid
01:37:31.943882 [NAS ] [I] New state DEREGISTERED
01:37:31.944089 [USIM] [I] Read Home PLMN Id=00101
01:37:31.944121 [NAS ] [E] Empty EIA list. Select at least one EIA algorithm.
01:37:31.944148 [NAS ] [E] Empty EEA list. Select at least one EEA algorithm.
01:37:31.944173 [NAS ] [I] Skip reading context from file.
01:37:31.955145 [NAS ] [I] Attach Request with cause mo-Data.
01:37:31.955165 [NAS ] [I] No PLMN selected. Starting PLMN Search...
01:37:31.955172 [NAS ] [I] Found PLMN:  Id=00101, TAC=65535
Found PLMN:  Id=00101, TAC=65535
01:37:31.955180 [NAS ] [I] Selecting Home PLMN Id=00101
01:37:31.955197 [NAS ] [I] Generating attach request
01:37:31.955207 [NAS ] [I] Generating PDN Connectivity Request
01:37:31.955211 [NAS ] [D] Requesting IPv4 PDN protocol
01:37:31.955215 [NAS ] [I] Requesting IMSI attach (IMSI=001010123456789)
01:37:31.955220 [NAS ] [D] Starting T3410
01:37:31.955229 [NAS ] [I] Proc "RRC Connect" - Starting...
NAS generated SDU (len=22):
[07 41 71 08 09 10 10 10 32 54 76 98 02 00 00 00 05 02 01 d0 11 d1 ];
01:37:31.955242 [NAS ] [I] Proc "RRC Connect" - Connection established correctly. Waiting for Attach
01:37:31.955247 [NAS ] [I] Proc "PLMN Search" - Starting...
01:37:31.955361 [NAS ] [E] Proc "RRC Connect" - Timeout or received attach reject while trying to attach
01:37:31.955368 [NAS ] [I] Proc "PLMN Search" - Completed with failure
01:37:31.955372 [NAS ] [D] Clearing EPS bearer context
01:37:31.955376 [NAS ] [I] New state DEREGISTERED
01:37:31.955379 [NAS ] [I] Attach Request from PLMN Search failed
01:37:31.955418 [NAS ] [I] Attach Request with cause mo-Data.
01:37:31.955421 [NAS ] [I] No PLMN selected. Starting PLMN Search...
01:37:31.955425 [NAS ] [I] Found PLMN:  Id=00101, TAC=65535
[... 98 lines cut ...]
             0010: 03 40 08 ae 5d 02 00 c2 81 34 01 4d 
01:37:32.001212 [NAS ] [I] Received Activate Dedicated EPS bearer context request (eps_bearer_id=6, linked_bearer_id=5, proc_id=0)
01:37:32.001235 [NAS ] [I] Traffic Flow Template: TFT OP code 0x1, Filter list size 1, Parameter list size 0
01:37:32.001260 [NAS ] [D] Not applying security for PDU. No context configured.
01:37:32.001284 [NAS ] [I] Sending Activate Dedicated EPS Bearer context accept (eps_bearer_id=6, proc_id=0)
             0000: 62 00 c6 
01:37:32.001329 [NAS ] [I] DL lcid PDU
             0000: 27 00 00 00 00 00 62 00 cd 24 
01:37:32.001374 [NAS ] [I] Integrity check ok. Local: count=1, Received: count=0  [00 00 00 00]
01:37:32.001397 [NAS ] [I] Update local count to received value 0
01:37:32.001420 [NAS ] [I] DL lcid Decrypted PDU
             0000: 27 00 00 00 00 00 62 00 cd 24 
01:37:32.001466 [NAS ] [I] Received Deactivate EPS bearer context request (eps_bearer_id=6, proc_id=0, cause=0x24)
01:37:32.001489 [NAS ] [I] Removed EPS bearer context (eps_bearer_id=6)
01:37:32.001512 [NAS ] [D] Not applying security for PDU. No context configured.
01:37:32.001536 [NAS ] [I] Sending Deactivate EPS Bearer context accept (eps_bearer_id=6, proc_id=0)
[... cut ...]
01:37:32.001581 [NAS ] [I] DL lcid PDU
             0000: 27 00 00 00 00 00 62 00 cd 24 
01:37:32.001626 [NAS ] [I] Integrity check ok. Local: count=1, Received: count=0  [00 00 00 00]
01:37:32.001648 [NAS ] [I] Update local count to received value 0
01:37:32.001671 [NAS ] [I] DL lcid Decrypted PDU
             0000: 27 00 00 00 00 00 62 00 cd 24 
01:37:32.001717 [NAS ] [I] Received Deactivate EPS bearer context request (eps_bearer_id=6, proc_id=0, cause=0x24)
01:37:32.001739 [NAS ] [E] EPS bearer doesn't exist (eps_bearer_id=6)

WT4444, 2020-07-11:
> Hi @WT4444 , if you continue to have problems please use the srslte-users mailing list - http://www.softwareradiosystems.com/mailman/listinfo/srslte-users
Do you mean let me send an email to srslte-users-owner@lists.softwareradiosystems.com mail

fazhiwang, 2021-03-06:
Have you solved your problem? I have the same problem.