srsran/srsRAN_4G #521: srsLTE installation warning

WT4444, 2020-07-11:
Install srsLTE according to the official website, execute make to compile and report a warning.

/srsLTE/build# make 
Scanning dependencies of target srslte_asn1
[  0%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_common.cc.o
cc1plus: warning: ‘-Werror=’ argument ‘-Werror=incompatible-pointer-types’ is not valid for C++
cc1plus: warning: unrecognized command line option ‘-Wno-parantheses’
[  1%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_rrc.cc.o
cc1plus: warning: ‘-Werror=’ argument ‘-Werror=incompatible-pointer-types’ is not valid for C++
cc1plus: warning: unrecognized command line option ‘-Wno-parantheses’
[  1%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_mme.cc.o
cc1plus: warning: ‘-Werror=’ argument ‘-Werror=incompatible-pointer-types’ is not valid for C++
cc1plus: warning: unrecognized command line option ‘-Wno-parantheses’
[  1%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_s1ap.cc.o
cc1plus: warning: ‘-Werror=’ argument ‘-Werror=incompatible-pointer-types’ is not valid for C++
cc1plus: warning: unrecognized command line option ‘-Wno-parantheses’
[  2%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/gtpc.cc.o
cc1plus: warning: ‘-Werror=’ argument ‘-Werror=incompatible-pointer-types’ is not valid for C++
cc1plus: warning: unrecognized command line option ‘-Wno-parantheses’
[  2%] Linking CXX static library libsrslte_asn1.a
[  2%] Built target srslte_asn1
Scanning dependencies of target gen_build_info

Operation System : ubuntu 18.04
Hardware : USRP-2943R
uhd driver versions: V3.14.1.1 release
srsLTE      version    ：release_18_09

MoneyCatcher, 2020-07-13:
Hi WT4444,

which gcc are you using.

This warning comes from your compiler.

Which is installed?

`gcc --version`

Maybe try another one:
https://linuxize.com/post/how-to-install-gcc-compiler-on-ubuntu-18-04/

Or try srsLTE 19.12 as an alternative on Ubuntu 18.04, if possible.

BR

WT4444, 2020-07-18:
Hello：
     gcc --version
    gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
    I have tried srsLTE 19.12 as an alternative on Ubuntu 18.04.
   Error still exists.

https://linuxize.com/post/how-to-install-gcc-compiler-on-ubuntu-18-04/

andrepuschmann, 2020-10-27:
Closing issue due to lack of feedback from the author. In general, we compile and run fine on Ubuntu 18.04

closed 2020-10-27 by andrepuschmann