srsran/srsRAN_4G #522: srsENB/UE with ZeroMQ - UEContextReleaseRequest from eNB with user-inactivity

2-face, 2020-07-11:
## Issue Description ##
While testing srsENB/UE with ZeroMQ (after following guidelines) I am getting UEContextReleaseRequest from eNB with cause user-inactivity.

## Setup Details ##
Latest release from github with ZeroMQ and commercial EPC.

## Expected Behavior ##
No UEContextReleaseRequest from gNB after ~300ms. It should appear only after **rrc_inactivity_timer** expirese.

## Actual Behaviour ##
Very quickly after ~300ms or less after ESM Information Response is sent back to MME, eNB sends UEContextReleaseRequest.

## Steps to reproduce the problem ##
ZeroMQ setup as per guidelines.

2-face, 2020-07-11:
[enb.log](https://github.com/srsLTE/srsLTE/files/4906788/enb.log)
[ue.log](https://github.com/srsLTE/srsLTE/files/4906789/ue.log)

andrepuschmann, 2020-10-27:
Could you check if the issue still exists in the 20.10 release?

ismagom, 2023-04-14:
Closing due to inactivity from user

closed 2023-04-14 by ismagom