srsran/srsRAN_4G #526: unable to build 20_04_1 with gcc-10

ZeroChaos-, 2020-07-23:
GCC 10 is a bit more strict, and the latest tag cannot be built with it.  I haven't tested git yet

```
>>> Emerging (1 of 1) net-wireless/srslte-20.04.1::maintainer-zero
 * srslte-20.04.1.tar.gz BLAKE2B SHA512 size ;-) ...                                                                                                                                   [ ok ]
>>> Unpacking source...
 * Unpacking srslte-20.04.1.tar.gz
>>> Source unpacked in /var/tmp/portage/net-wireless/srslte-20.04.1/work
>>> Preparing source in /var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1 ...
 * Working in BUILD_DIR: "/var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1_build"
 * Hardcoded definition(s) removed in CMakeLists.txt:
 *     set(CMAKE_BUILD_TYPE Release)
 *  set(CMAKE_BUILD_TYPE ${CMAKE_BUILD_TYPE} CACHE STRING "")
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1 ...
 * Working in BUILD_DIR: "/var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1_build"
cmake -C /var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DENABLE_UHD=yes -DENABLE_BLADERF=yes -DENABLE_SOAPYSDR=yes -DENABLE_ZEROMQ=yes -DENABLE_HARDSIM=yes -DCMAKE_BUILD_TYPE=Gentoo -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1_build/gentoo_toolchain.cmake  /var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1
loading initial cache file /var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1_build/gentoo_common_config.cmake
-- The C compiler identification is GNU 10.1.0
-- The CXX compiler identification is GNU 10.1.0
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_SYSTEM: Linux-5.7.8-pentoo
-- CMAKE_SYSTEM_PROCESSOR: x86_64
-- CMAKE_CXX_COMPILER: /usr/bin/x86_64-pc-linux-gnu-g++
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: x86_64-pc-linux-gnu-pkg-config (found version "0.29.2") 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.6-pl2
-- FFTW3F LIBRARIES: /usr/lib64/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: FFTW3F_STATIC_LIBRARY-NOTFOUND
-- FFTW3F INCLUDE DIRS: /usr/include
-- Found fftw3f: /usr/lib64/libfftw3f.so  
-- FFT_LIBRARIES: /usr/lib64/libfftw3f.so
-- Checking for module 'polarssl'
--   No package 'polarssl' found
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND
-- POLARSSL INCLUDE DIRS: POLARSSL_INCLUDE_DIRS-NOTFOUND
-- Could NOT find POLARSSL (missing: POLARSSL_LIBRARIES POLARSSL_INCLUDE_DIRS) 
-- Checking for module 'mbedtls'
--   No package 'mbedtls' found
-- MBEDTLS LIBRARIES: /usr/lib/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: MBEDTLS_STATIC_LIBRARIES-NOTFOUND
-- MBEDTLS INCLUDE DIRS: /usr/include
-- Found MBEDTLS: /usr/lib/libmbedcrypto.so  
-- Checking for module 'libpcsclite'
--   Found libpcsclite, version 1.8.25
-- Found PCSCLITE: /usr/lib64/libpcsclite.so  
-- PCSC LIBRARIES: /usr/lib64/libpcsclite.so
-- PCSC INCLUDE DIRS: /usr/include/PCSC
-- Building with PCSC support.
[... 150 lines cut ...]
/var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1/lib/include/srslte/common/move_callback.h: In member function ‘R srslte::task_details::empty_table_t<R, Args>::call(void*, Args&& ...) const’:
/var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1/lib/include/srslte/common/move_callback.h:33:11: error: ‘stderr’ was not declared in this scope
   33 |   fprintf(stderr, "ERROR: exception thrown due to bad function call (cause: %s)\n", cause);                            \
      |           ^~~~~~
/var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1/lib/include/srslte/common/move_callback.h:76:59: note: in expansion of macro ‘THROW_BAD_FUNCTION_CALL’
   76 |   R         call(void* src, Args&&... args) const final { THROW_BAD_FUNCTION_CALL("function ptr is empty"); }
      |                                                           ^~~~~~~~~~~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1/lib/include/srslte/common/move_callback.h:27:1: note: ‘stderr’ is defined in header ‘<cstdio>’; did you forget to ‘#include <cstdio>’?
   26 | #include <functional>
  +++ |+#include <cstdio>
   27 | #include <type_traits>
/var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1/lib/include/srslte/common/move_callback.h:33:3: error: there are no arguments to ‘fprintf’ that depend on a template parameter, so a declaration of ‘fprintf’ must be available [-fpermissive]
   33 |   fprintf(stderr, "ERROR: exception thrown due to bad function call (cause: %s)\n", cause);                            \
      |   ^~~~~~~
/var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1/lib/include/srslte/common/move_callback.h:76:59: note: in expansion of macro ‘THROW_BAD_FUNCTION_CALL’
   76 |   R         call(void* src, Args&&... args) const final { THROW_BAD_FUNCTION_CALL("function ptr is empty"); }
      |                                                           ^~~~~~~~~~~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1/lib/include/srslte/common/move_callback.h:33:3: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
   33 |   fprintf(stderr, "ERROR: exception thrown due to bad function call (cause: %s)\n", cause);                            \
      |   ^~~~~~~
/var/tmp/portage/net-wireless/srslte-20.04.1/work/srslte-20.04.1/lib/include/srslte/common/move_callback.h:76:59: note: in expansion of macro ‘THROW_BAD_FUNCTION_CALL’
   76 |   R         call(void* src, Args&&... args) const final { THROW_BAD_FUNCTION_CALL("function ptr is empty"); }
      |                                                           ^~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```

andrepuschmann, 2020-07-23:
It's already fixed in our dev repo. Check #496 until it becomes available officially.

closed 2020-07-23 by andrepuschmann

ZeroChaos-, 2020-07-23:
You have a non-public dev repo? Why not just do it in public so people can get fixes and help?  This seems very less than ideal.