srsran/srsRAN_4G #529: LTE links disconnected when distance longer than 4 meters

Torreskai0722, 2020-07-29:
## Issue Description ##

Steps:
1. Connect VERT2450 antenna to two USRP boards and each one connected to a x86-64 host machine. 2. Set up srsLTE on two host machine. All the configurations are using default value. In one host machine, run ```sudo srsepc``` and ```sudo srsenb```. In another host machine, run ```sudo srsue```.
3. LTE links can be set up and two host machines get IP address 172.16.0.1 and 172.16.0.x. However, when the distance of two USRP B210 boards larger than 3-4 meters, the connection lost.

## Setup Details ##
SDR devices: two USRP B210
Antennas: two VERT 2450
Host machine: Ubuntu 18.04
SDR dirver: UHD
srsLTE: 19.12

## Expected Behavior ##
the LTE connection can be maintained longer

## Actual Behaviour ##
when the distance of two USRP B210 boards larger than 3-4 meters, the connection lost.

## Steps to reproduce the problem ##
choose the default setup and increase the distance between the two devices

## Additional Information ##
N/A

andrepuschmann, 2020-09-14:
Can you please provide some logs and console traces for both units so we can check power levels, etc. It's likely something like that.

Torreskai0722, 2020-09-16:
Here are the logs for EPC, eNB, and UE. EPC and eNB is launched on one machine, UE is launched on another machine.

EPC:
```
---  Software Radio Systems EPC  ---

Reading configuration file /home/mist/.config/srslte/epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:001, MNC:01, PLMN: 61712
S1 Setup Request - TAC 7, B-PLMN 0
S1 Setup Request - Paging DRX 2
Sending S1 Setup Response
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- M-TMSI: 0x53a41069
Attach request -- eNB-UE S1AP Id: 1
Attach request -- Attach type: 1
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: false
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 1
PDN Connectivity Request -- ESM Information Transfer requested: false
UL NAS: Received Identity Response
ID Response -- IMSI: 001010123456789
Downlink NAS: Sent Authentication Request
UL NAS: Received Authentication Response
Authentication Response -- IMSI 001010123456789
UE Authentication Accepted.
Generating KeNB with UL NAS COUNT: 0
Downlink NAS: Sending NAS Security Mode Command.
UL NAS: Received Security Mode Complete
Security Mode Command Complete -- IMSI: 001010123456789
Getting subscription information -- QCI 7
Sending Create Session Request.
Creating Session Response -- IMSI: 1010123456789
Creating Session Response -- MME control TEID: 1
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
SPGW: Allocated Ctrl TEID 1
SPGW: Allocated User TEID 1
SPGW: Allocate UE IP 172.16.0.2
Received Create Session Response
Create Session Response -- SPGW control TEID 1
Create Session Response -- SPGW S1-U Address: 127.0.1.100
SPGW Allocated IP 172.16.0.2 to IMSI 001010123456789
Adding attach accept to Initial Context Setup Request
Initial Context Setup Request -- eNB UE S1AP Id 1, MME UE S1AP Id 1
Initial Context Setup Request -- E-RAB id 5
Initial Context Setup Request -- S1-U TEID 0x1. IP 127.0.1.100
Initial Context Setup Request -- S1-U TEID 0x1. IP 127.0.1.100
Initial Context Setup Request -- QCI 7
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x460003; eNB GTP-U Address 127.0.1.1
UL NAS: Received Attach Complete
Unpacked Attached Complete Message. IMSI 1010123456789
Unpacked Activate Default EPS Bearer message. EPS Bearer id 5
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_MODIFY_BEARER_REQUEST
Sending EMM Information
Received UE Context Release Request. MME-UE S1AP Id 1
There are active E-RABs, send release access bearers request
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_RELEASE_ACCESS_BEARERS_REQUEST
[... 90 lines cut ...]
Network attach successful. IP: 172.16.0.2
Software Radio Systems LTE (srsLTE)
/home/edge22/projects/srsLTE/lib/src/phy/phch/pusch.c.429: Error trying to allocate 3456 symbols but 1440 were allocated (tti=3467, short=0, L=10)

/home/edge22/projects/srsLTE/lib/src/phy/ue/ue_ul.c.301: Error encoding PUSCH

/home/edge22/projects/srsLTE/lib/src/phy/phch/pusch.c.429: Error trying to allocate 3456 symbols but 1440 were allocated (tti=3475, short=0, L=10)

/home/edge22/projects/srsLTE/lib/src/phy/ue/ue_ul.c.301: Error encoding PUSCH

/home/edge22/projects/srsLTE/lib/src/phy/phch/pusch.c.429: Error trying to allocate 3456 symbols but 1440 were allocated (tti=3483, short=0, L=10)

/home/edge22/projects/srsLTE/lib/src/phy/ue/ue_ul.c.301: Error encoding PUSCH

Received RRC Connection Release
RRC IDLE
^CStopping srsUE... Press Ctrl+C 9 more times to force stop
Random Access Transmission: seq=26, ra-rnti=0x2
Random Access Complete. 	c-rnti=0x48, ta=1
RRC Connected
---  exiting  ---

```

> Can you please provide some logs and console traces for both units so we can check power levels, etc. It's likely something like that.

andrepuschmann, 2020-09-16:
Please send log files too, and set them to `--log.all_level info`. Consult the docs for more info. Thanks

Torreskai0722, 2020-09-16:
[ue.log](https://github.com/srsLTE/srsLTE/files/5234065/ue.log)
[enb.log](https://github.com/srsLTE/srsLTE/files/5234070/enb.log)
[epc.log](https://github.com/srsLTE/srsLTE/files/5234072/epc.log)

andrepuschmann, 2020-09-21:
From the eNB logs it seems your uplink gets too weak over time, so the connection dies eventually. I guess this is because you are walking away. Maybe your gain and/or antenna settings aren't great. But it doesn't look like an srsLTE issue. Closing therefore

closed 2020-09-21 by andrepuschmann

Torreskai0722, 2020-09-22:
@andrepuschmann Do you have any advice on how to change the gain and antenna settings? Thanks!

hmartinezf, 2020-11-10:
VERT2450 shall feature some acceptable gain for reception in Band 7, and also good VSWR try to check it with nanoVNA V2.
Separate the antennas.
tx_gain & rx_gain are configurable in enb.conf
Check received signal level with Network Cell Info Lite android app. If level is good over distance then the uplink is not working, replace Rx antenna.

fatafalihhilmi, 2022-08-31:
can you send me enb.conf and epc.conf? thanks @Torreskai0722

Torreskai0722, 2022-09-02:
> can you send me enb.conf and epc.conf? thanks @Torreskai0722

Here is the link for all the config files: https://drive.google.com/drive/folders/1QZHRYhIeSb2XBCW2OMFpnDBjiu6n0RD8?usp=sharing