srsran/srsRAN_4G #531: Automatic selection of soapy device id

KloolK, 2020-08-06:
Hi,

I had some trouble getting soapy to select my LimeSDR-USB, because of the installed "audio" soapy module which detects sound cards. When checking the source code on how to tell soapy which device to use, I read the following lines and noted that it says "last device", but uses the MIN function where I would expect MAX to be used to select the highest and not the lowest available id. 

https://github.com/srsLTE/srsLTE/blob/d94078d543ceed275cd7b6dd2261592cbf1f0d38/lib/src/phy/rf/rf_soapy_imp.c#L328-L329

In my case this would have caused soapy to automatically pick the lime instead of the audio devices. I'm not sure about the general case, though.

Do you think this should be changed?

andrepuschmann, 2020-08-06:
The logic makes sure that whatever `dev_id` is selected, it is not larger than `length - 1`, which is the last available device. It just protects from selecting a device outside of the available ones. To me, the code looks ok. Could you perhaps share your console output and expected behavior.

KloolK, 2020-08-13:
I misinterpreted the comment and thought this line is supposed to select the last id in case multiple ones are found. All good then :)

However, it would have saved me some trouble if the `dev_id` parameter had been documented in the `enb.conf`. I'll add an example to the file and send a pull request later.

KloolK, 2020-08-14:
After playing around with it a little further I think there is something to improve here:

If I have the soapyaudio module installed, audio devices always occur before any Lime devices in the list of devices from Soapy:
```
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srslte/enb.conf...

Built in Release mode using commit f02bfe2cf on branch master.

Opening 1 channels in RF device=soapy with args=default
[INFO] [UHD] linux; GNU C++ version 10.1.0; Boost_107200; UHD_3.15.0.0-0-unknown
Soapy has found device #0: default_input=True, default_output=True, device_id=0, driver=audio, label=PulseAudio, 
Soapy has found device #1: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 3.0] 9060B00470B31, media=USB 3.0, module=FX3, name=LimeSDR-USB, serial=0009060B00470B31, 
Selecting Soapy device: 0
```

Since audio devices are not supported by srsLTE currently, I think Lime devices should be preferred over any other soapy devices by default. I would expect the following (note the device id is `1` instead of `0`):
```
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srslte/enb.conf...

Built in Release mode using commit f02bfe2cf on branch master.

Opening 1 channels in RF device=soapy with args=default
[INFO] [UHD] linux; GNU C++ version 10.1.0; Boost_107200; UHD_3.15.0.0-0-unknown
Soapy has found device #0: default_input=True, default_output=True, device_id=0, driver=audio, label=PulseAudio, 
Soapy has found device #1: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 3.0] 9060B00470B31, media=USB 3.0, module=FX3, name=LimeSDR-USB, serial=0009060B00470B31, 
Selecting Soapy device: 1
```

Looking at the code, the name of soapy devices is already checked, but does not influence the chosen device:
https://github.com/srsLTE/srsLTE/blob/f02bfe2cff6bc0685a0418ea61debde3e036de18/lib/src/phy/rf/rf_soapy_imp.c#L301-L304

andrepuschmann, 2020-10-27:
The code snippet referenced above indeed doesn't impact the device selection. The user is supposed to tell the driver which device to select if auto detection doesn't pick the correct one. I don't think that adding any heuristics here will be helpful. I am closing the issue as I don't see any problem in our code.

closed 2020-10-27 by andrepuschmann

Li-markus, 2021-01-19:
The problem is that how to tell the driver which device to select ? Especially when people is using the cell_search function of srsLTE.
I have the same problem recently, the Soapy always select the device 0 instead of 1

KloolK, 2021-01-19:
You can achieve that by using something like the following in your enb.conf:
`device_args = id=1`

See
https://github.com/srsLTE/srsLTE/blob/45486b6e2c622ab1d47906521662f087816d7d03/lib/src/phy/rf/rf_soapy_imp.c#L313-L329