srsran/srsRAN_4G #532: USRP B210  Error opening UHD code 44 in release_20_04

tomchean, 2020-08-11:
## When update srsLTE from release_19_12 to release_20_04, the UHD fails ##

When I use release_19_12, everything works well, however when I update to release_20_04, UHD return error code 44. 
and I note that the UHD driver version used is different in these two release, in release_19_12 is UHD_3.14.1.1-release, and in release_20_04 it shows UHD_003.010.003.000-0-unknown, I wonder which one is correct, and how can I specify the driver when run srsenb ?

## Setup Details ##
OS : Ubuntu-18.04 + USRPB210

## Additional Information ##
release_20_04 snapshot 

[image]

release_19_12 snapshot 
[image]

frankist, 2020-08-11:
Try using the LTS versions of the UHD. Either 3.9 or 3.15

tomchean, 2020-09-07:
I solve it after upgrade HUD to 3.15, thanks.

andrepuschmann, 2020-09-07:
Great. Thanks for the feedback.

closed 2020-09-07 by andrepuschmann

matthew-william-lock, 2020-10-04:
How do you specify the UHD version?

tomchean, 2020-10-28:
you can see CmakeCache to see where's the path of UHD library.