srsran/srsRAN_4G #539: Bugs: srsEPC crashed after illegal UE sends crafted NAS messages during Uplink NAS Transport

langeye0409, 2020-08-30:
## Issue Description ##
A bug can be triggered by crafted NAS messages, which makes EPC crashed.

## Setup Details ##
ubuntu 18.04,
RF front-end: USRP B210
srsLTE version: release 20.04.1

## Expected Behavior ##
Ignore illegal messages

## Actual Behaviour ##
srsepc crashed.

## Steps to reproduce the problem ##
As attached pcap file,
UE sends "Attach Request" with GUTI, and waits "Identity Request".
After receiving Identity Request, the UE sends 4 bytes or 5 bytes crafted NAS messages with security header type set to 1 (Integrity protected).

## Additional Information ##
[image]

[pcap.zip](https://github.com/srsLTE/srsLTE/files/5146754/pcap.zip)

djanderson, 2020-10-20:
It looks like the issue was in `security_128_eia1` and `snow3g_f9`, both of which were recently rewritten by @davidrupprecht in https://github.com/srsLTE/srsLTE/pull/497. I wonder if you can reproduce on the `next` branch?

andrepuschmann, 2020-10-27:
Thanks for reporting the issue. It's quite possible that the bug still exists.

davidrupprecht, 2021-04-22:
Dear @langeye0409 and @djanderson, thank you for the issue. Could you reproduce the issue with the srsLTE version 20.10.1. If yes, could you upload also the logs of the EPC. Thanks, David

andrepuschmann, 2021-05-14:
Closing issue due to user inactivity. Please reopen if it still exist.

closed 2021-05-14 by andrepuschmann