srsran/srsRAN_4G #541: srsLTE can not install

rembulan, 2020-09-02:
I get error when try to install srsLTE 

Scanning dependencies of target srslte_rf
[ 47%] Building C object lib/src/phy/rf/CMakeFiles/srslte_rf.dir/rf_soapy_imp.c.o
In file included from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:36:0:
/usr/include/SoapySDR/Types.h:25:3: error: conflicting types for ‘SoapySDRRange’
 } SoapySDRRange;
   ^~~~~~~~~~~~~
In file included from /usr/local/include/SoapySDR/Device.h:18:0,
                 from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Types.h:25:3: note: previous declaration of ‘SoapySDRRange’ was here
 } SoapySDRRange;
   ^~~~~~~~~~~~~
In file included from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:36:0:
/usr/include/SoapySDR/Types.h:33:3: error: conflicting types for ‘SoapySDRKwargs’
 } SoapySDRKwargs;
   ^~~~~~~~~~~~~~
In file included from /usr/local/include/SoapySDR/Device.h:18:0,
                 from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Types.h:33:3: note: previous declaration of ‘SoapySDRKwargs’ was here
 } SoapySDRKwargs;
   ^~~~~~~~~~~~~~
In file included from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:36:0:
/usr/include/SoapySDR/Types.h:39:30: error: conflicting types for ‘SoapySDRKwargs_fromString’
 SOAPY_SDR_API SoapySDRKwargs SoapySDRKwargs_fromString(const char *markup);
                              ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/SoapySDR/Device.h:18:0,
                 from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Types.h:39:30: note: previous declaration of ‘SoapySDRKwargs_fromString’ was here
 SOAPY_SDR_API SoapySDRKwargs SoapySDRKwargs_fromString(const char *markup);
                              ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:36:0:
/usr/include/SoapySDR/Types.h:45:21: error: conflicting types for ‘SoapySDRKwargs_toString’
 SOAPY_SDR_API char *SoapySDRKwargs_toString(const SoapySDRKwargs *args);
                     ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/SoapySDR/Device.h:18:0,
                 from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Types.h:45:21: note: previous declaration of ‘SoapySDRKwargs_toString’ was here
 SOAPY_SDR_API char *SoapySDRKwargs_toString(const SoapySDRKwargs *args);
                     ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:36:0:
/usr/include/SoapySDR/Types.h:50:5: error: redeclaration of enumerator ‘SOAPY_SDR_ARG_INFO_BOOL’
     SOAPY_SDR_ARG_INFO_BOOL,
     ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/SoapySDR/Device.h:18:0,
                 from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Types.h:50:5: note: previous definition of ‘SOAPY_SDR_ARG_INFO_BOOL’ was here
     SOAPY_SDR_ARG_INFO_BOOL,
     ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:36:0:
/usr/include/SoapySDR/Types.h:51:5: error: redeclaration of enumerator ‘SOAPY_SDR_ARG_INFO_INT’
     SOAPY_SDR_ARG_INFO_INT,
     ^~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/SoapySDR/Device.h:18:0,
                 from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Types.h:51:5: note: previous definition of ‘SOAPY_SDR_ARG_INFO_INT’ was here
     SOAPY_SDR_ARG_INFO_INT,
     ^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:36:0:
/usr/include/SoapySDR/Types.h:52:5: error: redeclaration of enumerator ‘SOAPY_SDR_ARG_INFO_FLOAT’
     SOAPY_SDR_ARG_INFO_FLOAT,
     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/SoapySDR/Device.h:18:0,
                 from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:
/usr/local/include/SoapySDR/Types.h:52:5: note: previous definition of ‘SOAPY_SDR_ARG_INFO_FLOAT’ was here
     SOAPY_SDR_ARG_INFO_FLOAT,
     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:36:0:
/usr/include/SoapySDR/Types.h:53:5: error: redeclaration of enumerator ‘SOAPY_SDR_ARG_INFO_STRING’
     SOAPY_SDR_ARG_INFO_STRING
[... 91 lines cut ...]
                                    ^
In file included from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:0:
/usr/local/include/SoapySDR/Device.h:294:19: note: expected ‘const SoapySDRKwargs * {aka const struct <anonymous> *}’ but argument is of type ‘SoapySDRKwargs * {aka struct <anonymous> *}’
 SOAPY_SDR_API int SoapySDRDevice_setupStream(SoapySDRDevice *device,
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:398:36: error: passing argument 7 of ‘SoapySDRDevice_setupStream’ from incompatible pointer type [-Werror=incompatible-pointer-types]
                                    &stream_args) != 0) {
                                    ^
In file included from /home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:31:0:
/usr/local/include/SoapySDR/Device.h:294:19: note: expected ‘const SoapySDRKwargs * {aka const struct <anonymous> *}’ but argument is of type ‘SoapySDRKwargs * {aka struct <anonymous> *}’
 SOAPY_SDR_API int SoapySDRDevice_setupStream(SoapySDRDevice *device,
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:433:31: error: invalid initializer
   SoapySDRRange tx_range    = SoapySDRDevice_getGainRange(handler->device, SOAPY_SDR_TX, 0);
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/theboel/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:434:31: error: invalid initializer
   SoapySDRRange rx_range    = SoapySDRDevice_getGainRange(handler->device, SOAPY_SDR_RX, 0);
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
lib/src/phy/rf/CMakeFiles/srslte_rf.dir/build.make:86: recipe for target 'lib/src/phy/rf/CMakeFiles/srslte_rf.dir/rf_soapy_imp.c.o' failed
make[2]: *** [lib/src/phy/rf/CMakeFiles/srslte_rf.dir/rf_soapy_imp.c.o] Error 1
CMakeFiles/Makefile2:3854: recipe for target 'lib/src/phy/rf/CMakeFiles/srslte_rf.dir/all' failed
make[1]: *** [lib/src/phy/rf/CMakeFiles/srslte_rf.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

andrepuschmann, 2020-09-02:
What version of srsLTE and Soapy? Please use the issue template and fill all relevant fields, otherwise your questions cannot be answered.

andrepuschmann, 2020-09-14:
Closing issue as requested information was not provided.

closed 2020-09-14 by andrepuschmann