srsran/srsRAN_4G #542: I installed and run srslte but it not connecting to my phone ? Why ?

muratoznn, 2020-09-02:
## Issue Description ##
[Describe the issue in detail]

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

suttonpd, 2020-09-03:
Please use the issues template to provide more information.

andrepuschmann, 2020-09-14:
Closing the issue since requested information was not provided.

closed 2020-09-14 by andrepuschmann